espressif/arduino-esp32 #8347: Error Compiling adc_types.h with Arduino IDE 2.1.0

Patronics, 2023-06-25:
### Board

ESP32-PICO-D4

### Device Description

Watchy V2.0

### Hardware Configuration

N/A

### Version

v2.0.9

### IDE Name

Arduino IDE 2.1.0

### Operating System

MacOS 

### Flash frequency

N/A

### PSRAM enabled

no

### Upload speed

N/A

### Description

When compiling from the Arduino IDE 2.1.0, I get the following error (but the same program compiles fine in the Arduino IDE 1.8.16). This seems to be the same error as in issue #6647, which is vague about how it was fixed, if it was at all (comments seem to say the issue is with PlatformIO, but I'm not using PlatformIO.

```In file included from /Users/patrick/Library/Arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32/include/driver/include/driver/adc.h:14:0,
                 from /Users/patrick/Library/Arduino15/packages/esp32/hardware/esp32/2.0.9/cores/esp32/esp32-hal-adc.c:16:
/Users/patrick/Library/Arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32/include/hal/include/hal/adc_types.h:90:27: error: expected ',' or '}' before '__attribute__'
     ADC_DIGI_FORMAT_12BIT __attribute__((deprecated)),  /*!<ADC to DMA data format,                [15:12]-channel, [11: 0]-12 bits ADC data (`adc_digi_output_data_t`). Note: For single convert mode. */
```

### Sketch

```cpp
the examples here: https://github.com/sqfmi/Watchy
```

### Debug Message

```plain
/Users/patrick/Library/Arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32/include/driver/include/driver/adc.h:14:0,
                 from /Users/patrick/Library/Arduino15/packages/esp32/hardware/esp32/2.0.9/cores/esp32/esp32-hal-adc.c:16:
/Users/patrick/Library/Arduino15/packages/esp32/hardware/esp32/2.0.9/tools/sdk/esp32/include/hal/include/hal/adc_types.h:90:27: error: expected ',' or '}' before '__attribute__'
     ADC_DIGI_FORMAT_12BIT __attribute__((deprecated)),  /*!<ADC to DMA data format,                [15:12]-channel, [11: 0]-12 bits ADC data (`adc_digi_output_data_t`). Note: For single convert mode. */
```

### Other Steps to Reproduce

It works fine in Arduino 1.8.16, but gives the above error with Arduino 2.1.0

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2023-06-26:
same here, can you please help with triage @P-R-O-C-H-Y?

P-R-O-C-H-Y, 2023-06-26:
@VojtechBartoska I am not able to reproduce the error with Arduino IDE 2.1.0.

P-R-O-C-H-Y, 2023-06-26:
@Patronics Can you please send more information on how to replicate the issue? Exact example will be handy :)

HenniePeters, 2023-08-04:
I had the same error in Arduino IDE 1.8.19
I could solve it by updating ESP32 in the boards manager from 2.0.4 to 2.0.11