espressif/arduino-esp32 #8348: Error when building a project for ESP32-C6 in Arduino 1.8.15

tankist-git-2, 2023-06-25:
### Board

ESP32-C6-WROOM-1

### Device Description

I use official ESP32-C6-WROOM-1 module which is installed into ESP32-DevKitC-32E board

### Hardware Configuration

Nothing use exept power, reset, boot and UART.

### Version

other

### IDE Name

Arduino IDE 1.8.15

### Operating System

Windows 10 21H2

### Flash frequency

40

### PSRAM enabled

no

### Upload speed

921600

### Description

1) I downloaded an archive with ESP32-C6 branch from https://github.com/espressif/arduino-esp32/tree/feature/esp32c6-support
2) I replaced my 2.0.9 folders in c:\Users\user\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.9 with new folders
3) I try to build empty project for ESP32-C6 module but I get an error:
```
���⥬� �� 㤠���� ���� 㪠����� ����.
exit status 1
```
The error is unreadable as you see.
I analyzed the commands and I see the last command try to execute file
`C:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.9/tools/esptool/esptool.exe`
which doesn't exist in my system. In "tools" directory even there is no a directory named "esptool"

I guess there is a bug in a build script or I didn't do some steps to make build possible because it's not mentioned.

You can see the detailed build information output below.

### Sketch

```cpp
void setup() {
}

void loop() {
}
```

### Debug Message

```plain
C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\user\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\user\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries D:\Doc\Projects\Arduino\libraries -fqbn=esp32:esp32:esp32c6:JTAGAdapter=default,CDCOnBoot=default,PartitionScheme=default,CPUFreq=160,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none,EraseFlash=none -vid-pid=1A86_7523 -ide-version=10815 -build-path C:\Users\user\AppData\Local\Temp\arduino_build_242956 -warnings=none -build-cache C:\Users\user\AppData\Local\Temp\arduino_cache_356019 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc.path=C:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s2-elf-gcc\esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc-esp-2021r2-patch5-8.4.0.path=C:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s2-elf-gcc\esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.mkspiffs.path=C:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.openocd-esp32.path=C:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\openocd-esp32\v0.11.0-esp32-20221026 -prefs=runtime.tools.openocd-esp32-v0.11.0-esp32-20221026.path=C:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\openocd-esp32\v0.11.0-esp32-20221026 -prefs=runtime.tools.esptool_py.path=C:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.5.1 -prefs=runtime.tools.esptool_py-4.5.1.path=C:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.5.1 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-esp-2021r2-patch5-8.4.0.path=C:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.mklittlefs.path=C:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933 -prefs=runtime.tools.mklittlefs-3.0.0-gnu12-dc7f933.path=C:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933 -prefs=runtime.tools.xtensa-esp-elf-gdb.path=C:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp-elf-gdb\11.2_20220823 -prefs=runtime.tools.xtensa-esp-elf-gdb-11.2_20220823.path=C:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp-elf-gdb\11.2_20220823 -prefs=runtime.tools.xtensa-esp32s3-elf-gcc.path=C:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s3-elf-gcc\esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.xtensa-esp32s3-elf-gcc-esp-2021r2-patch5-8.4.0.path=C:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s3-elf-gcc\esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.riscv32-esp-elf-gcc.path=C:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.riscv32-esp-elf-gcc-esp-2021r2-patch5-8.4.0.path=C:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.riscv32-esp-elf-gdb.path=C:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-elf-gdb\11.2_20220823 -prefs=runtime.tools.riscv32-esp-elf-gdb-11.2_20220823.path=C:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-elf-gdb\11.2_20220823 -verbose D:\Doc\Projects\Arduino\esp32c6_first\esp32c6_first.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\user\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\user\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries D:\Doc\Projects\Arduino\libraries -fqbn=esp32:esp32:esp32c6:JTAGAdapter=default,CDCOnBoot=default,PartitionScheme=default,CPUFreq=160,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none,EraseFlash=none -vid-pid=1A86_7523 -ide-version=10815 -build-path C:\Users\user\AppData\Local\Temp\arduino_build_242956 -warnings=none -build-cache C:\Users\user\AppData\Local\Temp\arduino_cache_356019 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc.path=C:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s2-elf-gcc\esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc-esp-2021r2-patch5-8.4.0.path=C:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s2-elf-gcc\esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.mkspiffs.path=C:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.openocd-esp32.path=C:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\openocd-esp32\v0.11.0-esp32-20221026 -prefs=runtime.tools.openocd-esp32-v0.11.0-esp32-20221026.path=C:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\openocd-esp32\v0.11.0-esp32-20221026 -prefs=runtime.tools.esptool_py.path=C:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.5.1 -prefs=runtime.tools.esptool_py-4.5.1.path=C:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.5.1 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-esp-2021r2-patch5-8.4.0.path=C:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.mklittlefs.path=C:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933 -prefs=runtime.tools.mklittlefs-3.0.0-gnu12-dc7f933.path=C:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933 -prefs=runtime.tools.xtensa-esp-elf-gdb.path=C:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp-elf-gdb\11.2_20220823 -prefs=runtime.tools.xtensa-esp-elf-gdb-11.2_20220823.path=C:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp-elf-gdb\11.2_20220823 -prefs=runtime.tools.xtensa-esp32s3-elf-gcc.path=C:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s3-elf-gcc\esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.xtensa-esp32s3-elf-gcc-esp-2021r2-patch5-8.4.0.path=C:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s3-elf-gcc\esp-2021r2-patch5-8.4.0 -prefs=runtime.tools.riscv32-esp-elf-gcc.path=C:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-el [... cut ...] lash_size 4MB -o "C:\\Users\\user\\AppData\\Local\\Temp\\arduino_build_242956\\esp32c6_first.ino.bootloader.bin" "C:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.9/tools/esp32-arduino-libs/esp32c6\\bin\\bootloader_qio_80m.elf" ) )
���⥬� �� 㤠���� ���� 㪠����� ����.
exit status 1
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-06-26:
Read the [manual installation instructions](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html)

SuGlider, 2023-06-26:
>I downloaded an archive with ESP32-C6 branch from https://github.com/espressif/arduino-esp32/tree/feature/esp32c6-support
I replaced my 2.0.9 folders in c:\Users\user\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.9 with new folders
I try to build empty project for ESP32-C6 module but I get an error:

This is not the correct way to install the C6 branch as an Arduino Board.
You must install it as a decribed above by @lbernstone.
After restarting Arduino IDE, you will see a new "ESP32 Arduino (in sketchbook)" board in the menu. This new board will have the ESP32-C6 listed there. You also **must** run `get.exe` from the your `Arduino Sketch Folder/ hardware/espressif/esp32/tools/get.exe` in order to download the toolchain for the C6.

[image]

me-no-dev, 2023-06-27:
I would suggest to NOT install it now and wait a bit until we release 2.0.10 and move 3.0.0 in master.

VojtechBartoska, 2023-08-18:
Closing as duplicate.. Please follow this issue https://github.com/espressif/arduino-esp32/issues/7713

closed 2023-08-18 by VojtechBartoska