espressif/arduino-esp32 #8350: Mqtt lib

podaen, 2023-06-26:
### Related area

Wifi

### Hardware specification

ESP32-S1-S2-S3-C3

### Is your feature request related to a problem?

No

### Describe the solution you'd like

Now Wifi seems to be working properly. Are there any plans to make a basic library for an ESP32 that uses the build in Arduino events?

### Additional context

https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/esp32/include/mqtt/esp-mqtt/include/mqtt_client.h

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

SuGlider, 2023-06-26:
@podaen - The `mqtt_client.h` is actually an IDF MQTT implementation that is not supported within ESP32 Arduino.
You may try to adapt it into an Arduino Sketch based on the IDF example from
https://github.com/espressif/esp-idf/blob/master/examples/protocols/mqtt5/main/app_main.c

VojtechBartoska, 2023-06-27:
Closing as asnwered, if needed you can reopen it.

closed 2023-06-27 by VojtechBartoska

podaen, 2023-06-27:
If it programs that fast as you are closing... I know it comes from IDF. I tried to make it work a year ago, but without success. I am short in time now... so I will leave it closed. Any example would be welcome to start with! I also build the Wifi connection back then completely with IDF's in the ESP Arduino core, but I couldn't connect. PS: It's not my first Mqtt. I was stuck on the connection, so in my case it didn't work.