espressif/arduino-esp32 #8353: Debug ESP32-C3 using JTAG over USB on Arduino IDE 2.x

KurtE, 2023-06-26:
### Board

ESP32-C3-DevKitC-02 and ESP32-C3-DevkitM-1 and LuotOS.com/t/esp32C3

### Device Description

I have tried 3 different setups for ESP32-C3

[image]

For the: ESP32-C3-DevKitC-02 and ESP32-C3-DevkitM-1 I use breadboard, to a USB breakout board.

For LuotOS version, the USB connector is actually connected to pins IO18 and IO19.  More details about this board up at: https://m.media-amazon.com/images/I/91S2q9qyLcL.pdf

### Hardware Configuration

Nothing other than mentioned above where IO18 and IO19 connected to USB connector.

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE 2.1 and and nightly build 20230625

### Operating System

Windows 11

### Flash frequency

80mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

Using the Arduino IDE 2.x debug facility on the ESP32-C3 does not appear to work.  Initially it looks like the debugger starts up.  But trying to do anything causes it to hang or crash.

Note: this is related to #8334 by @mjs513 but that issue started off as more of a question or issue on how to setup the JTAG for the ESP32-C3 and programming, and this one has more to do once the JTAG over USB is setup, and it appears like the debugger starts up.  You soon find that nothing appears to work.

If you feel like this is a duplicate, feel free to close it as such.  Some groups like to group things together and others like each specific thing broken out as a new issue.

### Sketch

```cpp
Any sketch will do, but current one:

#include "Freenove_WS2812_Lib_for_ESP32.h"

#define LEDS_COUNT  1
#define LEDS_PIN	8
#define CHANNEL		0

Freenove_ESP32_WS2812 strip = Freenove_ESP32_WS2812(LEDS_COUNT, LEDS_PIN, CHANNEL, TYPE_GRB);

void setup() {
  strip.begin();
  strip.setBrightness(20);  
}

void loop() {
  for (int j = 0; j < 255; j += 2) {
[... 119 lines cut ...]
Debug: 348978 743373 gdb_server.c:1533 gdb_read_memory_packet(): addr: 0x000000003fc9bfc0, len: 0x00000040
Debug: 348979 743374 target.c:2522 target_read_buffer(): reading buffer of 64 byte at 0x3fc9bfc0
Debug: 348980 743379 riscv-013.c:2859 log_mem_access_result(): Succeeded to read memory via system bus.
Debug: 348981 743379 gdb_server.c:407 gdb_log_outgoing_packet(): [esp32c3] sending packet: $a5a5a5a5a5a5a5a5a5a5a50a01000000000000000000000000000000000000000100000001000000000000004e0000420020033c0050c93f7cc0c93f00820042#6c
Debug: 348982 743383 gdb_server.c:390 gdb_log_incoming_packet(): [esp32c3] received packet: m3fc9c040,40
Debug: 348983 743383 gdb_server.c:1533 gdb_read_memory_packet(): addr: 0x000000003fc9c040, len: 0x00000040
Debug: 348984 743383 target.c:2522 target_read_buffer(): reading buffer of 64 byte at 0x3fc9c040
Debug: 348985 743387 riscv-013.c:2859 log_mem_access_result(): Succeeded to read memory via system bus.
Debug: 348986 743387 gdb_server.c:407 gdb_log_outgoing_packet(): [esp32c3] sending packet: $a5a5a5a5a5a5a5a5a5a5a5a582809235a5a5a5a5a5a5a5a5a5a5a5a5000000000100000000000000b0c0c93fe2850042a5a5a5a5a5a5a5a5a5a5a5a500000000#e2
Debug: 348987 743528 gdb_server.c:228 gdb_get_char_inner(): GDB connection closed by the remote client
Debug: 348988 743529 gdb_server.c:1115 gdb_connection_closed(): GDB Close, Target: esp32c3, state: halted, gdb_actual_connections=0
Debug: 348989 743529 target.c:1847 target_call_event_callbacks(): target event 8 (gdb-end) for core esp32c3
Debug: 348990 743529 esp32c3.c:177 esp32c3_handle_target_event(): 8
Debug: 348991 743531 esp_riscv.c:356 esp_riscv_handle_target_event(): 8
Debug: 348992 743531 target.c:1847 target_call_event_callbacks(): target event 23 (gdb-detach) for core esp32c3
Debug: 348993 743531 esp32c3.c:177 esp32c3_handle_target_event(): 23
Debug: 348994 743531 esp_riscv.c:356 esp_riscv_handle_target_event(): 23
Info : 348995 743531 server.c:569 server_loop(): dropped 'gdb' connection
[2023-06-26T14:24:38.569Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2023-06-27:
did you try adding a breakpoint in the loop and hitting the resume button for debug?

KurtE, 2023-06-27:
> did you try adding a breakpoint in the loop and hitting the resume button for debug?
Yes as well as other locations. 

[image]

The debug starts up. I press the continue button, it get to setup:
Debug tab pane shows:
```
Hardware assisted breakpoint 1 at 0x4200004c: file C:\Users\kurte\Documents\Arduino\Rainbow/Rainbow.ino, line 9.
Note: automatically using hardware breakpoints for read-only addresses.
[New Thread 1070463660]
[New Thread 1070177460]
[New Thread 1070169048]
[New Thread 1070174784]
[Switching to Thread 1070463660]
Thread 2 "loopTask" hit Temporary breakpoint 1, setup () at C:\Users\kurte\Documents\Arduino\Rainbow/Rainbow.ino:9
9	void setup() {
```
I hit the continue button and the looks like nothing happened, The screen looks the same as the image I showed above.

I restarted this.  I exited the IDE, turned off power to the ESP32-C3 and then turned on did a fresh upload, and then started up the debugger again
Here is the whole debug console and gdb server startup stuff...
Debug Console:
```
    ENOENT means program not found. If that is not the issue, please report this problem.
Launching gdb-server: "C:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\openocd-esp32\\v0.11.0-esp32-20221026/bin/openocd" -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s "c:\\Users\\kurte\\Documents\\Arduino\\Rainbow" -f "C:/arduino-ide_nightly/resources/app/plugins/cortex-debug/extension/support/openocd-helpers.tcl" -f debug.cfg -d3
    Please check TERMINAL tab (gdb-server) for output from C:\Users\kurte\AppData\Local\Arduino15\packages\esp32\tools\openocd-esp32\v0.11.0-esp32-20221026/bin/openocd
Finished reading symbols from objdump: Time: 26 ms
Finished reading symbols from nm: Time: 24 ms
Reading symbols from C:/Users\kurte\AppData\Local\Temp\arduino\sketches\E0082B7BB48CD9EC5B948DE681BAB496\Rainbow.ino.elf...
GNU gdb (esp-gdb) 11.2_20220823
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later 
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-w64-mingw32 --target=riscv32-esp-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
.
Find the GDB manual and other documentation resources online at:
    .
For help, type "help".
Type "apropos word" to search for commands related to "word".
Warning: 'set target-async', an alias for the command 'set mi-async', is deprecated.
Use 'set mi-async'.
0x40000000 in ?? ()
Program stopped, probably due to a reset and/or halt issued by debugger
JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Reset cause (3) - (Software core reset)
Hardware assisted breakpoint 1 at 0x4200004c: file C:\Users\kurte\Documents\Arduino\Rainbow/Rainbow.ino, line 9.
Note: automatically using hardware breakpoints for read-only addresses.
```
The server stuff is longer so included file instead:
[gdb-server-startup.txt](https://github.com/espressif/arduino-esp32/files/11882612/gdb-server-startup.txt)

I did the continue button (twice)... The only new stuff in debug console:
```
[New Thread 1070463660]
[New Thread 1070177460]
[New Thread 1070169048]
[New Thread 1070174784]
[Switching to Thread 1070463660]
Thread 2 "loopTask" hit Temporary breakpoint 1, setup () at C:\Users\kurte\Documents\Arduino\Rainbow/Rainbow.ino:9
```
And the GDB server one was again going endless with data, until I told the debugger to stop.
[gdb-server-aborted.txt](https://github.com/espressif/arduino-esp32/files/11882661/gdb-server-aborted.txt)

Note: I see in my temp directory a file that looks like the complete gdb log:
"C:\Users\kurte\AppData\Local\Temp\gdb-server-console-58868.log"
Which I thought about including here, but it is 10739kb in size and has 109K lines of text in it.
And probably not really needed as this is easy to reproduce.

Let me know if you need any additional information,
Kurt

VojtechBartoska, 2023-08-18:
Please help with triage of this ticket @me-no-dev, thanks

VojtechBartoska, 2023-12-12:
Hello, closing this as expired, if needed you can reopen.

closed 2023-12-12 by VojtechBartoska

eddiem3, 2024-01-06:
Is there any advice on resolving this issue?