espressif/arduino-esp32 #8354: ESP32 S3 Feather Stuck in constant Bootload loop.

potesd, 2023-06-26:
### Board

Adafruit ESP32 S3 Feather

### Device Description

Freshly purchased from Adafruit. 

### Hardware Configuration

There are two buttons attached to pins 12 and 13, and commonly grounded to Gnd.

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE 2.1.0

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

No matter what I do, the ESP will constantly enter Bootloader mode. Nothing has changed in my code, in the drivers I've used, I've tested this on multiple computers with the same results. I'm using the Adafruit ESP32 - S3 - Feather, and have only two buttons plugged in. My set up was working fine with no issues for over a week, only to start endlessly bootload looping after last Wednesday. I've spent the last few days fruitlessly trying to figure out the cause of the issue with no results. 

I now have 4 useless ESPs that do not work on any computer, and that constantly open the bootloader, with the Neo Pixel LED glowing green on plugging in. 

I can set the flash mode by holding down Boot while pressing and releasing Reset, and then flashing the ESP. Previously this would just launch my sketch after pressing reset, as the USB functionality of this set of boards is broken and doesn't allow an automatic reset, requiring a manual one. 

All of a sudden, hitting reset after flashing my device causes an infinite bootloader glitch that stops the ESP from showing up as a com port, and never actually launches the sketch. 

I'm getting desperate. 

### Sketch

```cpp
#include <WiFi.h>

#include <Bounce2.h>

// WiFi credentials

const char* ssid = "WIFI";

const char* password = "Pass";

// TCP server details

const char* serverIP = "192.168.1.45"; // Replace with your server's IP address

const int serverPort = 56629; // Replace with your server's port

const uint16_t localPort = 1337;

const char * localHost = "192.168.1.182";

WiFiServer server(localPort);

// Buttons

const int buttonPin1 = 12;

const int buttonPin2 = 13;

Bounce button1 = Bounce();

Bounce button2 = Bounce();

// Device details

const char* deviceID = "ESP32-1";

// WiFi client

WiFiClient client;

void setup() {

  Serial.begin(115200);

  server.begin();

  // Initialize buttons

  button1.attach(buttonPin1, INPUT_PULLUP);

  button1.interval(10);

  button2.attach(buttonPin2, INPUT_PULLUP);

  button2.interval(10);

  // Connect to Wi-Fi

  WiFi.begin(ssid, password);

  Serial.print("Connecting to WiFi...");

  while (WiFi.status() != WL_CONNECTED) {

    delay(500);

    Serial.print(".");

  }

  Serial.println("Connected!");

  // Connect to TCP server

  Serial.print("Connecting to server...");

  while (!client.connect(serverIP, serverPort)) {

    delay(500);

    Serial.print(".");

  }

  Serial.println("Connected!");

  Serial.println(client.remotePort());

  Serial.println(WiFi.localIP());

}

void startTcp(const char* host, int port) {

   // Stop server

    server.begin();

    client.write("Server Working"); //client.write is how we write communication back to the listener

 

}

void loop() {

  // Check for button presses

//startTcp(localHost, localPort);

  char c = client.read();

  Serial.write(c);

  button1.update();

  if (button1.fell()) {

    sendMessage("Button 1 pressed!");

  }

  button2.update();

  if (button2.fell()) {

    sendMessage("Button 2 pressed!");

  }

  // Check for incoming data from server

  if (client.available()) {

    String message = client.readStringUntil('\n');

    Serial.println("Received from server: " + message);

    sendHandshake();

  }

}

void sendMessage(const String& message) {

    client.connect(serverIP, serverPort);

  if (client.connected()) {

    client.println(message);

    Serial.println("Sent to server: " + message);

  }

}

void sendHandshake() {

  if (client.connected()) {

    String handshake = "Device ID: " + String(deviceID) +

                       ", Port: " + String(client.remotePort()) +

                       ", IP Address: " + String(client.remoteIP());

    server.println(handshake);

    Serial.println("Sent handshake to server: " + handshake);

  }

}
```

### Debug Message

```plain
No debug message, as the code flashes fine, the sketch just never runs, and never shows anything other than the following in Serial Monitor. 

this is what appears on Serial Monitor after successful flashing. Normally, I would hit the reset button and the sketch would launch, but as of Wednesday the 21st, this just launches an infinite bootloader loop. 
"rst:0xc (SW_CPU_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13220
ho 0 tail 12 room 4
load:0x40080400,len:3028
entry 0x400805e4
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x0 (DOWNLOAD(USB/UART0))
Saved PC:0x40041a76
waiting for download"
```

### Other Steps to Reproduce

Follow the Adafruit ESP32-S3 Feather Arduino IDE Set up https://learn.adafruit.com/adafruit-esp32-s3-feather/arduino-ide-setup

install the appropriate drivers, set up the correct board on board manager, and then flash your ESP with this code and press the Reset Button. 

I've now tried this on 4 BRAND new boards with no success, and all 4 boards getting stuck in infinite bootloading. 

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

potesd, 2023-06-26:
Please oh please help my poor dumb ass.

me-no-dev, 2023-06-27:
Looks like you are entering download mode. Are you holding something else?

SuGlider, 2023-06-27:
It seems related to #6762

SuGlider, 2023-06-27:
@potesd - Are you using the `Debug TX (DB) pin` from the Feather board to get those debug messages?

DonPavlov, 2023-08-08:
I had something similar happen yesterday. Try the following: 
Follow the steps here for the manual reset using esptool.py
So basically. Install latest python and add it to path, install latest esptool.
Download the bootloader factory images.
Then next bring it into download mode, by pressing boot, then reset then release reset then release boot.
Find out which COM devices are showing up, it may only flash briefly the right com device. Note down all showing up com devices.

Now its time to try out stuff:
Open a windows Powershell and run:
    python -m esptool --port COM88 chip_id
Use the correct port or try all out.

If the command doesn't work try spamming the command again and again by pressing up arrow on the terminal and then enter. (Up arrow scrolls through your latest issued commands).
At the same time try spamming the reset button. If the timing is just right it might work.

If this finally works you are getting closer and already have your solution.
Now navigate to the Downloads folder and spam do the same but for the flashing of the factory image.

In the Powershell: 
    cd ~\Downloads
    ls 

ls shows all the files in the directory and can help you select the right file, the Factory Bootloader.
now run
    esptool.py --port COM88 write_flash 0x0 Adafruit_Feather_ESP32S3_TFT_FactoryTest.bin

The same thing now again, try running the command while timing it correctly with pressing just the reset button on the device. Or trying to stay in the "ROM bootloader mode"

Now the best thing you can do after flashing is using the reset button right after to see your program start up.
Possible mitigations for that can be found here: 
https://github.com/espressif/arduino-esp32/issues/6762

This might help:
* In the Arduino IDE, select Tools > USB Mode > Hardware CDC and JTAG (pictured above).
* With arduino-cli, use --build-property=build.usb_mode=1 on the command line (untested).
* With platformio, use build_flags = -DARDUINO_USB_MODE=1 in platformio.ini.

SuGlider, 2023-08-17:
@potesd - I just got it to happen... I configured the Arduino IDE to enable PSRAM and then selected OPI PSRAM in a ESP32-S3 that only supports QSPI PSRAM.

Selecting back QSPI PSRAM made it work correctly.

The Serial Ouput that I get when OPI is selected looks like a infinite boot loop:
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
```

DonPavlov, 2023-08-22:
It is strange that you are using OPI PSRAM mode. Try rather using the esptool directly, as this takes the arduino abstractions away. 
Really try following my instructions above.

SuGlider, 2023-08-23:
Actually I just presented an example about how to make the S3 reset in constant Bootload loop.
I wanted to let @potesd know that selecting the correct OPI/QSPI PSRAM and Flash options is important to make the S3 work correctly.

kosmostrater, 2024-07-03:
Thx, as soon as I set PSRAM  to disable it programed.

rftafas, 2025-01-10:
No communication from OP. Assumed fixed/non-issue. Re-open if persistent with updated testing.

closed 2025-01-10 by rftafas