espressif/arduino-esp32 #8355: ESP32-S3-WROOM-2 gdbPath not configured correctly

JBelmont72, 2023-06-27:
### Board

ESP32_S3_WROOM_2

### Device Description

Error message reads"Please configure cortex-debug.armtoolchainpath or cortex-debug.gdbpath correctly"
for my ESP32-S3. Interesting that I can run the WiFi Scan example from the ESP32 library examples even with this error. But a simple blink example will produce serial print output but not light an LED.
My current set up in Arduino IDE is:[image]
[1.pdf](https://github.com/espressif/arduino-esp32/files/11876123/1.pdf)

### Hardware Configuration

for the led blink example tried GPIO 4, 7,8,15,16.

### Version

latest master (checkout manually)

### IDE Name

Arduino 2.0

### Operating System

Ventura 13.2.1 MAC OS

### Flash frequency

80 Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Error message reads"Please configure cortex-debug.armtoolchainpath or cortex-debug.gdbpath correctly"
for my ESP32-S3. Interesting that I can run the WiFi Scan example from the ESP32 library examples even with this error. But a simple blink example will produce serial print output but not light an LED.
My current set up in Arduino IDE is:[image]
[1.pdf](https://github.com/espressif/arduino-esp32/files/11876123/1.pdf)

### Sketch

```cpp
used the WIFI Scan example which works despite the error message and the basic blink with external LEDs on a variety of pins.
```

### Debug Message

```plain
Please configure cortex-debug.armtoolchainpath or cortex-debug.gdbpath correctly"
```

### Other Steps to Reproduce

I've used a variety of settings and both the USB and UART( I have the CP2102 driver). Other settings set the system into a panic or continuous reboot.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-06-27:
Blinking an LED should not depend on any of the settings in Arduino IDE.  What kind of LED is it?  A simple two pin LED typically needs a small resistor (100-220Ohm) to keep from burning out.  An RGB LED will need [different code](https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/GPIO/BlinkRGB/BlinkRGB.ino)

SuGlider, 2023-06-27:
Are you pressing the DEBUG button in the IDE GUI?

This message `"Please configure cortex-debug.armtoolchainpath or cortex-debug.gdbpath correctly"` only shows up when the Green Play button (Start Debugging) in the DEBUG session is pressed.
[image]

It is not possible to debug using the UART (CP2102).
For that, it is necessary to use the Native USB port configured as JTAG (Tools Menu-->JTG Adapter-->"Integrated USB JTAG"

JBelmont72, 2023-06-27:
I am redoing my ESP-IDF path and reloading everything to determine if I have a broken link somewhere. Working through the terminal commands as per the Espressif web site.

VojtechBartoska, 2023-08-18:
any updates @JBelmont72 ?

VojtechBartoska, 2023-08-21:
Closing, if needed you can reopen.

closed 2023-08-21 by VojtechBartoska