espressif/arduino-esp32 #8361: ESP32 Deep sleep API

savagerex, 2023-06-28:
### Board

ESP32 S3

### Device Description

Sensor -  IIS3dwb
External Flash

### Hardware Configuration

None

### Version

v2.0.8

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

I have two sample codes.

two codes use "esp_deep_sleep_start();"  API.

but two codes have different power consumption.

the problem is when codes use "esp_deep_sleep_start();" API, but power consumption is different.

below is more power consumption code.

### Sketch

```cpp
#include "SPI.h"
#include <SerialFlash.h>
#include "IIS3DWB.h"
#include "Sensor.h"
#include "WifiTask.h"
#include "BleTask.h"
#include "SensorPreferences.h"
#include <ESP32Time.h>
#include <WiFi.h>
#include <MQTT.h>
#include "LedBlinker.h"
#include "BoardConfig.h"

ESP32Time rtc;

#define EnableWifi 1
#define EnableBle 1
#define Enable_LedTask  1
[... 800 lines cut ...]

//  sensorValue = analogRead(sensorPin);

//  if(sensorValue <= 1701) {
//  Serial.println(sensorValue);
//  led_red_slow_blink();             
//  }

  client.loop();
  //client.publish(topic, "Hello world");
  //Serial.println(rtc.getDateTime());
  delay(1000);

}
```

### Debug Message

```plain
NONE
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-06-28:
There are different power domains which can be shut down during deep sleep.  The automatic method will shut down unused domains, so if you use a method that allows you turn off RTC peripherals (eg timer or ext1), you will draw less current during sleep.
https://docs.espressif.com/projects/esp-idf/en/v4.4.5/esp32/api-reference/system/sleep_modes.html#power-down-of-rtc-peripherals-and-memories

savagerex, 2023-06-28:
Got it, Thanks!!!

savagerex, 2023-06-29:
when i use timer & ext1 at same time.  power consumption is higher.

about this part, does your team can modify ??

because our product need  "timer & ext1" to wake up ESP32

SuGlider, 2023-06-29:
From the documentation: https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/system/sleep_modes.html#configuring-ios-deep-sleep-only

The circuit may also influence in the results. 

[image]

SuGlider, 2023-06-29:
@savagerex - I have done an experiment with using the Osciloscope. I have used an ESP32-S3-Mini Devkit + ESP-PROG + GPIO18 connected to 3.3v. A simplified sketch to just deep sleeep for 3 seconds and stay waked up for another 3 seconds.

I could not see any change in the current. When I comment out the line that sets GPIO18 to wakeup `esp_sleep_enable_ext0_wakeup(GPIO_NUM_18, 0)`, it also produces the exactly same current graph.

This is the OSciloscope Screenshot. Waked up = about 35mA and Deep Sleep = about 1mA.

[image]

This is the Sketch used to test it:

``` cpp
#define uS_TO_S_FACTOR 1000000  /* Conversion factor for micro seconds to seconds */
#define TIME_TO_SLEEP  3

void wakeup_reason() {
  esp_sleep_wakeup_cause_t wakeup_reason;

  wakeup_reason = esp_sleep_get_wakeup_cause();
  switch (wakeup_reason)
  {
    case ESP_SLEEP_WAKEUP_EXT0 : Serial.println("Wakeup caused by external signal using RTC_IO"); break;
    case ESP_SLEEP_WAKEUP_EXT1 : Serial.println("Wakeup caused by external signal using RTC_CNTL"); break;
    case ESP_SLEEP_WAKEUP_TIMER : Serial.println("Wakeup caused by timer"); break;
    case ESP_SLEEP_WAKEUP_TOUCHPAD : Serial.println("Wakeup caused by touchpad"); break;
    case ESP_SLEEP_WAKEUP_ULP : Serial.println("Wakeup caused by ULP program"); break;
    default :
      Serial.printf("Wakeup was not caused by deep sleep: %d\n", wakeup_reason);
      break;
  }
}

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);

  wakeup_reason();

  esp_sleep_enable_timer_wakeup(TIME_TO_SLEEP * uS_TO_S_FACTOR);
  esp_sleep_enable_ext0_wakeup(GPIO_NUM_18, 0);   //SR Board :0, EVB2 Board :1
  Serial.printf("\n\r=======\n\rhold %d sec ON\n\r=============\n\r", TIME_TO_SLEEP);
  delay(TIME_TO_SLEEP * 1000);
  Serial.printf("=====>> Going to sleep now : %d Sec Timeout\n\r", TIME_TO_SLEEP);

  esp_deep_sleep_start();

}

void loop() {}

```

SuGlider, 2023-06-29:
@savagerex  - Please try it executing `rtc_gpio_isolate(GPIO_NUM_18);`  in `setup()` before going into deep sleep.

savagerex, 2023-06-30:
I add rtc_gpio_isolate(GPIO_NUM_18);
the power consumption is 1mA.
we hope the power consumption is 6~7uA.

SuGlider, 2023-06-30:
Espressif technical customer support team from China has run a test with deep_sleep example in esp-idf to only module. hardware ESP32-S3-MINI-1-N8 and esp-idf version esp-idf v4.4.4 

Three Test cases: 
1) timer and ext0 wakeup source 
2) only timer 
3) only ext0 

The current in all three test cases is 8 μA.

The sketch posted to this issue uses the same IDF calls that we used in our testing.

We believe that there is something  in the current circuit design that causes the different current consumption.

SuGlider, 2023-06-30:
I also run an Arduino example with the 3 cases (both Timer + EXT0, just Timer and just EXT0). I used the S3 mini Devkit, therefore its circuit consumed more current than just the module.

The current consumption was exactly the same in all three cases.

I also can't see.any difference.
Therefore, we can't reproduce this issue.

It shall be something related to the circuit design and functionality.

VojtechBartoska, 2023-08-18:
Closing as answered.

closed 2023-08-18 by VojtechBartoska