espressif/arduino-esp32 #8374: Problem with PWM on GPIO5

brightproject, 2023-07-01:
### Board

ESP32-D0WD-V3 (revision 3)

### Device Description

Dev board ESP32-WROOM

### Hardware Configuration

```
// ESP32 pins used
#define TFT_CS   33  
#define TFT_DC   15  
#define TFT_RST  32 

#define TFT_WR    4 
#define TFT_RD    2

#define TFT_D0   12 
#define TFT_D1   13 
#define TFT_D2   26
#define TFT_D3   25
#define TFT_D4   17
#define TFT_D5   16
#define TFT_D6   27
#define TFT_D7   14
#define TFT_BL   21
// #define TFT_BL   5

#define BUTTON_A_PIN 39  
#define BUTTON_B_PIN 36  
```

### Version

v2.0.9

### IDE Name

Notepad ++ and arduino-cli

### Operating System

Windows 10 Pro

### Flash frequency

40 Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

I use a debug board to connect the display, using an 8-bit parallel interface, I connect two buttons and adjust the backlight at the display using PWM.
When I set TFT_BL  like GPIO21, the backlight is adjusted smoothly, my code first “lights” it smoothly, by means of a cycle for, and then after a short delay it reduces the backlight, also in the for cycle.
When I specify the backlight pin GPIO5, then the backlight simply turns on to the maximum and then just turns off without smooth.
At the beginning, I thought that GPIO5 was not designed to work with PWM, but the pinout picture says that it should work fine in PWM mode.
[image]

What could be the problem?

### Sketch

```cpp
//This is certainly not all the code and not all the functions, just a small excerpt from the code.

#define BLK_PWM_CHANNEL 7  // LEDC_CHANNEL_7	
#define TFT_BL   21
// #define TFT_BL   5	

// Init the back-light LED PWM
ledcSetup(BLK_PWM_CHANNEL, 44100, 8);
ledcAttachPin(TFT_BL, BLK_PWM_CHANNEL);
ledcWrite(BLK_PWM_CHANNEL, 80);

  for (float i = 1.0f; i <= Brightness; i += (float)Brightness / 500.0f)
  {
    Backlight(i);
    delay (3);
  }

  delay (150);
  for (float i = 1.0f; i <= Brightness; i += (float)Brightness / 500.0f)
  {
    Backlight(Brightness - i);
    delay (3);
  }
```

### Debug Message

```plain
No messages
```

### Other Steps to Reproduce

Nope

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brightproject, 2023-07-03:
Let me fix the problem a bit.
Pin 5 actually works as PWM and adjusts the backlight smoothly.
The point is the constant reloading of the module, when loading the next sketch.
```
#include <TFT_eSPI.h>

TFT_eSPI tft = TFT_eSPI();// create an object to work with the screen
TFT_eSprite sprite = TFT_eSprite(&tft); // create an object to work with the buffer

const int ledPin = 1; // LED buildin pin number
const int LED_PIN = 5; // pin number on which the display backlight is connected
const int LED_CHANNEL = 7; // PWM channel number for LED control
const int FREQ = 1000; // PWM frequency in Hz

const int BUTTON_1_PIN = 36; // botton DN
const int BUTTON_2_PIN = 39; // buttom UP

// rectangle where text is displayed
const int RECT_X = (360 / 2) - 75  ; 
const int RECT_Y = (240 / 2) - 30; 
const int RECT_WIDTH = 100; 
const int RECT_HEIGHT = 60; //

int brightness = 0; // initial LED brightness
int delayTime = map(brightness, 0, 255, 1000, 100); // calculation of delay time depending on brightness
unsigned long previousMillis = 0; // variable to store the previous millisecond value

void setup() {
  // PWM channel initialization
  ledcSetup(LED_CHANNEL, FREQ, 8); 
  // binding a PWM channel to a pin
  ledcAttachPin(LED_PIN, LED_CHANNEL);

  pinMode(BUTTON_1_PIN, INPUT);
  pinMode(BUTTON_2_PIN, INPUT);
  pinMode(ledPin, OUTPUT); 
  
  // screen initialization
  tft.init();
  tft.setRotation(1); // screen rotation 90 degrees
  tft.fillScreen(TFT_BLACK); // screen cleaning
  tft.setTextSize(3); // text size 3
  tft.setTextColor(TFT_WHITE); 

  // creating a sprite to work with the buffer
  sprite.setColorDepth(8); 
  sprite.createSprite(RECT_WIDTH-2, RECT_HEIGHT-2); // creating a sprite with the dimensions of a rectangle
  sprite.fillSprite(TFT_BLACK); // sprite cleaning
  sprite.setTextSize(5); 
  sprite.setTextColor(TFT_WHITE);
}

void loop() {
  // decrease in brightness when pressing the first button
  if (digitalRead(BUTTON_1_PIN) == LOW) {
    brightness = max(brightness - 10, 0); 
    ledcWrite(LED_CHANNEL, brightness);
    delayTime = map(brightness, 0, 255, 1000, 100); // recalculate the delay time depending on the brightness
    delay(50); 
  }

  // increase brightness by pressing the second button
  if (digitalRead(BUTTON_2_PIN) == LOW) {
    brightness = min(brightness + 10, 255); 
    ledcWrite(LED_CHANNEL, brightness);
    delayTime = map(brightness, 0, 255, 1000, 100); 
    delay(50); 
  }

  // displaying the current brightness on the screen through a sprite
  sprite.fillRect(0, 0, sprite.width(), sprite.height(), TFT_BLACK); 
  tft.setTextSize(5); 
  tft.setCursor(0, 0); // placing the cursor in the upper left corner of the screen
  tft.println("Brightness"); 
  sprite.setCursor(5, 10); // set the cursor to the top of the sprite
  sprite.println(brightness); 
  sprite.pushSprite(RECT_X+1, RECT_Y+1); 

  unsigned long currentMillis = millis(); // current value of milliseconds

  if (currentMillis - previousMillis >= delayTime) { // if enough time has passed
    previousMillis = currentMillis; // save the current value of milliseconds
    digitalWrite(ledPin, !digitalRead(ledPin)); // invert the state of the LED
  }
  
}
```
What is the strangeness of the following, if I use 2 buttons to give LOW signals to pins 39 and 36, then the code works, the display shows an inscription with brightness numbers, the PWM signal on pin 5 changes, the LED built into the board (on pin 1) also changes the glow periods, depending on the brightness value 0-255.
But as soon as I stop pressing the buttons, after 4-5 seconds the module will reboot, with the output of the following information in the com port.
```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13220
ho 0 tail 12 room 4
load:0x40080400,len:3028
entry 0x400805e4
```
My board got a little warm, and I decided to replace the AMC3.3 converter
Soldered it on the board, but nothing has changed.
Then I powered the board, directly, from a good power source - 3.3. volt.
But the reboots also continued.
With other sketches, the board does not reboot like this.
What could be the reason?
The idle board is reloaded.

sebi-rf, 2023-09-13:
II had the same problem! At mine "brown reset" did it.  So, I disabled brownout detector in setup function:

 WRITE_PERI_REG(RTC_CNTL_BROWN_OUT_REG, 0);
  Serial.println("Brownout detector disabled");

Firstly, include library: 

#include "soc/soc.h"
#include "soc/rtc_cntl_reg.h"

brightproject, 2023-09-18:
> II had the same problem! At mine "brown reset" did it. So, I disabled brownout detector in setup function:
> 
> WRITE_PERI_REG(RTC_CNTL_BROWN_OUT_REG, 0); Serial.println("Brownout detector disabled");
> 
> Firstly, include library:
> 
> #include "soc/soc.h" #include "soc/rtc_cntl_reg.h"

Thank you for your concern about my problem.
I work in the Arduino ecosystem, I don’t use the `ESP-IDF` core explicitly.
I catch the Brown detector for various reasons, the pins are not assigned correctly, or there are other conflicts.
Esp32 is a very unique microcontroller, but it makes up for it with its power.

closed 2025-01-16 by Parsaabasi