espressif/arduino-esp32 #8376: AsyncWebServer official integration

guestisp, 2023-07-02:
### Related area

WiFi

### Hardware specification

DevKit

### Is your feature request related to a problem?

no

### Describe the solution you'd like

Any chanche to have the ESPAsyncWebServer ported officially in the main repo ? 
That project seems to be dead, but the main developer is very active here.

Why not porting it officially ?

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

lbernstone, 2023-07-02:
A rewrite of WebServer based on esp_http(s)_server would not be a bad thing to put on the milestone list.

me-no-dev, 2023-07-02:
AsyncTCP might make it to the repo, but the server will likely stay separate (though I do plan to split it from ESP8266, because ESP32 and FreeRTOS require different approach).

Project is "dead" because I am very active here and this is actually my job. Will revive after Arduino 3.0.0 is out

me-no-dev, 2023-07-02:
> A rewrite of WebServer based on esp_http(s)_server would not be a bad thing to put on the milestone list.

That is planned actually

guestisp, 2023-07-02:
> AsyncTCP might make it to the repo, but the server will likely stay separate (though I do plan to split it from ESP8266, because ESP32 and FreeRTOS require different approach).
> 
> Project is "dead" because I am very active here and this is actually my job. Will revive after Arduino 3.0.0 is out

I understand, but at least some bug-fixing.. Features should wait, but there are some bugs that are very annoying (one over all, the queue size, there is a fork that adds support for chaning the queue size, that change should be merged)

HamzaHajeir, 2023-07-02:
I'm happy to invite you to use a library I maintain after the passage of its author about a year ago :\.

Feel free to use it, it's [H4AsyncWebServer](https://github.com/HamzaHajeir/H4AsyncWebServer).

Yet for TLS support you'll need to install a customized arduino core.

You can also check a complete H4Plugins environment [here](https://github.com/HamzaHajeir/H4Plugins_Env).

guestisp, 2023-07-02:
very very very interesting
i'll give it a try. do you know most frequent changes to do as drop in replacement of espasyncwebserver?

HamzaHajeir, 2023-07-02:
> very very very interesting i'll give it a try. do you know most frequent changes to do as drop in replacement of espasyncwebserver?

Ah that would require some good changes, as it has a totally different design. 

Basically it implies the usage of [H4 Timer](https://github.com/hamzahajeir/H4), it's a scheduler with good features, so requires to change your main code from `void setup() {} ..  void loop()` to `void h4setup(){}` and `void h4UserLoop(){}`. More to know about within its docs.

Then you can check the webserver's examples, beside how [H4Plugins starts](https://github.com/HamzaHajeir/h4plugins/blob/master/src/H4P_WiFi.cpp#L315-L378) it.

For request headers you can get them by calling `req_headers()` method out of the passed handler to the callback.

It's good to know that it currently lacks authentication, also it serves the file system automatically over the root (I might introduce a user API to allow overriding).

horendus, 2023-07-17:
I would be extremely grateful for an espAsyncWebServer update that adds full HTTPS support. To here its on the roadmap is great news!

closed 2025-01-16 by Parsaabasi