espressif/arduino-esp32 #8377: ESP32-S3 Hangs on boot when UART is printing

HamzaHajeir, 2023-07-02:
### Board

Espressif ESP32-S3-DevKitC-1-N8 (8 MB QD)

### Device Description

Official board by Espressif
Programming and powering by the UART port.

### Hardware Configuration

Unconnected to any external peripherals.

### Version

v2.0.9

### IDE Name

PlatformIO

### Operating System

Windows 11

### Flash frequency

40MHz (Default one)

### PSRAM enabled

yes

### Upload speed

115200

### Description

When data rushes to UART at application startup the MCU hangs.

This is observed in printing to UART within global constructors, wherein the Serial began in an earlier one.

This issue wasn't observed with normal ESP32.

The amount of data being successfully printed depends on the introduced delay (delayMs in the example sketch).

With no delay it printed 155 Bytes, and with a delay of uS it printed 182 Bytes, finally run with not problems when delay was 100uS.

### Sketch

```cpp
#include <Arduino.h>

int delayMs = 10;
class dummy {
  public:
  dummy () {
    Serial.begin(115200);
    for(int i=0;i<200;i++) {
      Serial.printf("0", i);
      delayMicroseconds(delayMs);
    }
  }
};
dummy d;
void setup() {
  // put your setup code here, to run once:
  pinMode(LED_BUILTIN, OUTPUT);
  Serial.begin(115200);
  Serial.printf("Hello World!\n");
}

void loop() {
  // put your main code here, to run repeatedly:
  digitalWrite(LED_BUILTIN, HIGH);
  delay(1000);
  digitalWrite(LED_BUILTIN, LOW);
  delay(1000);
}
```

### Debug Message

```plain
cmd _addLocal()
_addLocal() end
[   208][D][esp32-hal-rmt.c:615] rmtInit():  -- TX RMT - CH 0 - 1 RAM Blocks - pin 48

Adding locals
wifi _addLocal()
_addLocal() end
_ESP-ROM:esp32s3-20210327

Build:Mar 27 2021

rst:0x10 (RTCWDT_RTC_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)

SPIWP:0xee

mode:DIO, clock div:1

load:0x3fce3808,len:0x44c

load:0x403c9700,len:0xbe4

load:0x403cc700,len:0x2a68

entry 0x403c98d4
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

JorikWit, 2023-07-03:
try 
Flash Mode -> OPI 80MHz

brightproject, 2023-07-03:
Now I did one "[research](https://github.com/espressif/arduino-esp32/issues/8374#issuecomment-1618760514)" about resetting the microcontroller, and found the reason, this is the installed core 0 in the arduino IDE.
Tell me, are you flashing the microcontroller through an arduino IDE?
What do you have selected in the board settings?

HamzaHajeir, 2023-07-03:
I'm not sure what that issue brings to us, I use PlatformIO.

The software does work ok after setup() is succeeded to be called.

brightproject, 2023-07-03:
> The software does work ok after setup() is succeeded to be called.

Show me your compilation settings.
Are you specifying which `core 0` or `core 1` to use?

HamzaHajeir, 2023-07-03:
I don't override any such configuration. And other than the exact situation
described OP, everything works as expected.

Rhvarrier, 2023-07-08:
I'm having the same issue. I'm using the following [compilation settings ](https://github.com/sivar2311/freenove-esp32-s3-platformio/blob/73a8694c03023057fe3b727259a86caaa350c612/freenove_esp32_s3_wroom.json)

SuGlider, 2023-08-18:
After investigating this issue, I found out a few things. 
As conclusion, I see that it is due to the way how IDF UART API, used in Arduino, works.
The IDF UART driver freezes, apparently, depending of the "timing" of its execution.
It can't be fixed at this time.

A few facts:
1- The 1st stage boot loader will set the Watchdog Timer to reset the ESP32-S3 after about 8 seconds. 
This WDT resets the S3 with the message "rst:0x10 (RTCWDT_RTC_RST)"

2- IDF UART driver needs FreeRTOS resources to be set in order to run correctly.

3- C++ constructors are executed before the `main()` function is executed. This creates a problem because the IDF Driver can't run well and the program ends up with potential "inconsistencies".

I have tested the same sketch running direct IDF UART writing functions, and it fails in the same way:
``` cpp
#include "driver/uart.h"

class dummy {
  public:
  dummy () {
    Serial.begin(115200);
    for(int i=0;i<200;i++) {
      char c = '0';
      uart_write_bytes(0, &c, 1);
      delayMicroseconds(delayMs);
    }
  }
};
``` 

As a workaround, I used `printf()` instead of `Serial.printf()` which uses some ROM functions instead of IDF UART drivers.
So, in case it is necessary to print something in a C++ Constructor, this is the way to do it.

``` cpp
class dummy {
  public:
  dummy () {
    for(int i=0;i<200;i++) {
      printf("%d", i%10);
      delayMicroseconds(delayMs);
    }
    printf("\nConstructor finished\n");
  }
};

```

HamzaHajeir, 2023-08-18:
Thanks for the investigation.

But what can make printing at constructors succeeding in the older ESP32.

If I remember correctly, there was a similar issue regarding ESP32 that was
fixed by somehow. But couldn't find it.

SuGlider, 2023-08-18:
What do you mean with "older ESP32"?
In Arduino Core 1.0.6, or any 1.0.x?
Or on the ESP32 chip with Arduino Core 2.0.x?

I tested the issue with ESP32 + Core 2.0.11 and time = 1 us.
It hangs as well.

HamzaHajeir, 2023-08-21:
If I remember it correctly, it was printing okay, then made workarounds to disable on the S3 recently.

I'll re-check at soonest anyway.

AAJAY5, 2023-09-01:
Use snippet shown below. Flush every time instead of delay. 
```
#include <Arduino.h>

int delayMs = 10;
class dummy {
  public:
  dummy () {
    Serial.begin(115200);
    while(!Serial);
    for(int i=0;i<200;i++) {
      Serial.printf("0",i);
      Serial.flush();    
    }
  }
};
dummy d;
void setup() {
  // put your setup code here, to run once:
  pinMode(LED_BUILTIN, OUTPUT);
  Serial.begin(115200);
  while(!Serial);
  Serial.printf("Hello World!\n");
}

void loop() {
  // put your main code here, to run repeatedly:
  digitalWrite(LED_BUILTIN, HIGH);
  delay(1000);
  digitalWrite(LED_BUILTIN, LOW);
  delay(1000);
}```

closed 2025-01-10 by rftafas