espressif/arduino-esp32 #8381: Mqtt Library - Arduino

savagerex, 2023-07-04:
### Board

ESP32 S3

### Device Description

Sensor - IIS3DWB
Flash

### Hardware Configuration

None

### Version

v2.0.9

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

We need Mqtt library - arduino from ESP team.

because we want to transfer large amount of data.

please help us.

Thanks

### Sketch

```cpp
none
```

### Debug Message

```plain
none
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

savagerex, 2023-07-04:
i use the mqtt lib from third party.
but their lib is be limited by ram.

So i hope esp team can make mqtt lib by psram.

savagerex, 2023-07-05:
below is the sample code

#include <WiFi.h>
#include <MQTT.h>

const char* ssid = "ASUS-38";     //SW1-WIFI6   ASUS-38   vivo Y21 
const char* pass = "12345678";   //wireless    12345678   1234567890

const char* mqtt_server = "192.168.50.100";   //broker.emqx.io  192.168.50.128
const char *topic = "DataTopic";

//uint8_t DataBufRead[18500];
uint8_t* DataBufRead;

WiFiClient net;
//MQTTClient client(1700000);  
MQTTClient client(162000);

void connect() {
  Serial.print("checking wifi...");
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(1000);
  }

  Serial.print("\nconnecting...");

  while (!client.connect("arduino", "public", "public")) {
    Serial.print(".");
    delay(1000);
  }

  Serial.println("\nconnected!");

  
}

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);

  DataBufRead = (uint8_t*)ps_calloc(162500, sizeof(uint8_t));

  delay(10);
    WiFi.begin(ssid,pass);

    Serial.print("Connecting.");
    while(WiFi.status() != WL_CONNECTED) {
        Serial.print(".");
        delay(500);
    }

    Serial.print("WiFi connected - IP address: ");
    Serial.println(WiFi.localIP());

    client.begin(mqtt_server, net);

  Serial.print("\nconnecting...");
  while (!client.connect("arduino", "public", "public")) {
    Serial.print(".");
    delay(1000);
  }

  Serial.println("\nconnected!");

  client.loop();

  if (!client.connected()) {
    connect();
  }

  for (uint32_t i = 0; i<161999; i++)
  {
    DataBufRead[i] = 'a';
  }

   char *DataBuf = (char*)DataBufRead;
   bool PubData = client.publish(topic, DataBuf, 162000, false, 0);  //18000 //6000
   Serial.println(PubData);

   
   Serial.println("Mqtt Finish");
}

void loop() {
  // put your main code here, to run repeatedly:

}

savagerex, 2023-07-05:
if declare  MQTTClient client(162000);

then publish is fail.

Serial.println(PubData);    always get 0.

1: Success
0: fail

lbernstone, 2023-07-05:
psram must be initialized (in initArduino) before it can be used.  Instantiate your object with a new call in setup(). You can see an example of how to use new in https://github.com/espressif/arduino-esp32/blob/master/libraries/SPI/examples/SPI_Multiple_Buses/SPI_Multiple_Buses.ino

HamzaHajeir, 2023-07-05:
You're free to test and try out [H4AsyncMQTT](https://github.com/hamzahajeir/H4AsyncMQTT), which under TLS testing for now.

bertmelis, 2023-07-05:
Even that lib limits to available heap. I have a lib that uses a callback so you can fetch payload data in chunks. Filling up memory with very big payloads might succeed but it's better to leave some space for for example incoming data that needs processing.

closed 2025-01-15 by Parsaabasi