espressif/arduino-esp32 #8383: Debug not supported on ESP32-S2-DevKitM-1U-N4R2

mjs513, 2023-07-04:
### Board

ESP32-S2-DevKitM-1U-N4R2

### Device Description

Espressif ESP32-S2-DevKitM-1U-N4R2 (https://www.amazon.com/dp/B0B97H9D4F?psc=1&ref=ppx_yo2ov_dt_b_product_details) on breadboard.

### Hardware Configuration

Nothing else is connected to the board except the jtag connections.

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE 2.1.1 Released Version

### Operating System

Windows 10 x64

### Flash frequency

80Mz

### PSRAM enabled

no

### Upload speed

921600

### Description

Using a ESP-Prog board with Interface 0 updated to WinUSB v6.1.7600.16385 and core debug level set to Debug with adapter set to FTDI Arduino debugger fails to start.  Expected behavior is it starts.

### Sketch

```cpp
/*
  BlinkRGB

  Demonstrates usage of onboard RGB LED on some ESP dev boards.

  Calling digitalWrite(RGB_BUILTIN, HIGH) will use hidden RGB driver.
    
  RGBLedWrite demonstrates controll of each channel:
  void neopixelWrite(uint8_t pin, uint8_t red_val, uint8_t green_val, uint8_t blue_val)

  WARNING: After using digitalWrite to drive RGB LED it will be impossible to drive the same pin
    with normal HIGH/LOW level
*/
//#define RGB_BRIGHTNESS 64 // Change white brightness (max 255)

// the setup function runs once when you press reset or power the board

void setup() {
  // No need to initialize the RGB LED
  Serial.begin(115200);
}

// the loop function runs over and over again forever
void loop() {
#ifdef RGB_BUILTIN
  digitalWrite(RGB_BUILTIN, HIGH);   // Turn the RGB LED white
  delay(1000);
  digitalWrite(RGB_BUILTIN, LOW);    // Turn the RGB LED off
  delay(1000);

  neopixelWrite(RGB_BUILTIN,RGB_BRIGHTNESS,0,0); // Red
  delay(1000);
  neopixelWrite(RGB_BUILTIN,0,RGB_BRIGHTNESS,0); // Green
  delay(1000);
  neopixelWrite(RGB_BUILTIN,0,0,RGB_BRIGHTNESS); // Blue
  delay(1000);
  neopixelWrite(RGB_BUILTIN,0,0,0); // Off / black
  delay(1000);
  Serial.println("end cycle");
#endif
}
```
```

### Debug Message

```plain
[2023-07-04T13:56:12.306Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
"C:\\Users\\Merli\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\openocd-esp32\\v0.11.0-esp32-20221026/bin/openocd" -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s "d:\\Users\\Merli\\Documents\\Arduino\\BlinkRGB_ESP32C3" -f "F:/Arduino-2-0/resources/app/plugins/cortex-debug/extension/support/openocd-helpers.tcl" -f debug.cfg
Open On-Chip Debugger v0.11.0-esp32-20221026 (2022-10-26-14:48)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
CDRTOSConfigure
Info : Listening on port 50001 for tcl connections
Info : Listening on port 50002 for telnet connections
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi tdo_sample_edge falling"
Info : clock speed 20000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0xfc800231 (mfg: 0x118 (Southland Microsystems), part: 0xc800, ver: 0xf)
Warn : JTAG tap: esp32.cpu0       UNEXPECTED: 0xfc800231 (mfg: 0x118 (Southland Microsystems), part: 0xc800, ver: 0xf)
Error: JTAG tap: esp32.cpu0  expected 1 of 1: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0xffffffff (mfg: 0x7ff (<invalid>), part: 0xffff, ver: 0xf)
Warn : JTAG tap: esp32.cpu1       UNEXPECTED: 0xffffffff (mfg: 0x7ff (<invalid>), part: 0xffff, ver: 0xf)
Error: JTAG tap: esp32.cpu1  expected 1 of 1: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Error: Trying to use configured scan chain anyway...
Error: esp32.cpu1: IR capture error; saw 0x1f not 0x01
Warn : Bypassing JTAG setup events due to errors
Error: [esp32.cpu0] Unexpected OCD_ID = 00000000
Warn : target esp32.cpu0 examination failed
Info : starting gdb server for esp32.cpu0 on 50000
Info : Listening on port 50000 for gdb connections
Info : [esp32.cpu1] Debug controller was reset.
Info : accepting 'gdb' connection on tcp/50000
Error: Target not examined yet
Error executing event gdb-attach on target esp32.cpu0:

Warn : No symbols for FreeRTOS!
Error: Target not halted
Error: auto_probe failed
Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
Error: attempted 'gdb' connection rejected
[2023-07-04T13:56:17.018Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...
```
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mjs513, 2023-07-05:
Added info.  After some playing around with drivers and changing the adapter speed to 9000 it did recognize the chip however still does not start:
```
Waiting for gdb server to start...[2023-07-05T17:47:12.626Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
"C:\\Users\\Merli\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\openocd-esp32\\v0.11.0-esp32-20221026/bin/openocd" -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s "d:\\Users\\Merli\\Documents\\Arduino\\BlinkRGB_ESP32C3" -f "F:/Arduino-2-0/resources/app/plugins/cortex-debug/extension/support/openocd-helpers.tcl" -f debug.cfg
Open On-Chip Debugger v0.11.0-esp32-20221026 (2022-10-26-14:48)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
CDRTOSConfigure
Info : Listening on port 50001 for tcl connections
Info : Listening on port 50002 for telnet connections
Info : clock speed 9000 kHz
Info : JTAG tap: esp32s2.cpu tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : starting gdb server for esp32s2 on 50000
Info : Listening on port 50000 for gdb connections
Info : [esp32s2] Target halted, PC=0x4004C18C, debug_reason=00000001
Info : [esp32s2] Reset cause (8) - (Main WDT1 core reset)
Error: [esp32s2] DSR (8022CC13) indicates DIR instruction generated an exception!
Info : [esp32s2] Target halted, PC=0x4004C18C, debug_reason=00000001
Info : [esp32s2] Reset cause (8) - (Main WDT1 core reset)
Info : [esp32s2] Halt cause (0) - (Illegal instruction)
Error: [esp32s2] DSR (80008413) indicates DIR instruction generated an exception!
Error: [esp32s2] DSR (8000CC13) indicates DIR instruction generated an exception!
Info : [esp32s2] Target halted, PC=0x4004C18C, debug_reason=00000001
Info : [esp32s2] Reset cause (8) - (Main WDT1 core reset)
Info : [esp32s2] Debug controller was reset.
Info : [esp32s2] Core was reset.
Info : [esp32s2] Debug controller was reset.
Info : [esp32s2] Core was reset.
Error: [esp32s2] DSR (8020CC13) indicates DIR instruction generated an exception!
Info : [esp32s2] Target halted, PC=0x4001276A, debug_reason=00000001
Info : [esp32s2] Reset cause (8) - (Main WDT1 core reset)
Error: [esp32s2] DSR (8022CC13) indicates DIR instruction generated an exception!
Info : [esp32s2] Target halted, PC=0x4004C18C, debug_reason=00000001
Info : [esp32s2] Disabling LDDR32.P/SDDR32.P
Info : [esp32s2] Halt cause (0) - (Illegal instruction)
Error: [esp32s2] DSR (8000CC13) indicates DIR instruction generated an exception!
Info : [esp32s2] Target halted, PC=0x4004C18C, debug_reason=00000001
Info : [esp32s2] Reset cause (8) - (Main WDT1 core reset)
Info : [esp32s2] Debug controller was reset.
Info : [esp32s2] Core was reset.
Error: [esp32s2] DSR (8020CC13) indicates DIR instruction generated an exception!
Info : [esp32s2] Target halted, PC=0x4001276A, debug_reason=00000001
Info : [esp32s2] Reset cause (8) - (Main WDT1 core reset)
Info : [esp32s2] Target halted, PC=0x4004C18C, debug_reason=00000001
Info : accepting 'gdb' connection on tcp/50000
Info : [esp32s2] Debug controller was reset.
Info : [esp32s2] Core was reset.
Info : [esp32s2] Target halted, PC=0x4001276A, debug_reason=00000000
Info : [esp32s2] Reset cause (8) - (Main WDT1 core reset)
Warn : [esp32s2] target not halted
Error: read_memory: read at 0x3f4c1010 with width=32 and count=1 failed
Error executing event gdb-attach on target esp32s2:
C:/Users/Merli/AppData/Local/Arduino15/packages/esp32/tools/openocd-esp32/v0.11.0-esp32-20221026/bin/../share/openocd/scripts/memory.tcl:87: Error: memread32: read_memory: failed to read memory
in procedure 'esp32s2_memprot_is_enabled' called at file "C:/Users/Merli/AppData/Local/Arduino15/packages/esp32/tools/openocd-esp32/v0.11.0-esp32-20221026/bin/../share/openocd/scripts/target/esp32s2.cfg", line 63
in procedure 'get_mmr_bit' called at file "C:/Users/Merli/AppData/Local/Arduino15/packages/esp32/tools/openocd-esp32/v0.11.0-esp32-20221026/bin/../share/openocd/scripts/target/esp32s2.cfg", line 36
in procedure 'memread32' called at file "C:/Users/Merli/AppData/Local/Arduino15/packages/esp32/tools/openocd-esp32/v0.11.0-esp32-20221026/bin/../share/openocd/scripts/mmr_helpers.tcl", line 79
at file "C:/Users/Merli/AppData/Local/Arduino15/packages/esp32/tools/openocd-esp32/v0.11.0-esp32-20221026/bin/../share/openocd/scripts/memory.tcl", line 87
Warn : No symbols for FreeRTOS!
Error: Target not halted
Error: auto_probe failed
Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
Error: attempted 'gdb' connection rejected
[2023-07-05T17:47:15.092Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...
```
@KurtE is having similar issues.

mjs513, 2023-07-06:
Update (6/6)
Finally got it working by changing the adapter speed in esp32s2_kaluga_v1.cfg to 5000.  But this seems to be dangerous as that config may be used for other boards.  Not sure if there is a way to change the adapter speed in debug.custom for the S2 board?

Also how would I use a Segger J-link Mini with the S2 since there does not appear to be an option to do that?

denizzzka, 2024-08-15:
Run into the same issue with device marked as: LuatOS ESP32-C3 Development Board (CORE-ESP32)

denizzzka, 2024-08-15:
Solved by changing `openocd` option from `-f target/esp32.cfg` to `-f board/esp32c3-builtin.cfg`

closed 2025-01-15 by Parsaabasi