espressif/arduino-esp32 #8387: ESP32c3 DeepSleep GPIO Wake Up : Not able to get the GPIO which trigger wake up

spilz87, 2023-07-06:
### Board

ESP32-C3

### Device Description

I use a custom board with an ESP32-C3 and a rotary encoder

### Hardware Configuration

rotary encoder connected to PIN 0, 1, 2.
 

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

To save battery, I want to put ESP in Deep Sleep and it wake up when the rotaty encoder (or Rotary Knob) is modified.
So I used esp_deep_sleep_enable_gpio_wakeup() to wake up the device according to the used pins.
the ESP wake up correctly.

But in rotation it seems it changes to fast and I'm not able to detect which pin was triggering the wake up :(

Is there a way to know which GPIO triggered the wake up ?

is there any other solution to get it ?
(i was thinking using" void RTC_IRAM_ATTR esp_wake_deep_sleep(void)" but i didn't succed)

thanks in advance for your help

regards

### Sketch

```cpp
gpio_deep_sleep_hold_dis();
esp_sleep_config_gpio_isolate();
gpio_set_direction(gpio_num_t(0), GPIO_MODE_INPUT);
gpio_set_direction(gpio_num_t(1), GPIO_MODE_INPUT);
gpio_set_direction(gpio_num_t(2), GPIO_MODE_INPUT);
esp_deep_sleep_enable_gpio_wakeup(1<<0, ESP_GPIO_WAKEUP_GPIO_LOW); 
esp_deep_sleep_enable_gpio_wakeup(1<<1, ESP_GPIO_WAKEUP_GPIO_LOW); 
esp_deep_sleep_enable_gpio_wakeup(1<<2, ESP_GPIO_WAKEUP_GPIO_LOW); 
esp_deep_sleep_start();
```

### Debug Message

```plain
None
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-07-06:
https://docs.espressif.com/projects/esp-idf/en/v4.4.5/esp32c3/api-reference/system/sleep_modes.html#_CPPv432esp_sleep_get_gpio_wakeup_statusv

spilz87, 2023-07-06:
thanks for your reply, 

I tryed it but it seems it goes too slow to catch which one is trigger first on the rotary encoder :(

I put this to try to get it : 

`#define ENCODER_A_PIN   0
#define ENCODER_B_PIN   1
#define ENCODER_S_PIN   2

uint64_t gpioWakeUp = esp_sleep_get_gpio_wakeup_status();
  
  pinMode(PROG_PIN, INPUT);
  pinMode(ENCODER_A_PIN, INPUT);
  pinMode(ENCODER_B_PIN, INPUT);
  pinMode(ENCODER_S_PIN, INPUT);

  // datasESPNOW[1] = digitalRead(ENCODER_A_PIN);
  // datasESPNOW[2] = digitalRead(ENCODER_B_PIN);
  // datasESPNOW[3] = digitalRead(ENCODER_S_PIN);

  pinMode(LED_R_PIN, OUTPUT);
  pinMode(LED_G_PIN, OUTPUT);
  pinMode(LED_B_PIN, OUTPUT);

  digitalWrite(LED_R_PIN, 0);
  digitalWrite(LED_G_PIN, 0);
  digitalWrite(LED_B_PIN, 0);

  digitalWrite(LED_R_PIN, bitRead(gpioWakeUp, ENCODER_A_PIN));
  digitalWrite(LED_G_PIN, bitRead(gpioWakeUp, ENCODER_B_PIN));
  digitalWrite(LED_B_PIN, bitRead(gpioWakeUp, ENCODER_S_PIN));`

When I "press" (ENCODER_S_PIN) Led Blue turn ON
but when I turn the rotary encoder, Red and Green Leds turn ON whatever I turn Left or Right :(

Any suggestion ?

spilz87, 2023-07-06:
If I turn slowly, it does the difference, but if I turn it normally, it miss it :(

SuGlider, 2023-07-12:
@spilz87 - I have done a testing with `esp_sleep_get_gpio_wakeup_status()`.

I found out that it returns the GPIOs that are in LOW (wakeup mask), but the this function also changes its return value, while running the sketch, if any other GPIO goes to LOW. 
The final result is that whenever all the 3 GPIOs go to LOW, the function will return the mask incluing all of them (7). 

Example of `setup()`:
``` cpp
void setup() {
  Serial.begin(115200);

  uint8_t states[10], count = 50;
  while (count) {
    for (uint8_t i = 0; i < sizeof(states); i++) {
      states[i] = esp_sleep_get_gpio_wakeup_status();
      delay(10);
    }
    for (uint8_t i = 0; i < sizeof(states); i++) {
      Serial.print(states[i]);
      Serial.print(" ");
    }
    Serial.println();
    count--;
  }
 
  //Increment boot number and print it every reboot
  ++bootCount;
  Serial.println("Boot number: " + String(bootCount));
  Serial.println();

  //Go to sleep now
  Serial.println("Going to sleep now");

  gpio_deep_sleep_hold_dis();
  esp_sleep_config_gpio_isolate();
  gpio_set_direction(gpio_num_t(0), GPIO_MODE_INPUT);
  gpio_set_direction(gpio_num_t(1), GPIO_MODE_INPUT);
  gpio_set_direction(gpio_num_t(2), GPIO_MODE_INPUT);
  esp_deep_sleep_enable_gpio_wakeup(1 << 0, ESP_GPIO_WAKEUP_GPIO_LOW);
  esp_deep_sleep_enable_gpio_wakeup(1 << 1, ESP_GPIO_WAKEUP_GPIO_LOW);
  esp_deep_sleep_enable_gpio_wakeup(1 << 2, ESP_GPIO_WAKEUP_GPIO_LOW);
  esp_deep_sleep_start();

  Serial.println("This will never be printed");
}

void loop() {}
```

As result, reading many `esp_sleep_get_gpio_wakeup_status()` values, may help to find out the direction of the Rotary Encoder.

SuGlider, 2023-07-12:
@spilz87 - Great! I guess that the issue can be closed. Feel free to open it again whenever necessary.

closed 2023-07-12 by SuGlider

Wujianlingabc, 2025-08-13:
when light_sleep wakeup can not get trigger gpio,What is the solution to my problem?