espressif/arduino-esp32 #8391: Problem stack size SET_LOOP_TASK_STACK_SIZE(12*1024);

robertorojosil, 2023-07-09:
### Board

ESP32 Wrover Module

### Device Description

ESP32 board connected to LoRa module

### Hardware Configuration

GPIO conected between ESP32 and LoRa

### Version

v1.0.6

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

40MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Hello, good morning everyone,

I was developing my program in Arduino and when running my program I got the following errors:
Guru Meditation Error: Core 1 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (loopTask)
After a search on the internet, most users agreed on the problem of the stack and the increase of its size, so I added the command: SET_LOOP_TASK_STACK_SIZE( 16*1024 );

This was done by researching here:
https://github.com/espressif/arduino-esp32/pull/6025
https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/ArduinoStackSize/ArduinoStackSize.ino
https://github.com/espressif/arduino-esp32/pull/6025/files/301f93d9ae2c6e22a7c35993559b580403e07b77
https://github.com/espressif/arduino-esp32/issues/7824

However when I added the command and compiled, I got this error: 
Compilation error: expected constructor, destructor, or type conversion before '((' token

I compile with the ESP32 Wrober module support board, and all the ESP32 libraries are added as I saw in various tutorials. I attach the code in case it might help.

If anyone knows why this error occurs and what I should change I would be very grateful.

Thanks and best regards

### Sketch

```cpp
#include <SPI.h>
#include <LoRa.h>
#include <Arduino.h>

//define the pins used by the transceiver module
#define ssl 5
#define rst 14
#define dio0 2

extern "C" {
#include "aes.h"
#include "aes_tiny.h"
#include "cbd.h"
#include "indcpa.h"
#include "kem.h"
#include "ntt.h"
#include "params.h"
#include "poly.h"
#include "polyvec.h"
#include "randombytes.h"
#include "reduce.h"
#include "sha2.h"
#include "symmetric-aes.h"
#include "symmetric.h"
#include "verify.h"
}

uint8_t pk1[PQCLEAN_KYBER51290S_CLEAN_CRYPTO_PUBLICKEYBYTES];
uint8_t sk1[PQCLEAN_KYBER51290S_CLEAN_CRYPTO_SECRETKEYBYTES];
uint8_t ct1[PQCLEAN_KYBER51290S_CLEAN_CRYPTO_CIPHERTEXTBYTES];
uint8_t ss1[PQCLEAN_KYBER51290S_CLEAN_CRYPTO_BYTES];

uint8_t pk2[PQCLEAN_KYBER51290S_CLEAN_CRYPTO_PUBLICKEYBYTES];
uint8_t sk2[PQCLEAN_KYBER51290S_CLEAN_CRYPTO_SECRETKEYBYTES];
uint8_t ct2[PQCLEAN_KYBER51290S_CLEAN_CRYPTO_CIPHERTEXTBYTES];
//uint8_t ss2[PQCLEAN_KYBER51290S_CLEAN_CRYPTO_BYTES]

int packetCount1 = 4;
int packetLength1 = 200;

int packetSize2;
int packetLength2 = 200;
int bytes_received2 = 0;

int packetSize3;
int packetCount3 = 4;
int packetLength3 = 200;

int packetSize4;
int packetCount4 = 1;
int packetLength4 = 16;

struct AES_ctx ctx;
uint8_t info[16] = "Hello World";

SET_LOOP_TASK_STACK_SIZE(12*1024);

void setup() {
  //Inicialización del Monitor Serial
  Serial.begin(115200);
  while (!Serial);
  Serial.println("LoRa Receiver");

  //Configuración pines módulo LoRa
  LoRa.setPins(ssl, rst, dio0);
  
  //Reemplaza LoRa.negin(---E-) con la frecuencia de tu localización 
  //433E6 para Asia
  //866E6 para Europe
  //915E6 para Norte America
  while (!LoRa.begin(866E6)) {
    Serial.println(".");
    delay(500);
  }

  LoRa.setSyncWord(0xF3);
  Serial.println("LoRa Initializing OK!");
  uxTaskGetStackHighWaterMark(NULL);
}

void loop() {
  //Se genera pk1 y sk1
  PQCLEAN_KYBER51290S_CLEAN_crypto_kem_keypair(pk1, sk1);

  //Impresión por pantalla de pk1
  Serial.println("Clave publica pk1");
  for (int i = 0; i < PQCLEAN_KYBER51290S_CLEAN_CRYPTO_PUBLICKEYBYTES; i++) {
    Serial.println(pk1[i]);
  }

  //Envió de pk1 mediante LoRa
  /*for (int p = 0; p < packetCount1; p++) {
    LoRa.beginPacket();
    for (int i = 0; i < packetLength1; i++) {
      LoRa.write(pk1[p*packetLength1 + i]);
    }
    LoRa.endPacket();
  }

  //Recpeción de pk2 mediante LoRa
  while (bytes_received2 < PQCLEAN_KYBER51290S_CLEAN_CRYPTO_PUBLICKEYBYTES) {
    packetSize2 = LoRa.parsePacket();
    if (packetSize2) {
      for (int i = 0; i < packetLength2; i++) {
        pk2[bytes_received2] = LoRa.read();
        bytes_received2++;
      }
    }
  }

  //Impresión por pantalla de pk2
  Serial.println("Clave publica pk2");
  for (int i = 0; i < PQCLEAN_KYBER51290S_CLEAN_CRYPTO_PUBLICKEYBYTES; i++) {
    Serial.println(pk2[i]);
  }

  delay(1000);*/

  PQCLEAN_KYBER51290S_CLEAN_crypto_kem_enc(ct1, ss1, pk1);

  Serial.println("Texto cifrado ct1");
  for (int i = 0; i < PQCLEAN_KYBER51290S_CLEAN_CRYPTO_CIPHERTEXTBYTES; i++) {
    Serial.println(ct1[i]);
  }

  /*Serial.println("Texto aleatorio ss1");
  for (int i = 0; i < PQCLEAN_KYBER51290S_CLEAN_CRYPTO_BYTES; i++) {
    Serial.println(ss1[i]);
  }*/

  /*for (int p = 0; p < packetCount3; p++) {
    LoRa.beginPacket();
    for (int i = 0; i < packetLength3; i++) {
      LoRa.write(ct1[p*packetLength3 + i]);
    }
    LoRa.endPacket();
  }*/

  //Implementación AES

  /*AES_init_ctx(&ctx, ss1);
  AES_ECB_encrypt(&ctx, info);
  //AES_CBC_encrypt_buffer(&ctx, info, 16);

  Serial.println("Info");
  for (int i = 0; i < 16; i++) {
    Serial.println(info[i]);
  }

  for (int p = 0; p < packetCount4; p++) {
    LoRa.beginPacket();
    for (int i = 0; i < packetLength4; i++) {
      LoRa.write(info[p*packetLength4 + i]);
    }
    LoRa.endPacket();
  }*/

  while (1);
}
```

### Debug Message

```plain
Compilation error: expected constructor, destructor, or type conversion before '(' token
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-07-09:
This was never supported in 1.0.6.  If you want to change the stack size in 1.0.6, your best bet is to change main.cpp.

robertorojosil, 2023-07-09:
I found the error, I take this opportunity to tell you here.

Apparently, and looking at the screenshot you gave me of the examples, it seemed that a part of the ESP32 library was missing. I uninstalled it and reinstalled it but the examples were still missing.

So I decided to uninstall everything I had about Arduino on the computer, reboot it and reinstall it again. And once everything was installed again, the function was there and everything worked correctly. 

So nothing, thank you very much to all who helped and I leave this here in case someone has the same error in the future.

mrengineer7777, 2023-07-09:
Closing as completed. Thanks @lbernstone for the assist.

closed 2023-07-09 by mrengineer7777