espressif/arduino-esp32 #8392: WiFiSecureClient connected() detection

gasparfm, 2023-07-09:
### Board

ESP32-D0WDQ6-V3

### Device Description

M5Stack Core2

### Hardware Configuration

M5Stack Core2

### Version

v1.0.6

### IDE Name

Arduino IDE

### Operating System

macOS 12.13.1

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

1500000

### Description

I'm using ArduinoWebsockets and I have a problem detecting socket disconnection. For example, when I create a WebSocketClient and suddenly the server breaks (I have a local server on my computer and kill the process), the device still detecting the connection. It keeps the connection about 1 minute, it also keeps sending packets.

I've seen ArduinoWebsockets, strongly relies on WiFiSecureClient::connected() method (I'm using SSL), but this method isn't returning the right value.

After some tests, I've done a mix of WiFiClient::connected() and WiFiSecureClient::connected() and now works right. I'm reporting an issue instead of a PR because I haven't tested it in other cases and newer versions (but I saw newer versions have the same old code for this).

```c++
uint8_t WiFiClientSecure::connected()
{
  if (_connected) {
    uint8_t dummy = 0;
    int res = read(&dummy, 0);
    if (res<0) {
      switch (errno) {
        case EWOULDBLOCK:
        case ENOENT:
          _connected = true; 
          break;
        case ENOTCONN:
        case EPIPE:
        case ECONNRESET:
        case ECONNREFUSED:
        case ECONNABORTED:
          _connected = false; 
          log_d("Disconnected. RES: %d, ERRNO: %d\n", res, errno);
          break;
        default: 
          log_d("Unexpected. RES: %d, ERRNO: %d\n", res, errno);
          break;
      }
    }
  }
  return _connected;
}
```

### Sketch

```cpp
https://github.com/Links2004/arduinoWebSockets/blob/master/examples/esp32/WebSocketClientSSL/WebSocketClientSSL.ino
++
/*
 * WebSocketClientSSL.ino
 *
 *  Created on: 10.12.2015
 *
 *  note SSL is only possible with the ESP8266
 *
 */

#include <Arduino.h>

#include <WiFi.h>
#include <WiFiMulti.h>
#include <WiFiClientSecure.h>

#include <WebSocketsClient.h>

WiFiMulti WiFiMulti;
WebSocketsClient webSocket;

#define USE_SERIAL Serial1

void hexdump(const void *mem, uint32_t len, uint8_t cols = 16) {
	const uint8_t* src = (const uint8_t*) mem;
	USE_SERIAL.printf("\n[HEXDUMP] Address: 0x%08X len: 0x%X (%d)", (ptrdiff_t)src, len, len);
	for(uint32_t i = 0; i < len; i++) {
		if(i % cols == 0) {
			USE_SERIAL.printf("\n[0x%08X] 0x%08X: ", (ptrdiff_t)src, i);
		}
		USE_SERIAL.printf("%02X ", *src);
		src++;
	}
	USE_SERIAL.printf("\n");
}

void webSocketEvent(WStype_t type, uint8_t * payload, size_t length) {

    switch(type) {
        case WStype_DISCONNECTED:
            USE_SERIAL.printf("[WSc] Disconnected!\n");
            break;
        case WStype_CONNECTED:
            {
                USE_SERIAL.printf("[WSc] Connected to url: %s\n",  payload);

			    // send message to server when Connected
				webSocket.sendTXT("Connected");
            }
            break;
        case WStype_TEXT:
            USE_SERIAL.printf("[WSc] get text: %s\n", payload);

			// send message to server
			// webSocket.sendTXT("message here");
            break;
        case WStype_BIN:
            USE_SERIAL.printf("[WSc] get binary length: %u\n", length);
            hexdump(payload, length);

            // send data to server
            // webSocket.sendBIN(payload, length);
            break;
		case WStype_ERROR:			
		case WStype_FRAGMENT_TEXT_START:
		case WStype_FRAGMENT_BIN_START:
		case WStype_FRAGMENT:
		case WStype_FRAGMENT_FIN:
			break;
    }

}

void setup() {
    // USE_SERIAL.begin(921600);
    USE_SERIAL.begin(115200);

    //Serial.setDebugOutput(true);
    USE_SERIAL.setDebugOutput(true);

    USE_SERIAL.println();
    USE_SERIAL.println();
    USE_SERIAL.println();

      for(uint8_t t = 4; t > 0; t--) {
          USE_SERIAL.printf("[SETUP] BOOT WAIT %d...\n", t);
          USE_SERIAL.flush();
          delay(1000);
      }

    WiFiMulti.addAP("SSID", "passpasspass");

    //WiFi.disconnect();
    while(WiFiMulti.run() != WL_CONNECTED) {
        delay(100);
    }

    webSocket.beginSSL("192.168.0.123", 81);
    webSocket.onEvent(webSocketEvent);

}

void loop() {
    webSocket.loop();
}
```

### Debug Message

```plain
No output when disconnect occurs.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2023-07-09:
Hi thanks for contributing to the community. Are you really using 1.0.6? That has been out of support for some time.

SuGlider, 2023-07-18:
@gasparfm - Not sure if your project uses Arduino Core 1.0.6, but we are not supporting it anymore.
My suggestion would be to try it with the latest Arduino Core 2.0.10.
LwIP has changed a lot and there are many bug fixes and new/replaced APIs.

VojtechBartoska, 2023-10-02:
closing as expired, if needed you can reopen.

closed 2023-10-02 by VojtechBartoska