espressif/arduino-esp32 #8393: BasicOTA isn't able to update: `[ERROR]: No response from device`

beta-tester, 2023-07-09:
### Board

ESP32 Dev Module (NodeMCU ESP32 from JOY-iT)

### Device Description

hi,
i am using Arduino IDE 1.8.19 for Linux ARM 32 on my Raspberry Pi (32bit).
and i try to get the `BasicOTA` example on my `ESP32 Dev Module` (NodeMCU from JOY-iT) to run.

### Hardware Configuration

nothing attached.

### Version

v2.0.9

### IDE Name

Arduino IDE 1.8.19 for Linux ARM 32bit

### Operating System

RaspiOS Bullseye 32bit desktop

### Flash frequency

80

### PSRAM enabled

no

### Upload speed

921600

### Description

when i try to OTA update the sketch from Arduino IDE, it fails with `[ERROR]: No response from device`.

i tried the different partition schemes:
- Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS)
- Minimal SPIFFS (1.9MB APP with OTA/190KB SPIFFS)

but always the same result when i try to OTA update the device.

### Sketch

```cpp
#include <WiFi.h>
#include <ESPmDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>

const char* ssid       = "myssid";
const char* password   = "mypassword";

void setup() {
  Serial.begin(115200);
  Serial.println("Booting");
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while (WiFi.waitForConnectResult() != WL_CONNECTED) {
    Serial.println("Connection Failed! Rebooting...");
    delay(5000);
    ESP.restart();
  }

  //ArduinoOTA.setHostname("myesp32");

  //ArduinoOTA.setPassword("admin");

  ArduinoOTA
    .onStart([]() {
      String type;
      if (ArduinoOTA.getCommand() == U_FLASH)
        type = "sketch";
      else // U_SPIFFS
        type = "filesystem";

      // NOTE: if updating SPIFFS this would be the place to unmount SPIFFS using SPIFFS.end()
      Serial.println("Start updating " + type);
    })
    .onEnd([]() {
      Serial.println("\nEnd");
    })
    .onProgress([](unsigned int progress, unsigned int total) {
      Serial.printf("Progress: %u%%\r", (progress / (total / 100)));
    })
    .onError([](ota_error_t error) {
      Serial.printf("Error[%u]: ", error);
      if (error == OTA_AUTH_ERROR) Serial.println("Auth Failed");
      else if (error == OTA_BEGIN_ERROR) Serial.println("Begin Failed");
      else if (error == OTA_CONNECT_ERROR) Serial.println("Connect Failed");
      else if (error == OTA_RECEIVE_ERROR) Serial.println("Receive Failed");
      else if (error == OTA_END_ERROR) Serial.println("End Failed");
    });

  ArduinoOTA.begin();

  Serial.println("Ready");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
}

void loop() {
  ArduinoOTA.handle();
}
```

### Debug Message

```plain
on the ESP32 Dev Board i see the following output when DEBUG is enabled (verbose) :

19:22:56.656 -> [   264][I][ArduinoOTA.cpp:141] begin(): OTA server at: myesp32.local:3232
19:22:56.656 -> Ready
19:22:56.656 -> IP address: 192.168.1.4
19:24:33.634 -> [ 97256][D][Updater.cpp:133] begin(): OTA Partition: app1
19:24:33.634 -> Start updating sketch
19:24:33.667 -> Progress: 0%
[100257][I][WiFiClient.cpp:253] connect(): select returned due to timeout 3000 ms for fd 49
19:24:36.647 -> Error[2]: Connect Failed
19:24:36.647 -> [100258][E][Updater.cpp:284] end(): premature end: res:0, pos:0/759152
19:24:36.647 -> 
19:24:36.647 -> Error[4]: End Failed
19:24:36.680 -> [100283][E][ArduinoOTA.cpp:356] _runUpdate(): Update ERROR: Aborted
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-07-10:
Maybe a firewall/netfilter running on your rpi?

beta-tester, 2023-07-10:
> Maybe a firewall/netfilter running on your rpi?

it is completely off / no rules enabled:
```iftables
$ sudo iptables -L -v -n
Chain INPUT (policy ACCEPT 0 packets, 0 bytes)
 pkts bytes target     prot opt in     out     source               destination         

Chain FORWARD (policy ACCEPT 0 packets, 0 bytes)
 pkts bytes target     prot opt in     out     source               destination         

Chain OUTPUT (policy ACCEPT 0 packets, 0 bytes)
 pkts bytes target     prot opt in     out     source               destination
```

i think if there were firewall rule issues, then Arduino IDE would not see the OTA device as OTA able device and would not show it in the list of active OTA devices.

maybe i should mention, that the Raspberry Pi is acting as a WiFi access point for the OTA devices - using `hostapd` and `dnsmasq`, so the Ethernet (LAN) and WiFi (WLAN) has different IP address ranges.

c---, 2023-10-01:
I'm seeing this issue running Arduino IDE 1.8.19 on Arch Linux.

The OTAWebUpdater does work but BasicOTA does not. It never opens the update port so gets a connection refused error.

I asked Bard to run it also and supposedly it sees the same problem. Of course with Bard who knows if it would work anyway. Just another anecdotal data point.

danieljharris, 2024-03-25:
Having the same issue with Arduino IDE 2.3.2 on Windows 11 with ESP32-WROOM-32.

### Debug Message
```
[ 22307][D][Updater.cpp:133] begin(): OTA Partition: app1
Start updating sketch
[ 25308][I][WiFiClient.cpp:260] connect(): select returned due to timeout 3000 ms for fd 49
Error[2]: Connect Failed
[ 25309][E][Updater.cpp:284] end(): premature end: res:0, pos:0/798448

Error[4]: End Failed
[ 25334][E][ArduinoOTA.cpp:356] _runUpdate(): Update ERROR: Aborted
```

closed 2025-01-15 by Parsaabasi