espressif/arduino-esp32 #8396: Can't compile with BluetoothSerial.h

antekgort200, 2023-07-10:
### Board

m5stampC3U

### Device Description

plain board

### Hardware Configuration

nothing attached

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

linux mint

### Flash frequency

160MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

I use one of examples from IDE but when i compile i get
```

bt_classic_device_discovery:8:2: error: #error Serial Bluetooth not available or not enabled. It is only available for the ESP32 chip.
 #error Serial Bluetooth not available or not enabled. It is only available for the ESP32 chip.
  ^~~~~
exit status 1
#error Serial Bluetooth not available or not enabled. It is only available for the ESP32 chip.
```

### Sketch

```cpp
#include "BluetoothSerial.h"

//#define USE_PIN // Uncomment this to use PIN during pairing. The pin is specified on the line below
const char *pin = "1234"; // Change this to more secure PIN.

String device_name = "ESP32-BT-Slave";

#if !defined(CONFIG_BT_ENABLED) || !defined(CONFIG_BLUEDROID_ENABLED)
#error Bluetooth is not enabled! Please run `make menuconfig` to and enable it
#endif

#if !defined(CONFIG_BT_SPP_ENABLED)
#error Serial Bluetooth not available or not enabled. It is only available for the ESP32 chip.
#endif

BluetoothSerial SerialBT;

void setup() {
  Serial.begin(115200);
  SerialBT.begin(device_name); //Bluetooth device name
  Serial.printf("The device with name \"%s\" is started.\nNow you can pair it with Bluetooth!\n", device_name.c_str());
  //Serial.printf("The device with name \"%s\" and MAC address %s is started.\nNow you can pair it with Bluetooth!\n", device_name.c_str(), SerialBT.getMacString()); // Use this after the MAC method is implemented
  #ifdef USE_PIN
    SerialBT.setPin(pin);
    Serial.println("Using PIN");
  #endif
}

void loop() {
  if (Serial.available()) {
    SerialBT.write(Serial.read());
  }
  if (SerialBT.available()) {
    Serial.write(SerialBT.read());
  }
  delay(20);
}
```

### Debug Message

```plain
bt_classic_device_discovery:8:2: error: #error Serial Bluetooth not available or not enabled. It is only available for the ESP32 chip.
 #error Serial Bluetooth not available or not enabled. It is only available for the ESP32 chip.
  ^~~~~
exit status 1
#error Serial Bluetooth not available or not enabled. It is only available for the ESP32 chip.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-07-10:
ESP32-C3 is not capable of using bluetooth classic, only BLE.

SuGlider, 2023-07-11:
@antekgort200 
```
bt_classic_device_discovery:8:2: error: #error Serial Bluetooth not available or not enabled. It is only available for the ESP32 chip.
 #error Serial Bluetooth not available or not enabled. It is only available for the ESP32 chip.
  ^~~~~
exit status 1
#error Serial Bluetooth not available or not enabled. It is only available for the ESP32 chip.
```

The error message says all necessary information. 
Serial Bluetooth based on SPP on top of RFCOMM only works with the "old" ESP32 SoC.
All other chips, C3, S3, C6, H2, etc only support BLE 5.0+ which has no RFCOMM.

closed 2023-07-11 by SuGlider