espressif/arduino-esp32 #8397: When ASyncUDP with Wireguard, packet receive causes Guru Meditation

AndyW999, 2023-07-10:
### Board

ESP32 DEVKITC

### Device Description

Just the devkit, nothing else.

### Hardware Configuration

No

### Version

other

### IDE Name

PlatformIO

### Operating System

Win10

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

When using with PlatformIO and receiving a Wireguard packet there is a Guru Meditation Error.

If WiFiUdp is used it works OK but packets cannot be processed fast enough this is why I want to use AsyncUDP .

AsyncUDP works OK without Wireguard.

### Sketch

```cpp
#include <Arduino.h>
#include "WiFi.h"
#include <WireGuard-ESP32.h>

#include "AsyncUDP.h"
const int udpPort = 7071;

// WiFi configuration --- UPDATE this configuration for your WiFi AP
char ssid[] = "";
char password[] = "";

// WireGuard configuration --- UPDATE this configuration from JSON
char private_key[] = "";  // [Interface] PrivateKey
IPAddress local_ip(10,10,10,3);            // [Interface] Address
char public_key[] = "";     // [Peer] PublicKey
char endpoint_address[] = "10.0.0.14";    // [Peer] Endpoint
int endpoint_port = 51820;              // [Peer] Endpoint
IPAddress remote_ip = {10,10,10,1};
void WiFiEvent(WiFiEvent_t event);
AsyncUDP udp;
void udpRXCallBack(AsyncUDPPacket packet);

static WireGuard wg;

void setup()
{
    Serial.begin(115200);
    Serial.println("Connecting to the AP...");
    WiFi.onEvent(WiFiEvent);
    WiFi.begin(ssid);
    while( !WiFi.isConnected() ) {
        delay(1000);
    }
    Serial.println("Connected. Initializing WireGuard...");
    Serial.println(local_ip); 
    Serial.println(private_key); 
    Serial.println(endpoint_address); 
    Serial.println(public_key); 
    Serial.println(endpoint_port, DEC); 
    wg.begin(
        local_ip,
        private_key,
        endpoint_address,
        public_key,
        endpoint_port);
    
    udp.onPacket([](AsyncUDPPacket packet) { udpRXCallBack(packet); });
    udp.listen(udpPort);
}

void loop()
{

}

void WiFiEvent(WiFiEvent_t event)
{
  switch(event) 
  {
    case SYSTEM_EVENT_STA_CONNECTED:
    {
      Serial.println("ESP32 Connected to WiFi Network");
      break;
    }
    case SYSTEM_EVENT_STA_GOT_IP:
    {
      //When connected set 
        Serial.println("WiFi connected!");
        Serial.println(WiFi.localIP());  
        Serial.println(WiFi.subnetMask());  
        Serial.println(WiFi.gatewayIP());  
        Serial.println(WiFi.dnsIP(0));  
        Serial.println(WiFi.dnsIP(1)); 
        break;
    }
      case SYSTEM_EVENT_STA_DISCONNECTED:
      {
          Serial.println("WiFi lost connection");
          break;
      } 
    }
}

void udpRXCallBack(AsyncUDPPacket packet)
{
  uint64_t uptime_msec = millis();
    uint8_t buffer[8];
    buffer[ 0] = (uptime_msec >>  0) & 0xff;
    buffer[ 1] = (uptime_msec >>  8) & 0xff;
    buffer[ 2] = (uptime_msec >> 16) & 0xff;
    buffer[ 3] = (uptime_msec >> 24) & 0xff;
    buffer[ 4] = (uptime_msec >> 32) & 0xff;
    buffer[ 5] = (uptime_msec >> 40) & 0xff;
    buffer[ 6] = (uptime_msec >> 48) & 0xff;
    buffer[ 7] = (uptime_msec >> 56) & 0xff;
    udp.writeTo(buffer, 8, remote_ip, udpPort);
}
```

### Debug Message

```plain
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400895c3  PS      : 0x00060330  A0      : 0x800e7101  A1      : 0x3ffceeb0  
A2      : 0x00000000  A3      : 0x3ffb5644  A4      : 0x00000000  A5      : 0x00000001
A6      : 0x00060f20  A7      : 0x00000001  A8      : 0x800e6f42  A9      : 0x3ffcee60  
A10     : 0x00000001  A11     : 0x3ffb51f0  A12     : 0x3ffb51a4  A13     : 0x00000000
A14     : 0x3ffb51f0  A15     : 0x00000000  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x40089611  LEND    : 0x40089621  LCOUNT  : 0xfffffff3

Backtrace: 0x400895c0:0x3ffceeb0 0x400e70fe:0x3ffceec0 0x400e6d3d:0x3ffceee0 0x400e6d90:0x3ffcef00 0x400e6dee:0x3ffcef20 
0x400d7eba:0x3ffcef40 0x400d81a3:0x3ffcef70 0x400d81db:0x3ffcf000 0x400d8210:0x3ffcf020

  #0  0x400895c0:0x3ffceeb0 in strcmp at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/strcmp.S:467
  #1  0x400e70fe:0x3ffceec0 in esp_netif_get_handle_from_ifkey at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_netif/esp_netif_objects.c:182
  #2  0x400e6d3d:0x3ffceee0 in netif_from_if at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/tcpip_adapter/tcpip_adapter_compat.c:76
  #3  0x400e6d90:0x3ffcef00 in netif_from_if at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/tcpip_adapter/tcpip_adapter_compat.c:74
  #4  0x400e6dee:0x3ffcef20 in tcpip_adapter_get_netif at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/tcpip_adapter/tcpip_adapter_compat.c:226
  #5  0x400d7eba:0x3ffcef40 in AsyncUDPPacket::AsyncUDPPacket(AsyncUDP*, pbuf*, ip_addr const*, unsigned short, netif*) at C:/Users/Andy/.platformio/packages/framework-arduinoespressif32/libraries/AsyncUDP/src/AsyncUDP.cpp:334
  #6  0x400d81a3:0x3ffcef70 in AsyncUDP::_recv(udp_pcb*, pbuf*, ip_addr const*, unsigned short, netif*) at C:/Users/Andy/.platformio/packages/framework-arduinoespressif32/libraries/AsyncUDP/src/AsyncUDP.cpp:700
  #7  0x400d81db:0x3ffcf000 in AsyncUDP::_s_recv(void*, udp_pcb*, pbuf*, ip_addr const*, unsigned short, netif*) at C:/Users/Andy/.platformio/packages/framework-arduinoespressif32/libraries/AsyncUDP/src/AsyncUDP.cpp:709
  #8  0x400d8210:0x3ffcf020 in _udp_task(void*) at C:/Users/Andy/.platformio/packages/framework-arduinoespressif32/libraries/AsyncUDP/src/AsyncUDP.cpp:137
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-07-10:
writeTo needs an interface as well.  `udp.writeTo(buffer, 8, remote_ip, udpPort, TCPIP_ADAPTER_IF_STA);

AndyW999, 2023-07-11:
Actually it does not.

If you comment out wg.begin and set IPAddress remote_ip to the sending IP instead of the wg interface it works OK with - 

udp.writeTo(buffer, 8, remote_ip, udpPort);

AndyW999, 2023-07-14:
I attached a J-Link and when it crashes I get this -

Thread 
14 "async_udp" received signal SIGTRAP, Trace/breakpoint trap.
[Switching to Thread 1073533816]
strcmp () at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/strcmp.S:470
470	/builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/strcmp.S: No such file or directory.

Any help?

AndyW999, 2023-07-31:
Well as I was so underwhelmed by the response I used this...

IPAddress remote_ip {0, 0, 0, 0};			// IP address of the local WG interface
IPAddress local_ip {0, 0, 0, 0};           	// IP address of the remote WG interface
uint8_t rx_buffer[1600];					// received data

void udp_rtp_init(uint16_t port)
{
  err_t err;
   
  /* Create a new UDP control block  */
  upcb_rtp = udp_new();
  ip_set_option(upcb_rtp, SOF_BROADCAST);

  if(upcb_rtp)
  {
	udp_rtp_port = port;
	if(upcb_rtp)
	{
		udp_rtp_port = port;
		/* Bind the upcb to the UDP_PORT port */
		IP4_ADDR( &udp_rtp_addr.u_addr.ip4,  remote_ip[0], remote_ip[1], remote_ip[2], remote_ip[3]);
		IP4_ADDR( &wg_local_addr.u_addr.ip4,  local_ip[0], local_ip[1],local_ip[2], local_ip[3]);
		err = udp_bind(upcb_rtp, &wg_local_addr, port);
 
		if(err == ERR_OK)
		{
			/* Set a receive callback for the upcb */
			udp_recv(upcb_rtp, udp_rtp_receive_callback, NULL);
		}
		else
		{
			printf("can not bind pcb");
		}
		upcb_rtp->remote_port = udp_rtp_port;
  }
  else
  {
    printf("can not create pcb");
  } 
}

void udp_rtp_receive_callback(void *arg, struct udp_pcb *upcb, struct pbuf *p, const ip_addr_t *addr, short unsigned int port)
{
  uint16_t i, j;

  j = p->len;
  memcpy(rx_buffer, p->payload, p->len);

  return;
}

void udp_rtp_tx(unsigned char *data, unsigned short int len)
{
  if(udp_ready == true)
  {
    if (upcb_rtp->remote_port != (uint16_t)0)
    {
      struct pbuf *p = pbuf_alloc(PBUF_TRANSPORT, len, PBUF_RAM);
      memcpy(p->payload, data, len);
      udp_sendto(upcb_rtp, p, &udp_rtp_addr, udp_rtp_port);
      pbuf_free(p);
    }
  }
}

closed 2023-07-31 by AndyW999