espressif/arduino-esp32 #8400: Add compile time ADC attenuation init overrides

tablatronix, 2023-07-10:
### Related area

ESP32 ASDC Attenuation

### Hardware specification

ESP32-SX

### Is your feature request related to a problem?

No way to set adc atten in platform/env

### Describe the solution you'd like

Allow Attenuation overrides for adcinit for setting during compile

https://github.com/espressif/arduino-esp32/blob/855f5261913c74e9f4f99fc0de184c847cc4db59/cores/esp32/esp32-hal-
adc.c#L88C40-L88C40

Reasoning is that without runtime and or code control you cannot change this, and also perhaps for boards that might need a specific atten set, this is just the init default, so no issue should be caused here.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

VojtechBartoska, 2023-07-12:
@P-R-O-C-H-Y can you please help with triage?

P-R-O-C-H-Y, 2023-07-12:
@tablatronix Sounds reasonable. If possible please open a PR and do the changes on [esp-idf-v5.1-libs](https://github.com/espressif/arduino-esp32/tree/esp-idf-v5.1-libs) branch. Thanks!

P-R-O-C-H-Y, 2024-01-23:
@tablatronix Hello, any news about this? Thanks

tablatronix, 2024-01-23:
oh geez, I am going to say , I had an uncommited fork and i lost it, or my cat ate it.
Really though, I totally forgot about this, I will take a look at my stashes