espressif/arduino-esp32 #8403: is it support ESP32 board for Arduino newest version 2.1.1 ?why I meet error

woodgreat, 2023-07-12:
### Board

ESP32-CAM

### Device Description

ESP32-CAM

### Hardware Configuration

no need connect or yes ,the bug is on also

### Version

v2.0.9

### IDE Name

arduino 2.1.1

### Operating System

windows10

### Flash frequency

40mhz

### PSRAM enabled

no

### Upload speed

11520

### Description

------------------------------------ code
void setup() {

}

void loop() {

}
------------------------------------ output
ϵͳ�Ҳ���ָ����·����

exit status 1

Compilation error: exit status 1
-------------------------------------
why?

### Sketch

```cpp
void setup() {

}

void loop() {

}
```

### Debug Message

```plain
ϵͳ�Ҳ���ָ����·����

exit status 1

Compilation error: exit status 1
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-07-12:
@woodgreat - I have just updated my Arduino IDE from 2.1.0 to 2.1.1, keeping the same ESP32 Arduino Core 2.0.9 that was already installed before.

I could compile and upload an empty sketch as well as some other examples.
Please make sure that you select again the ESP32 target board and its related serial port for uploading.

[image]

woodgreat, 2023-07-14:
OK,I will try more than once

woodgreat, 2023-07-14:
maybe I use Offline ESP32 library, Is that cause the problem?where is your lib came from?

lbernstone, 2023-07-14:
Stable release version is 2.0.9
https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html

wz12378, 2023-07-21:
arduino ide 2.1.1 ,Versionv2.0.9.My situation is the same as yours,how can I solve it.help

SuGlider, 2023-07-21:
@wz12378 - What is exactly the problem?

wz12378, 2023-07-23:
> - 问题到底是什么？

like this：
ϵͳ�Ҳ���ָ����·����

exit status 1

Compilation error: exit status 1

SuGlider, 2023-07-24:
If the problem is that the Console Ouput seems weird with  `ϵͳ�Ҳ���ָ����·����` Serial Console output, please make sure that it has been changed to 115200 instead of usual 9600 default baud rate.

RayC012, 2023-07-27:
did you put your file under some folders that consist of prohibited character like" $%&"

VojtechBartoska, 2023-08-21:
this is supposed to work, can we close this issue @woodgreat?

closed 2023-09-04 by woodgreat