espressif/arduino-esp32 #8406: BLE Transfer data

savagerex, 2023-07-13:
### Board

ESP32 S3

### Device Description

Sensor - IIS3DWB

Flash

### Hardware Configuration

NONE

### Version

v2.0.9

### IDE Name

Arduino

### Operating System

Windows 11

### Flash frequency

80M

### PSRAM enabled

no

### Upload speed

921600

### Description

When i use mobile APP transfer data to ESP32 S3 by BLE.

but after several times, esp32 will reset.

### Sketch

```cpp
#include <Arduino.h>
#include <BLEDevice.h>
#include <BLEServer.h>
#include <BLEUtils.h>
#include <BLE2902.h>
#include <nvs.h>
#include <nvs_flash.h>
#include "Sensor.h"
#include "BleTask.h"
#include "SensorPreferences.h"
#include "LedBlinker.h"
#include "BoardConfig.h"
#include <WiFi.h>
#include <ArduinoJson.h>

BLEServer *pServer = NULL;
static BLECharacteristic * pTxCharacteristic = NULL;
static bool deviceConnected = false;
static bool oldDeviceConnected = false;
static uint8_t txValue = 0;

#define SERVICE_UUID           "6E400001-B5A3-F393-E0A9-E50E24DCCA9E" // UART service UUID
#define CHARACTERISTIC_UUID_RX "6E400002-B5A3-F393-E0A9-E50E24DCCA9E"
[... 947 lines cut ...]
        }
        
  } else {
    Serial.println("Received invalid JSON");
  }
  
}
```

### Debug Message

```plain
Received message {"SsidPrim":"ASUS-38", "PwPrim":"12345678"} over Bluetooth

Backtrace: 0xfffffffe:0x80381fb8 |<-CORRUPTED

ELF file SHA256: ecc258b6f93a6885

Rebooting...
x⸮⸮ESP-ROM:esp32s3-20210327
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

savagerex, 2023-07-13:
Somteimes debug message

Received message {"SsidPrim":"ASUS-38", "PwPrim":"12345678"} over Bluetooth
Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (BTC_TASK) 
Core  0 register dump:
PC      : 0x40379801  PS      : 0x00060e36  A0      : 0x3fcbd130  A1      : 0x3fcbd070  
A2      : 0x3fcbd60c  A3      : 0x3fcbd372  A4      : 0x00000002  A5      : 0x3c12b968  
A6      : 0x02cc3bcc  A7      : 0x00ffffff  A8      : 0x00000001  A9      : 0x3fcbd60c  
A10     : 0x00000034  A11     : 0x3fcd4878  A12     : 0x00000068  A13     : 0xbaad5678  
A14     : 0x00060820  A15     : 0x00000001  SAR     : 0x00000017  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0x00000000  

Backtrace: 0x403797fe:0x3fcbd070 0x3fcbd12d:0x3fcbd150 |<-CORRUPTED

ELF file SHA256: 16a5d47fb756a2a4

Rebooting...

SuGlider, 2023-07-13:
@savagerex - 

Based on the message `Debug exception reason: Stack canary watchpoint triggered (BTC_TASK)`, it looks like the Bluetooth processing task has a small stack size for the processing done in `BLECharacteristicCallbacks()`. 

`BLECharacteristicCallbacks()` calls `ParseReceivedData()` which may be the place where the BLE processing task stack overflows.

My suggestion would be to just copy the BLE received data into a HEAP buffer (global array while in `BLECharacteristicCallbacks()`) and then signal to some other task (or even in loop()) to process the data outside of `BLECharacteristicCallbacks()`.

A separated task can be created with a bigger stack size. 
It is also possible to increase the stack size of the task that processes `loop()`.

This is a similar issue to https://github.com/espressif/arduino-esp32/issues/7428
`CONFIG_BT_BTC_TASK_STACK_SIZE=3072` - 3KB is the BTC task stack size. 
It is defined in the sdkconfig of Arduino.

If necessary (which I think, it may not be the case), it is possible to turn this project into an Arduino as IDF Component project and then modify the BTC STACK SIZE to whatever you may need.

I think that it is always possible to rewrite the code and tasks architecture to solve it. 
But, of course, this is your call.

savagerex, 2023-07-17:
OK!!! Thanks, it is solved.

closed 2023-07-17 by VojtechBartoska