espressif/arduino-esp32 #8410: LittleFS mount & initialization problem in esp32cam,can any one plz help

hakam007, 2023-07-14:
### Board

esp32cam

### Device Description

esp32cam ai thinker

### Hardware Configuration

esp32cam 

### Version

v2.0.0

### IDE Name

arduino ide

### Operating System

window10

### Flash frequency

40mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

esp32cam littlefs mount problem ,plz help 

Connecting to WiFi.........
 IP Address: http://192.168.100.19
 ./components/esp_littlefs/src/littlefs/lfs.c:1229:error: Corrupted dir pair at {0x0, 0x1}
 E (3385) esp_littlefs: mount failed,  (-84)
 E (3388) esp_littlefs: Failed to initialize LittleFS
 Taking a photo...
 Picture file name: /photo.jpg
 Failed to open file in writing mode

### Sketch

```cpp
/*********
  Rui Santos
  Complete instructions at https://RandomNerdTutorials.com/esp32-cam-projects-ebook/
  
  Permission is hereby granted, free of charge, to any person obtaining a copy of this software and associated documentation files.
  The above copyright notice and this permission notice shall be included in all copies or substantial portions of the Software.
*********/

#include "esp_camera.h"
#include "SPI.h"
#include "driver/rtc_io.h"
#include <ESP_Mail_Client.h>
#include <FS.h>
#include <WiFi.h>

// REPLACE WITH YOUR NETWORK CREDENTIALS
const char* ssid = "dycepm";
const char* password = "pspcl123";
[... 271 lines cut ...]

   // You need to clear sending result as the memory usage will grow up.
   smtp.sendingResult.clear();
  }
}
```

### Debug Message

```plain
esp32cam littlefs mount problem ,plz help 

Connecting to WiFi.........
 IP Address: http://192.168.100.19
 ./components/esp_littlefs/src/littlefs/lfs.c:1229:error: Corrupted dir pair at {0x0, 0x1}
 E (3385) esp_littlefs: mount failed,  (-84)
 E (3388) esp_littlefs: Failed to initialize LittleFS
 Taking a photo...
 Picture file name: /photo.jpg
 Failed to open file in writing mode
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-07-14:
@hakam007 - Have you tried to run the LittleFS example `LITTLEFS_test.ino` from the ESP32 examples?
Go to the Arduino IDE Menu:
File-->Examples-->Examples for ESP32 DevModule-->LittleFS-->Open `LITTLEFS_test`

Read the example. I guess that there is something missing in esp32Camera sketch, like initialization.

``` cpp
    if(!LittleFS.begin(FORMAT_LITTLEFS_IF_FAILED)){
        Serial.println("LittleFS Mount Failed");
        return;
    }
```

Jason2866, 2023-07-14:
Use latest Arduino Core 2.0.9.  You wrote you are using Arduino core 2.0.0.  LittleFS was not working correctly with 2.0.0

VojtechBartoska, 2023-08-18:
Closing as solved in newer versions.

closed 2023-08-18 by VojtechBartoska