espressif/arduino-esp32 #8412: Caught in constant reboots when using interrupts

Tinyu-Zhao, 2023-07-15:
### Board

ESP32-D0WDQ6-V3

### Device Description

[M5Paper COMM EDition](https://shop.m5stack.com/products/m5paper-esp32-development-kit-comm-edition)

### Hardware Configuration

[Schematic](https://m5stack.oss-cn-shenzhen.aliyuncs.com/resource/docs/schematic/Core/m5paper/M5_PAPER_SCH.pdf)

### Version

v2.0.5

### IDE Name

PlatfromIO and Arduino

### Operating System

macOS 11.7.8 or Windows

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

Constant reboot after program execution.

[image]

I suspect a conflict between Wire and IRQ, if I comment out the [interrupts();](https://github.com/Tinyu-Zhao/M5EPD/blob/7e8ad5726f5684064879b0fa1e9a867373b1687d/src/utility/GT911.cpp#L32) or [Wire part of GT911::begin](https://github.com/Tinyu-Zhao/M5EPD/blob/7e8ad5726f5684064879b0fa1e9a867373b1687d/src/utility/GT911.cpp#L39C3-L53C6) in the program, the program won't get stuck in a reboot.

`volatile uint8_t gt911_irq_trigger = 0;

void IRAM_ATTR ___GT911IRQ___() {
    noInterrupts();
    gt911_irq_trigger = 1;
    interrupts();
}

esp_err_t GT911::begin(uint8_t pin_sda, uint8_t pin_scl, uint8_t pin_int) {
    log_d("GT911: I2C Initialization");
    pinMode(pin_int, INPUT);  // Startup sequence PIN part

    Wire.begin((int)pin_sda, (int)pin_scl,
               (uint32_t)100000U);  // Note: SHT3x sensor built into M5Paper
                                    // only seems to work with default 100kHz
    delay(100);

    Wire.beginTransmission(0x14);
    if (Wire.endTransmission()) {
        Wire.beginTransmission(0x5D);
        if (Wire.endTransmission()) {
            // log_e("Touch screen IIC connection error");
            return ESP_FAIL;
        }
        _iic_addr = 0x5D;
        // log_e("Wire.beginTransmission(0x5D)");
    }
    // log_d("Wire.endTransmission()=%d", Wire.endTransmission(true));

    delay(100);
    attachInterrupt(pin_int, ___GT911IRQ___, FALLING);
    return ESP_OK;
}`

### Sketch

```cpp
https://github.com/Tinyu-Zhao/M5EPD/blob/main/%20main.cpp
```

### Debug Message

```plain
<img width="949" alt="image" src="https://github.com/espressif/arduino-esp32/assets/44396641/8834eaf4-c56d-4f95-8848-f38abd059e5f">

Guru Meditation Error or Backtrace:
PC: 0x40084373: esp_spiram_writeback_cache at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_hw_support/port/esp32/spiram.c line 257
EXCVADDR: 0x3ffbe11c

Decoding stack results
0x40084370: esp_spiram_writeback_cache at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_hw_support/port/esp32/spiram.c line 257
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-07-17:
@Tinyu-Zhao - Please try using the latest Arduino Core 2.0.10 and let me know if the issue continues.

Tinyu-Zhao, 2023-07-17:
Using the latest 2.0.10 core, the problem still exists.
[image]
`Debug Message:
ELF file SHA256: f82a5b93590fc88a

Rebooting...

assert failed: spinlock_acquire spinlock.h:122 (result == core_id || result == SPINLOCK_FREE)

Backtrace: 0x40086389:0x3ffbc4c0 0x400d9e35:0x3ffbc4d0 0x40083900:0x3ffbc520 0x40084890:0x3ffbc540 0x4008386d:0x3ffbc600 0x400884b5:0x3ffbc620 0x4008d559:0x3ffbc640 0x4008b213:0x3ffbc770 0x400d939c:0x3ffbc7b0 0x400d93eb:0x3ffbc7e0 0x400d97da:0x3ffbc800 0x400896b1:0x3ffbc820
`

`
ESP Exception Decoding stack results
0x400d9e35: panic_handler at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/port/panic_handler.c line 124
0x40083900: xt_unhandled_exception at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/port/panic_handler.c line 219
0x4008386d: panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c line 408
0x400884b5: esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c line 137
0x4008d559: __assert_func at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/assert.c line 85
0x4008b213: xPortEnterCriticalTimeout at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_hw_support/include/soc/spinlock.h line 122
0x400d939c: esp_task_wdt_reset at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/port/xtensa/include/freertos/portmacro.h line 578
0x400d93eb: idle_hook_cb at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/task_wdt.c line 80
0x400d97da: esp_vApplicationIdleHook at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/freertos_hooks.c line 51
0x400896b1: prvIdleTask at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/tasks.c line 3987

`

SuGlider, 2023-07-18:
@Tinyu-Zhao 

Please try this code:

``` cpp
void IRAM_ATTR GT911IRQ() {
  // removed interrupts()/noInterrupts() from the GPIO ISR
  gt911_irq_trigger = 1;
}
```

Is there a reason for using `interrupts()/noInterrupts()`?

Tinyu-Zhao, 2023-07-21:
After testing, removing "interrupts()/noInterrupts()" doesn't affect the functionality and stops the constant rebooting.

This is the code left by a former colleague many years ago, before esp 2.0.5 this problem did not occur, do you know what causes this problem?

Is it because of this?
[image]

SuGlider, 2023-07-21:
ESP32 Arduino runs on top of FreeRTOS.
It doesn't make much sense to disable interrupts when running a RTOS.
The software shall use FreeRTOS resources to achieve results.

SuGlider, 2023-07-21:
>Is it because of this?

Yes, I think that before 2.0.5 `sei()` and `cli()` had no code.

Tinyu-Zhao, 2023-07-21:
Thank you for your answer and help. Have fun.☺️

closed 2023-07-21 by Tinyu-Zhao