espressif/arduino-esp32 #8413: pCharacteristic->notify() memory leak

raphael-bmec-co, 2023-07-15:
### Board

Esp32devkit

### Device Description

Plane module

### Hardware Configuration

NA

### Version

v2.0.9

### IDE Name

CLion with platformIO

### Operating System

Windows 11

### Flash frequency

80

### PSRAM enabled

yes

### Upload speed

115200

### Description

When calling notify @ ~10x/second with an MTU of 50-247, some backlog occurs, causing a memory leak and eventually the server becomes unresponsive to the client and logs `esp_ble_gatts_send_ notify: rc=-1 Unknown ESP_ERR error`.

The issue can be resolved by changed from 'notify' to 'indicate'. However, this is less performant and there is a clear leak if notifies fail to be received in full by a client.

### Sketch

```cpp
The BLE notify example sketch works fine. Just change the MTU - needs to be done client side to 247.
```

### Debug Message

```plain
esp_ble_gatts_send_ notify: rc=-1 Unknown ESP_ERR error
```

### Other Steps to Reproduce

Issues exists from 2.0.3 to 2.0.9. I have found from scattered mentions around the internet from years ago about a buffer getting full. I suspect the memory is still allocated as the read is still in progress when the next notify comes along and the memory is not freed correctly clogging up the buffer.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

raphael-bmec-co, 2023-07-17:
Seems related to, or the same as, https://github.com/nkolban/esp32-snippets/issues/773.

SuGlider, 2023-07-17:
@raphael-bmec-co - Could you please post a basic BLE sketch example that replicates the issue (including an explanation about the necessary condition for the issue to happen)?

tichise, 2023-09-23:
I too am having trouble finding a solution to this problem.

closed 2025-01-09 by Parsaabasi

Nawor3565, 2025-10-18:
@Parsaabasi Can confirm that the issue is still present in the latest arduino-esp32 version.