espressif/arduino-esp32 #8416: interrupt problem

savagerex, 2023-07-17:
### Board

ESP32 S3

### Device Description

Sensor - IIS3DWB
Flash

### Hardware Configuration

NONE

### Version

v2.0.9

### IDE Name

Arduino

### Operating System

Windows 11

### Flash frequency

80M

### PSRAM enabled

no

### Upload speed

921600

### Description

when i use interrupt.

Sometimes the value "Isrk" will overflow.

the code as below.

void IRAM_ATTR IntOneIsr() {
     //DryFlag = 1;    
     //DataReady = true; 
       IsrK++;
       if(IsrK==6)
       {
         IsrK=0;
         DataReady = true;
       } 
     
}

### Sketch

```cpp
#include "SPI.h"
#include "SPIFFS.h"
#include <SerialFlash.h>
#include "BoardConfig.h"
#include "Sensor.h"
#include "IIS3DWB.h"
#include <WiFi.h>
#include <MQTT.h>

const char* ssid = "ASUS_24";     //SW1-WIFI6   ASUS-38   oppo          ASUS_24
[... 346 lines cut ...]
//            packet_counter = 0;
//          }
//          packet_counter++;
//          j = 0;
//
//       }
//         
//
//
//  // Next data in 100 milliseconds
//  //delay(1000);
//  //delayMicroseconds(211);   //250
//
//}
```

### Debug Message

```plain
NONE
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-07-17:
@savagerex - I'm sorry, but the issue is not clear. Please give us more details and a longer description.

In a first glance, I think that  `lsrK` shall be declared `volatile` because it is only modified inside a ISR. Marking `lsrK` as volatile, it will prevent the compiler to optimize it.

SuGlider, 2023-07-17:
Another thing would be to make sure that `lsrK` is correctly initialized.
`uint32_t IsrK = 0;`

savagerex, 2023-07-18:
ok!!  thanks.

VojtechBartoska, 2023-08-18:
Closing this as asnwered.

closed 2023-08-18 by VojtechBartoska