espressif/arduino-esp32 #8420: A fatal error occurred: Failed to connect to ESP32: No serial data received.

Mosazghi, 2023-07-17:
### Board

ESP32 DOIT

### Device Description

DEVKIT 1 

### Hardware Configuration

no

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

W11

### Flash frequency

80

### PSRAM enabled

yes

### Upload speed

115200

### Description

A fatal error occurred: Failed to connect to ESP32: No serial data received.

### Sketch

```cpp
BLINK EXAMPLE
```

### Debug Message

```plain
Sketch uses 373089 bytes (28%) of program storage space. Maximum is 1310720 bytes.
Global variables use 25632 bytes (7%) of dynamic memory, leaving 302048 bytes for local variables. Maximum is 327680 bytes.
"C:\Users\fest1\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.5.1/esptool.exe" --chip esp32 --port "COM6" --baud 115200  --before default_reset --after hard_reset write_flash  -z --flash_mode dio --flash_freq 40m --flash_size 4MB 0x1000 "C:\Users\fest1\AppData\Local\Temp\arduino\sketches\153CC2C6F9AD81EC5064C858CEB5DCCC/lineTrackingtest2.0.ino.bootloader.bin" 0x8000 "C:\Users\fest1\AppData\Local\Temp\arduino\sketches\153CC2C6F9AD81EC5064C858CEB5DCCC/lineTrackingtest2.0.ino.partitions.bin" 0xe000 "C:\Users\fest1\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.10/tools/partitions/boot_app0.bin" 0x10000 "C:\Users\fest1\AppData\Local\Temp\arduino\sketches\153CC2C6F9AD81EC5064C858CEB5DCCC/lineTrackingtest2.0.ino.bin" 
esptool.py v4.5.1
Serial port COM6
Connecting......................................

A fatal error occurred: Failed to connect to ESP32: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
Failed uploading: uploading error: exit status 2
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-07-18:
@Mosazghi - I have a board with exactly the same problem.
For some reason, when I pulse BOOT button while it is displaying the dots "..." it connects. If I don't press BOOT, sometimes it connects, but most of the times, it just displays the same error as you have pointed out.

It only happens with this specific board, which I think may have some problem...

Give it a try... Pulse the BOOT button when it is printing `Connecting......................................`

Mosazghi, 2023-07-18:
Tried but doesnt solve anything. Also i forgot to mention this: the board gets really hot when i plug the USB in and smells burnt. Maybe i could provide some additional info:

`// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(LED_BUILTIN, HIGH);  // turn the LED on (HIGH is the voltage level)
  delay(1000);                      // wait for a second
  digitalWrite(LED_BUILTIN, LOW);   // turn the LED off by making the voltage LOW
  delay(1000);                      // wait for a second
}
`
Upload Speed: 115200.

SuGlider, 2023-07-18:
>Tried but doesnt solve anything. Also i forgot to mention this: the board gets really hot when i plug the USB in and smells burnt.

hmmm... I think that the CP2102 chip of your board may be damaged. This UART-USB chip is the one responsable for toggling the DTR/RTS pins to start an upload or to reset the board (always commanded by the ESPTOOL within the Arduino IDE).

Solution in that case could be to put the ESP32 board into "Waiting for Upload" state manually, by keeping BOOT button pressed and then pulsing the EN/RESET button. If it doesn't work... it may require using other way to upload, such as Wireless OTA.... OR just getting a new board.

[image]

closed 2023-07-20 by Mosazghi

ahappycatdog, 2024-03-24:
so how to solve this problem

Maykordi, 2024-05-15:
ihave the same probleme can u help me : Sketch uses 1146537 bytes (36%) of program storage space. Maximum is 3145728 bytes.
Global variables use 52460 bytes (16%) of dynamic memory, leaving 275220 bytes for local variables. Maximum is 327680 bytes.
esptool.py v4.5.1
Serial port COM5
Connecting......................................

A fatal error occurred: Failed to connect to ESP32: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
Failed uploading: uploading error: exit status 2