espressif/arduino-esp32 #8421: Why is the version 2.0.10? Is that a error?

V205Arduino, 2023-07-17:
### Board

NA

### Device Description

NA

### Hardware Configuration

NA

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

macOS

### Flash frequency

NA

### PSRAM enabled

yes

### Upload speed

NA

### Description

Why is the new version 2.0.10?
Should it be 2.1.0?
Thanks.

### Sketch

```cpp
NA
```

### Debug Message

```plain
NA
```

### Other Steps to Reproduce

Not really.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

V205Arduino, 2023-07-17:
Is it intentional?

janczeresnia, 2023-07-17:
In the new version 2.0.10 there is an error when compiling any sketch.
In version 2.0.9 everything works fine.

>: -c: line 1: unexpected EOF while searching for a match `''
>: -c: line 2: syntax error: unexpected end of file
exit status 2
/usr/share/arduino/arduino-builder return 2
Compilation error for ESP32S3 Dev Module

V205Arduino, 2023-07-17:
Not sure if that is related......

V205Arduino, 2023-07-17:
You can use the old one instead.

lbernstone, 2023-07-17:
It is intentional.  This is a patch version (bugfixes).  The next minor version would be 2.1.0.  However, the next version is likely to be 3.0.0.  A change in IDF qualifies for a major version.  I'm not sure exactly what would make for a minor version, since we've never had one of those :)

V205Arduino, 2023-07-17:
Thanks.
Make sense.

SuGlider, 2023-07-18:
> In the new version 2.0.10 there is an error when compiling any sketch. In version 2.0.9 everything works fine.

Please open a separated issue related to this problem.
I have tested 2.0.10 with the S3 using a couple examples from the IDE Menu and I got no errors. But I have used Win10.

Maybe it only happens with MacOS/Linux.

closed 2023-07-18 by SuGlider