espressif/arduino-esp32 #8422: timerAttachInterrupt with object method callback

radiantBear, 2023-07-18:
### Board

ESP32-D0WD-V3

### Device Description

Just a plain module on a breadboard

### Hardware Configuration

Nothing is connected.

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

Presumably 80MHz; not relevant to issue

### PSRAM enabled

yes

### Upload speed

921600

### Description

I'm trying to use timer interrupts and alarms within a class to contain its update needs. The function to call is a method in the class. But when passing that into `timerAttachInterrupt()` I'm getting this error: **`Invalid use of non-static member function 'void Test::testFn()'`**. 

However, _I cannot simply make the method static because I need to modify other members specific to my object instance_.

To get around this issue in the past, I've passed a lambda function as a callback with `this` specified, and then used that to call the class method (`[&]() {update();}`). However, when I try that with `timerAttachInterrupt()`, I get the error: **`Cannot convert 'Test::Test()::<lambda()>' to 'void (*)()'`**.

I found a similar issue from a few years ago (#3465), and tried using <FunctionalInterrupt.h> and the example in the Ticker library as suggested. (I also tried commenting on that issue, but it's marked as closed, so I'm creating a new issue.) However, their suggested code produces the error **`Invalid use of member function 'void Test::testFn()' (did you forget the '()' ?)`**, and I'm not sure how to remedy that. Nothing I've found searching online forums has provided any explanation of what's failing in this case either.

How can timer interrupts be done within classes? I can't make the timer ISR method static as I want it to be object dependant to process the attribute values for the particular object.

**P.S.**
Interestingly, `attachInterrupt()` (located in [FunctionalInterrupt.h](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-timer.c)) does at least allow the use of lambda function callbacks, which `timerAttachInterrupt()` does not. I assume this is because that function uses the type `std::function<void(void)>` for its callback, while `timerAttachInterrupt()` uses `void (*fn)(void)`, though I don't know why this difference exists. Would it be possible to change `timerAttachInterrupt()` to use the same callback type?

### Sketch

```cpp
#include "esp_timer.h"

class Test {
    public:
        Test(void) {
            esp_timer_handle_t testTimer;
            esp_timer_create_args_t timerConfig;
            timerConfig.arg = 0;
            timerConfig.callback = reinterpret_cast<esp_timer_cb_t>(testFn);
            timerConfig.dispatch_method = ESP_TIMER_TASK;
            timerConfig.name = "Test_Timer";
            esp_timer_create(&timerConfig, &testTimer);
            esp_timer_start_periodic(testTimer, 1000000);
        }

        void testFn(void) {
            Serial.print("Test succeeded! Time elapsed: ");
            Serial.printf("%lu.%03lu sec\n", millis() / 1000, millis() % 1000);
        }
};

void setup() {
    // put your setup code here, to run once:

    Serial.begin(115200);
    while(!Serial);

    Test justATestObject;
}

void loop() {
    // put your main code here, to run repeatedly:
}
```

### Debug Message

```plain
Invalid use of member function 'void Test::testFn()' (did you forget the '()' ?)
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-07-18:
@radiantBear - Is it what you are looking for?

``` cpp
#include "esp_timer.h"

class Test {
    public:
        Test(void) {
            esp_timer_handle_t testTimer;
            esp_timer_create_args_t timerConfig;
            timerConfig.arg = this;
            timerConfig.callback = reinterpret_cast<esp_timer_cb_t>(testFn);
            timerConfig.dispatch_method = ESP_TIMER_TASK;
            timerConfig.name = "Test_Timer";
            esp_timer_create(&timerConfig, &testTimer);
            esp_timer_start_periodic(testTimer, 1000000);
        }
        
        static void testFn(void *arg) {
          Test *obj = (Test *)arg;
          obj->myTestFn();
        }
          
        void myTestFn(void) {
            Serial.print("Test succeeded! Time elapsed: ");
            Serial.printf("%lu.%03lu sec\n", millis() / 1000, millis() % 1000);
        }
};

void setup() {
    // put your setup code here, to run once:

    Serial.begin(115200);
    while(!Serial);

    Test justATestObject;
}

void loop() {
    // put your main code here, to run repeatedly:
}
```

radiantBear, 2023-07-18:
@SuGlider Thank you, that's great! Not sure why I didn't think of that...

I guess there was a second part to my question that I didn't state very clearly, though. Is it possible to have the same functionality with `timerAttachInterrupt()`? It doesn't seem as though there's an equivalent method for setting the callback arguments for that function. Also, could you please clarify the difference between the functions in `esp-hal-timer.h` [(here)](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-timer.h) and `esp_timer.h` [(here)](https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/esp32/include/esp_timer/include/esp_timer.h)? Why are there two different groups of functions to use the ESP32's hardware timers?

SuGlider, 2023-07-18:
> Is it possible to have the same functionality with timerAttachInterrupt()? It doesn't seem as though there's an equivalent method for setting the callback arguments for that function. Also, could [...]

The `esp32-hal-timer` layer has been refactored for the next Arduino Core 3.0.0 version.
It has some breaking changes that can be seen in the [header file](https://github.com/espressif/arduino-esp32/blob/esp-idf-v5.1-libs/cores/esp32/esp32-hal-timer.h)
We think that it makes the whole API simpler. Please let me know your opinion.

Anyway, I agree that we can add `std::function` to it and make it closer to the GPIO Interrupt API for C++.

Regarding the difference of the 2 layers, the `esp32-hal` layer is the interface from Arduino API to the IDF layer. By other hand the sdk esp32 files are just pure IDF header files, meaning that those are the IDF functions. `esp32-hal` calls functions from the IDF sdk and expose an API closer to Arduino, whenever possible.
Both layer are in C. Your project can use any of them, one or the other. 
Mixing both may require reading the code and avoiding conflicts, therefore, maybe not so simple.

radiantBear, 2023-07-18:
Okay, that makes sense. Thank you for the clarification! I agree as well; outside of needing to call a class method from a timer interrupt, the `esp32-hal-timer` certainly simplifies API use. Thank you for moving to add `std::function` to it as well, that will be very helpful!

closed 2023-07-18 by radiantBear