espressif/arduino-esp32 #8423: PSRAM in ESP32

savagerex, 2023-07-18:
### Board

ESP32-WROVER-E

### Device Description

NONE

### Hardware Configuration

NONE

### Version

v2.0.9

### IDE Name

Arduino

### Operating System

Windows 11

### Flash frequency

80M

### PSRAM enabled

yes

### Upload speed

921600

### Description

when i use this code in ESP32-S3- DevKitC-1, it's ok.

but i use this code in ESP32-WROVER-E. then ESP32 always reset.

code as below.

### Sketch

```cpp
#include <WiFi.h>
#include <MQTT.h>

const char* ssid = "ASUS-38";     //SW1-WIFI6   ASUS-38   vivo Y21 
const char* pass = "12345678";   //wireless    12345678   1234567890

const char* mqtt_server = "192.168.50.129";   //broker.emqx.io  192.168.50.128
const char *topic = "DataTopic";

//uint8_t DataBufRead[18500];
uint8_t* DataBufRead;

WiFiClient net;
MQTTClient client(1700500);  
//MQTTClient client(162500);
//MQTTClient client(20500);

void connect() {
  Serial.print("checking wifi...");
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(1000);
  }

  Serial.print("\nconnecting...");

  while (!client.connect("arduino", "public", "public")) {
    Serial.print(".");
    delay(1000);
  }

  Serial.println("\nconnected!");

  
}

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);

  DataBufRead = (uint8_t*)ps_calloc(1700500, sizeof(uint8_t));
//  DataBufRead = (uint8_t*)ps_calloc(162500, sizeof(uint8_t));
//  DataBufRead = (uint8_t*)ps_calloc(20500, sizeof(uint8_t));

  delay(10);
    WiFi.begin(ssid,pass);

    Serial.print("Connecting.");
    while(WiFi.status() != WL_CONNECTED) {
        Serial.print(".");
        delay(500);
    }

    Serial.print("WiFi connected - IP address: ");
    Serial.println(WiFi.localIP());

    client.begin(mqtt_server, net);

  Serial.print("\nconnecting...");
  while (!client.connect("arduino", "public", "public")) {
    Serial.print(".");
    delay(1000);
  }

  Serial.println("\nconnected!");

  client.loop();

  if (!client.connected()) {
    connect();
  }

  for (uint32_t i = 0; i<1699999; i++)
  //for (uint32_t i = 0; i<161999; i++)
  //for (uint32_t i = 0; i<19999; i++)
  {
    DataBufRead[i] = 'a';
  }

   char *DataBuf = (char*)DataBufRead;
   bool PubData = client.publish(topic, DataBuf, 1700000, false, 0);  //18000 //6000
   Serial.println(PubData);

   
   Serial.println("Mqtt Finish");
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Debug Message

```plain
...WiFi connected - IP address: 192.168.50.150

connecting...Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4014a63b  PS      : 0x00060f30  A0      : 0x800d4ec8  A1      : 0x3ffb2120  
A2      : 0xffffffff  A3      : 0x0019f294  A4      : 0x00000010  A5      : 0x00000004  
A6      : 0x00000000  A7      : 0x00000fff  A8      : 0x3ffb2160  A9      : 0x00000000  
A10     : 0x00000d6a  A11     : 0x00000000  A12     : 0x00000d6a  A13     : 0x00007540  
A14     : 0x00000008  A15     : 0x3ffc4550  SAR     : 0x0000001c  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x4008a58c  LEND    : 0x4008a5a8  LCOUNT  : 0x00000000  

Backtrace: 0x4014a638:0x3ffb2120 0x400d4ec5:0x3ffb2140 0x400d4b9e:0x3ffb2190 0x400d4881:0x3ffb21e0 0x400d28bf:0x3ffb2250 0x400d616a:0x3ffb2290

ELF file SHA256: 40e317c64fdab27d

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-07-18:
@savagerex - Please test `DataBufRead` to make sure it is not NULL before using it.

SuGlider, 2023-07-18:
@savagerex - Please make sure you have selected the `ESP32 Wrover Module` board in the IDE menu. 
It enables PSRAM by default.

savagerex, 2023-07-18:
@SuGlider 

 i have enabled PSRAM in ESP32 Wrover Module

SuGlider, 2023-07-18:
@savagerex - The `StoreProhibited` error and `EXCVADDR: 0x00000000` indicate that the application tried to store data in a NULL pointer. I guess that the issue is that `DataBufRead` is NULL right after executing `DataBufRead = (uint8_t*)ps_calloc(1700500, sizeof(uint8_t));`. 

Please test `if (DataBufRead != NULL)` to make sure that this is not the problem.

SuGlider, 2023-07-18:
You can also use the `ESP32 Exception Decoder` for finding out where did it break.
`Backtrace: 0x4014a638:0x3ffb2120 0x400d4ec5:0x3ffb2140 0x400d4b9e:0x3ffb2190 0x400d4881:0x3ffb21e0 0x400d28bf:0x3ffb2250 0x400d616a:0x3ffb2290`

This link may help to understand how to install it and use it:
https://www.donskytech.com/esp32-exception-decoder/

savagerex, 2023-07-18:
@SuGlider i modified the code.

it seems not problem for  DataBufRead = (uint8_t*)ps_calloc(1700500, sizeof(uint8_t));

the code as below

#include <WiFi.h>
#include <MQTT.h>

const char* ssid = "ASUS-38";     //SW1-WIFI6   ASUS-38   vivo Y21 
const char* pass = "12345678";   //wireless    12345678   1234567890

const char* mqtt_server = "192.168.50.129";   //broker.emqx.io  192.168.50.128
const char *topic = "DataTopic";

//uint8_t DataBufRead[18500];
uint8_t* DataBufRead;

WiFiClient net;
MQTTClient client(1700500);  
//MQTTClient client(162500);
//MQTTClient client(20500);

void connect() {
  Serial.print("checking wifi...");
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(1000);
  }

  Serial.print("\nconnecting...");

  while (!client.connect("arduino", "public", "public")) {
    Serial.print(".");
    delay(1000);
  }

  Serial.println("\nconnected!");

  
}

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);

//  DataBufRead = (uint8_t*)ps_calloc(1700500, sizeof(uint8_t));
  
//  DataBufRead = (uint8_t*)ps_calloc(162500, sizeof(uint8_t));
//  DataBufRead = (uint8_t*)ps_calloc(20500, sizeof(uint8_t));

//  delay(10);
//    WiFi.begin(ssid,pass);
//
//    Serial.print("Connecting.");
//    while(WiFi.status() != WL_CONNECTED) {
//        Serial.print(".");
//        delay(500);
//    }
//
//    Serial.print("WiFi connected - IP address: ");
//    Serial.println(WiFi.localIP());
//
//
//    client.begin(mqtt_server, net);
//
//
//
//  Serial.print("\nconnecting...");
//  while (!client.connect("arduino", "public", "public")) {
//    Serial.print(".");
//    delay(1000);
//  }
//
//  Serial.println("\nconnected!");
//
//  client.loop();
//
//  if (!client.connected()) {
//    connect();
//  }

  DataBufRead = (uint8_t*)ps_calloc(1700500, sizeof(uint8_t));    // 1700500

  for (uint32_t i = 0; i<1699999; i++)       // 1699999
  //for (uint32_t i = 0; i<161999; i++)
  //for (uint32_t i = 0; i<19999; i++)
  {
    DataBufRead[i] = 'a';
  }

  Serial.println(DataBufRead[0]);

//   char *DataBuf = (char*)DataBufRead;
//   bool PubData = client.publish(topic, DataBuf, 1700000, false, 0);  //18000 //6000
//   Serial.println(PubData);

   
   Serial.println("Mqtt Finish");
}

void loop() {
  // put your main code here, to run repeatedly:

}

savagerex, 2023-07-18:
When the program executes here ( while (!client.connect("arduino", "public", "public")) { ) .

esp32 will reset

SuGlider, 2023-07-18:
@savagerex - Please enable the `Core Debug Level:` in the IDE Board Menu to `"Verbose"` and post here the output messages.

savagerex, 2023-07-19:
I think ESP32-WROVER-E have problem.

i use this sample code in ESP32-WROVER-E. it's ok.

code as below. 

// This example uses an ESP32 Development Board
// to connect to shiftr.io.
//
// You can check on your device after a successful
// connection here: https://www.shiftr.io/try.
//
// by Joël Gähwiler
// https://github.com/256dpi/arduino-mqtt

#include <WiFi.h>
#include <MQTT.h>

const char ssid[] = "ASUS-38";
const char pass[] = "12345678";

//const char *topic = "DataTopic";
//uint8_t* DataBufRead;

WiFiClient net;
//MQTTClient client(1700500);
MQTTClient client;

unsigned long lastMillis = 0;

void connect() {
  Serial.print("checking wifi...");
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(1000);
  }

  Serial.print("\nconnecting...");
  while (!client.connect("arduino", "public", "public")) {
    Serial.print(".");
    delay(1000);
  }

  Serial.println("\nconnected!");

  client.subscribe("/hello");
  // client.unsubscribe("/hello");
}

void messageReceived(String &topic, String &payload) {
  Serial.println("incoming: " + topic + " - " + payload);

  // Note: Do not use the client in the callback to publish, subscribe or
  // unsubscribe as it may cause deadlocks when other things arrive while
  // sending and receiving acknowledgments. Instead, change a global variable,
  // or push to a queue and handle it in the loop after calling `client.loop()`.
}

void setup() {
  Serial.begin(115200);
  WiFi.begin(ssid, pass);

  //DataBufRead = (uint8_t*)ps_calloc(1700500, sizeof(uint8_t));

  // Note: Local domain names (e.g. "Computer.local" on OSX) are not supported
  // by Arduino. You need to set the IP address directly.
  client.begin("192.168.50.129", net);
  client.onMessage(messageReceived);

  connect();

  client.loop();
  delay(10);  // <- fixes some issues with WiFi stability

  if (!client.connected()) {
    connect();
  }

//  for (uint32_t i = 0; i<1699999; i++)       // 1699999
//  //for (uint32_t i = 0; i<161999; i++)
//  //for (uint32_t i = 0; i<19999; i++)
//  {
//    DataBufRead[i] = 'a';
//  }
//
//  Serial.println(DataBufRead[0]);
//
//   char *DataBuf = (char*)DataBufRead;
//   bool PubData = client.publish(topic, DataBuf, 1700000, false, 0);  //18000 //6000
//   Serial.println(PubData);
//
//   Serial.println("Mqtt Finish");
}

void loop() {
//  client.loop();
//  delay(10);  // <- fixes some issues with WiFi stability
//
//  if (!client.connected()) {
//    connect();
//  }
//
//  // publish a message roughly every second.
//  if (millis() - lastMillis > 1000) {
//    lastMillis = millis();
//    client.publish("/hello", "world");
//  }
}

savagerex, 2023-07-19:
if declare "MQTTClient client;" ,  it's ok.

if declare "MQTTClient client(1700500);" , it's fail.

SuGlider, 2023-07-19:
> if declare "MQTTClient client;" , it's ok.
> 
> if declare "MQTTClient client(1700500);" , it's fail.

#### There is nothing that we can do to fix it here in ESP32 Arduino Github.

This seems to be something related to the MQTT Library and not to the ESP32 Arduino Core.
https://github.com/256dpi/arduino-mqtt  is the place for this question/issue.

This is the code from the Library construtor:

``` cpp
// from MQTTClient.h
  explicit MQTTClient(int bufSize = 128) : MQTTClient(bufSize, bufSize) {}
  MQTTClient(int readSize, int writeBufSize);

// from MQTTClient.cpp
MQTTClient::MQTTClient(int readBufSize, int writeBufSize) {
  // allocate buffers
  this->readBufSize = (size_t)readBufSize;
  this->writeBufSize = (size_t)writeBufSize;
  this->readBuf = (uint8_t *)malloc((size_t)readBufSize + 1);
  this->writeBuf = (uint8_t *)malloc((size_t)writeBufSize);
}
```

As you can see it will try to allocate 1.7MB twice. The code also doesn't test the result from `malloc()` to make sure it is not `NULL`. Therefore, the Library code generates the `StoreProhibited` Panic and Reset.

lbernstone, 2023-07-19:
I don't think there is any reason to have the buffer size larger than the maximum message size.

SuGlider, 2023-07-19:
The theoretical maximum length of an MQTT packet is 268,435,456 bytes, equivalent to 256 MB.
But it also depends on the MQTT Server and Client settings.
https://www.emqx.com/en/blog/best-practices-of-maximum-packet-size-in-mqtt

savagerex, 2023-07-19:
if declare "MQTTClient client(1700500);"
i use ESP32 S3. it's ok

so i think MQTT lib, it is not problem

CyBaart, 2023-07-19:
@savagerex @SuGlider 

Greetings from Cyprus.

Sorry to hijack this page but can I please ask you to to take a look at an old post https://github.com/espressif/arduino-esp32/issues/6796#issuecomment-1641456394

I am going bald pulling my hair out!

Thank you, Steve (CyBaart)

SuGlider, 2023-07-19:
> if declare "MQTTClient client(1700500);" i use ESP32 S3. it's ok
> 
> so i think MQTT lib, it is not problem

@savagerex - There is a reason for it to work fine on the S3 and not in the WROVER-E.
This is related to PSRAM available space and how much of it is addressable by the chip.
The S3 can address the whole 8MB PSRAM.
The ESP32 can't. It can only address 4MB. Therefore, when the MQTT tries to allocate 1.7MB x 2, it may reach the 4MB limit and fails `malloc()`, making the pointer of MQTT buffer NULL. The code MUST always check such returns and exception situation. The MQTT Lib doesn't test for NULL and this is the main issue.

ESP32-WROVER-E or any other ESP32 with PSRAM will suffer with the same problems. 
There is a HiMem IDF API to switch 4MB memory banks in the ESP32, but it can't use the whole 8MB as a continuous PSRAM addressable space.

The ESP32-S3 has no such problem and is able address the whole 8MB PSRAM continuosly.
https://esp32.com/viewtopic.php?t=29938#:~:text=There%27s%20no%20need%20for%20himem,space%20for%204MiB%20of%20PSRAM.
```
There's no need for himem on the ESP32-S3. 
Himem is a workaround on the original ESP32 to work around the fact that it only has addressing space 
for 4MiB of PSRAM. 
That addressing space got expanded to 32MiB (shared with flash) for data on the ESP32-S3; 
it means you can simply enable PSRAM and allocate the entire 8MiB using malloc()/calloc or 
heap_caps_alloc().
```

This is the documentation about ESP32 Himem:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/himem.html

savagerex, 2023-07-20:
ok ,i see.

do you have "himem sample code" on arduino ????

savagerex, 2023-07-20:
Can you teach me to use himem???

lbernstone, 2023-07-20:
https://github.com/espressif/esp-idf/tree/master/examples/system/himem

lbernstone, 2023-07-20:
If you really want to bake your noodle, you should be able to modify https://github.com/lbernstone/psram_custom_allocator to use himem.

savagerex, 2023-07-20:
OK, Thanks

SuGlider, 2023-07-20:
> Can you teach me to use himem???

From the [documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/himem.html):

The ESP32 can access external SPI RAM transparently, so you can use it as normal memory in your program code. However, because the address space for external memory is limited in size, only the first 4MiB can be used as such. Access to the remaining memory is still possible, however this needs to go through a bankswitching scheme controlled by the himem API.

Specifically, what is implemented by the himem API is a bankswitching scheme. Hardware-wise, the 4MiB region for external SPI RAM is mapped into the CPU address space by a MMU, which maps a configurable 32K bank/page of external SPI RAM into each of the 32K pages in the 4MiB region accessed by the CPU. For external memories that are <=4MiB, this MMU is configured to unity mapping, effectively mapping each CPU address 1-to-1 to the external SPI RAM address.

In order to use the himem API, you have to enable it in the menuconfig using [CONFIG_SPIRAM_BANKSWITCH_ENABLE](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-spiram-bankswitch-enable), as well as set the amount of banks reserved for this in [CONFIG_SPIRAM_BANKSWITCH_RESERVE](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-spiram-bankswitch-reserve). This decreases the amount of external memory allocated by functions like malloc(), but it allows you to use the himem api to map any of the remaining memory into the reserved banks.

The himem API is more-or-less an abstraction of the bankswitching scheme: it allows you to claim one or more banks of address space (called ‘regions’ in the API) as well as one or more of banks of memory to map into the ranges.

savagerex, 2023-07-20:
OK!!! Thanks

CyBaart, 2023-07-20:
@savagerex 

Again, my apologies for posting here. Can you please take another look at an old post https://github.com/espressif/arduino-esp32/issues/6796#issuecomment-1641456394  I would really appreciate your help and if you visit Cyprus (Paphos, not Limassol...) I will buy you several large drinks! Is it possible to discuss in messenger or WhatsApp etc so I am not posting irrelevant things on this forum?

Cheers, Steve

savagerex, 2023-07-21:
@SuGlider 

In fact, I don't know how to use Hinmem.

if i want to locate the PSRAM.
the code as below
uint8_t* DataBufRead;
DataBufRead = (uint8_t*)ps_calloc(600, sizeof(uint8_t));

if  want to locate the Himem
????
how to write??? 
Do you write the sample code for me???

i found the example code in website.
https://github.com/arkhipenko/ESP32Himem

but I don't know how to use it 

please help me.

Thanks a lot

SuGlider, 2023-07-21:
I have found a Library that may work with Arduino ... I didn't test it.
https://github.com/arkhipenko/ESP32Himem

savagerex, 2023-07-21:
@SuGlider 

yes, i found this lib too.
but I don't know how to use it.

i hope you can write the code for locate himem (1728000).

Thanks a lot!!!

SuGlider, 2023-07-22:
Hi @savagerex - 

I have tested the IDF example and the ESP32Himem Library.
None of them work within Arduino Core 2.0.10.

The single fact of including `esp32/himem.h` into the sketch causes a reset.

@me-no-dev - Do you how to use Himem within Arduino Core 2.0.10?

me-no-dev, 2023-07-24:
@SuGlider  We do not support HIMEM in Arduino. People can use it through "as-component" and get support from the IDF team. This is not a trivial feature :)

SuGlider, 2023-07-24:
Thanks @me-no-dev!

SuGlider, 2023-07-24:
@savagerex - I think that this issue is fully clarified. Let me know if you have further related questions.

savagerex, 2023-07-24:
OK, Thanks

savagerex, 2023-07-24:
@me-no-dev 

what do you mean "as-component"  ????

lbernstone, 2023-07-25:
https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html

savagerex, 2023-07-25:
as-component means esp-idf_component  ???

lbernstone, 2023-07-25:
Yes.  If your time is valuable, and you really need the additional psram, then I would strongly suggest you use an S3, which directly can access all the spiram using the psram libraries.

savagerex, 2023-07-25:
OK!!! I see

SuGlider, 2023-07-25:
I'll close this issue by now. Please feel free to open it again whenever necessary.

closed 2023-07-25 by SuGlider