espressif/arduino-esp32 #8424: Core Compilation failure

josesnchz, 2023-07-18:
### Board

Adafruit ESP32 Feather - HUZZAH32

### Device Description

Plain Module in breadboard

### Hardware Configuration

Nothing attached to the dev board. Using USB power. 

### Version

v2.0.9

### IDE Name

Arduino IDE 

### Operating System

Ubuntu 22.04 with Kernel 5.15.0-76

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

Running version v2.0.10, using any sketch, when it comes the moment to compile the core, I always get the same error. Downgrading to v2.0.9 solves the problem.

### Sketch

```cpp
/*
  Blink

  Turns an LED on for one second, then off for one second, repeatedly.

  Most Arduinos have an on-board LED you can control. On the UNO, MEGA and ZERO
  it is attached to digital pin 13, on MKR1000 on pin 6. LED_BUILTIN is set to
  the correct LED pin independent of which board is used.
  If you want to know what pin the on-board LED is connected to on your Arduino
  model, check the Technical Specs of your board at:
  https://www.arduino.cc/en/Main/Products

  modified 8 May 2014
  by Scott Fitzgerald
  modified 2 Sep 2016
  by Arturo Guadalupi
  modified 8 Sep 2016
  by Colby Newman

  This example code is in the public domain.

  https://www.arduino.cc/en/Tutorial/BuiltInExamples/Blink
*/

// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second
}
```

### Debug Message

```plain
Compiling core...
bash -c "echo '-DARDUINO_CORE_BUILD' > /tmp/arduino_build_925905/file_opts"
>: -c: line 1: unexpected EOF while looking for matching `''
>: -c: line 2: syntax error: unexpected end of file
exit status 2
Error compiling for board Adafruit ESP32 Feather.
```

### Other Steps to Reproduce

Tried executing the bash command that fails manually. It does not fail.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-07-18:
@me-no-dev - I think that this is the second issue related to Linux that I see.

The other similar one was reported in https://github.com/espressif/arduino-esp32/issues/8421#issuecomment-1638824609

@PilnyTomas - Can you please test it with Linux and let us know? Thanks.

PilnyTomas, 2023-07-19:
Linux Fedora 37
Arduino 2.0.10
Sketch Blink
Compiling for

- Adafruit ESP32 Feather
- Adafruit Feather ESP32 V2

Results - both compiled successfully

josesnchz, 2023-07-19:
When using the apt-obtained version of the Arduino IDE, which I guess it is 1.8.19, the 2.0.10 version triggers the error I documented, which can be solved by downgrading to 2.0.9.
Using the latest version of the Arduino IDE (2.1.1) makes the error with 2.0.10 dissapear. Everything gets compiled correctly now.

pennam, 2023-07-19:
removing the single quote around `'-DARDUINO_CORE_BUILD'` in `platform.txt` file seems to fix the issue with the apt obtained version of the ide.

https://github.com/pennam/arduino-esp32/compare/arduino_nano_esp32...eof_fix

aalucia, 2023-07-24:
Hi,
same issue (: -c: line 1: unexpected EOF while searching for a match `'')  on:
 
SUSE Linux Leap 15.5
IDE 1.8.12
Board Lolin 32 S2 Mini

Cheers.

closed 2023-07-24 by SuGlider

aalucia, 2023-07-24:
> A new Arduino Core 2.0.11 has been released.
> https://github.com/espressif/arduino-esp32/releases/tag/2.0.11
>
> It shall fix the Linux problem using the Fix that @pillo79 has posted in #8433.

Cool !!!
Thanks.

PR #8433 (2023-07-20): 2.0.10 escaping fix
## Description of Change
Fix escaping issues on 2.0.10 (Windows path and IDE 1.x). 

## Tests scenarios
Tested on clean Windows 11 box, Arduino IDE 2.1.1, username with spaces, 2.0.10. The issue is present and is fixed by these commits.
Includes patch by @pennam fixing #8424 for IDE 1.8.x.  

## Related links
Closes #8424 and fixes Windows path issues reported in Arduino forums e.g. [here](https://forum.arduino.cc/t/compilation-error-exit-status-1/1149342) and [here](https://forum.arduino.cc/t/compiling-for-esp32-fails-access-is-denied/1149574)