espressif/arduino-esp32 #8426: I2S pin setting does not work for ESP32-S3

janczeresnia, 2023-07-18:
### Board

ESP32-S3 Dev Kit N8R8 Waveshare

### Device Description

https://www.waveshare.com/wiki/ESP32-S3-DEV-KIT-N8R8

### Hardware Configuration

MAX 98357A connected to various GPIO ports in the microcontroller. I tested all connections to most GPIOs.

### Version

v2.0.9

### IDE Name

Arduino IDE 1.8.19

### Operating System

Linux Gentoo

### Flash frequency

80 Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

The problem is that no matter how I set the pins for I2S, the program always returns an error:
i2s_set_pin failed; attempted settings: SCK=14; FS=25; DIN=35; DOUT=26
The error appears only for the ESP32-S3 board with other boards working fine.

### Sketch

```cpp
/*
 * https://github.com/espressif/arduino-esp32/blob/master/libraries/I2S/examples/SimpleTone/SimpleTone.ino
 */

#include <I2S.h>
const int frequency = 1000; // frequency of square wave in Hz
const int amplitude = 500; // amplitude of square wave
const int sampleRate = 8000; // sample rate in Hz
const int BITS_PER_SAMPLE = 16;

const int halfWavelength = (sampleRate / frequency); // half wavelength of square wave

short sample = amplitude; // current sample value
int count = 0;
// I2S_PHILIPS_MODE / I2S_RIGHT_JUSTIFIED_MODE / I2S_LEFT_JUSTIFIED_MODE
i2s_mode_t I2S_MODE = I2S_PHILIPS_MODE; // I2S decoder is needed

void setup() {
  Serial.begin(115200);
  while (!Serial or (millis() < 1000)) {delay(200);}
  Serial.println(F("\nESP32-S3 I2S DAC Simple Tone\n"));

  // start I2S at the sample rate with 16-bits per sample
  if (!I2S.begin(I2S_MODE, sampleRate, BITS_PER_SAMPLE)) {
    Serial.println("Failed to initialize I2S!");
    while (true); // do nothing
  }

  I2S.setSckPin(4);  // BCLK
  I2S.setFsPin(5);   // LRCLK
  I2S.setDataPin(18); // DIN
}

void loop() {
    if (count % halfWavelength == 0 ) {
      // invert the sample every half wavelength count multiple to generate square wave
      sample = -1 * sample;
    }

    if(I2S_MODE == I2S_PHILIPS_MODE) { // write the same sample twice, once for Right and once for Left channel
      // write the same sample twice, once for Right and once for Left channel
      if(BITS_PER_SAMPLE == 32) {
        I2S.write(sample*65536); // Convert 16bit to 32bit
        I2S.write(sample*65536);
      } else if (BITS_PER_SAMPLE == 24) {
        I2S.write(sample*256); // Convert 16bit to 24bit
        I2S.write(sample*256);
      } else if (BITS_PER_SAMPLE == 16) {
        I2S.write(sample);  // Right channel
        I2S.write(sample);  // Left channel
      }
    } else if (I2S_MODE == I2S_RIGHT_JUSTIFIED_MODE || I2S_MODE == I2S_LEFT_JUSTIFIED_MODE){
      // write the same only once - it will be automatically copied to the other channel
      if(BITS_PER_SAMPLE == 32) {
        I2S.write(sample*65536); // Convert 16bit to 32bit
      } else if (BITS_PER_SAMPLE == 24) {
        I2S.write(sample*256); // Convert 16bit to 24bit
      } else if (BITS_PER_SAMPLE == 16) {
        I2S.write(sample);
      }
    }
    // increment the counter for the next sample
    count++;
}
```

### Debug Message

```plain
(17) I2S: i2s_set_pin(295): ws_io_num invalid
[   103][E][I2S.cpp:341] _applyPinSetting(): i2s_set_pin failed; attempted settings: SCK=14; FS=25; DIN=35; DOUT=26
[   106][E][I2S.cpp:214] _installDriver(): could not apply pin setting during driver install
[   115][E][I2S.cpp:291] begin(): ERROR: failed to install driver
Failed to initialize I2S!
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2023-07-18:
pins should be set before `I2S.begin()`

janczeresnia, 2023-07-18:
Thank you. My mistake everything works fine. Regards

closed 2023-07-18 by janczeresnia