espressif/arduino-esp32 #8428: Adafruit QT Py ESP32-S3 does not run the code. Disconnects after uploading

Richrfl, 2023-07-18:
### Board

Adafruit QT Py ESP32-S3

### Device Description

Just running the three lines code "hola word"

### Hardware Configuration

Nothing attached to the board

### Version

other

### IDE Name

Arduino IDE V2.1.1

### Operating System

Windows 11

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

The upload finished ok, but I got nothing in the Serial Monitor. I tried other codes and the result is the same. 
The Serial Monitor is set to 9600. I added a 500ms delay between the Serial.begin and Serial.print sentences and makes no difference. If I press RST, then the board disconnects. Also, unless I do a Boot sequence (hold BOOT press RST release BOOT), the IDE will not "see" the COM port. After this process, the led in the board will only blink when I hook up the USB Cable, but never again the rainbow sequence.

### Sketch

```cpp
void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);
  delay(500);
  Serial.println("Hello World!");
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Debug Message

```plain
No warning or error messages
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-07-18:
Enable CDC
[image]

SuGlider, 2023-07-18:
@Richrfl - I think that the problem is related to the fact that when the `Adafruit QT Py ESP32-S3` board is selected in the Arduino IDE, it defines USB mode as `USB-OTG (TinyUSB)`.

The ESP32-S3 has two mode for the USB: TinyUSB (OTG) or  Hardware CDC.

With OTG, the Serial Monitor must be closed and reopen after uploading, because Windows must renumerate USB in order to make it available. OTG allows other USB functions such as the MSC, whenever it is availble. Adafruit has an specific firmware that allows Windows to open a new drive in the computer to upload firmware with drag and drop of the .bin file.

With Hardware CDC, it is not necessary to close/open the Serial Monitor every single time that a new sketch is uploaded. The point here is that the USB only works as CDC (serial communication).
It is a lot easier to use this mode for Serial communications.

>The Serial Monitor is set to 9600. I added a 500ms delay between the Serial.begin and Serial.print sentences and makes no difference. 

For USB the baudrate is irrelevant.

>If I press RST, then the board disconnects. 

RST make the USB device to reset (ESP32-S3) and Windows will detect it and disable the COM port.

>Also, unless I do a Boot sequence (hold BOOT press RST release BOOT), the IDE will not "see" the COM port. 

When doing it, the ESP32-S3 will enter in Download Mode. In this mode it will enable the Hardware CDC and a COM port shall be visible in Windows. When uploading, the ESPTOOL will try to put the S3 in this mode and use such port.

Setting the IDE to use `Hardware CDC and JTAG` in `USB Mode:` option will make the whole experience better, but USB OTG won't be available... If you just want to print something in the Serial Monitor, use the Hardware CDC mode.

>After this process, the led in the board will only blink when I hook up the USB Cable, but never again the rainbow sequence.

Not sure about it. What is the sketch that you are running?

SuGlider, 2023-07-18:
Try the setup below in the IDE. 

In order to flash a new firmware in this configuration, I had to open the Device Manager, put the S3 in Download Mode (BOOT + Pulsing RESET/EN) and then select the new COM port in the IDE. Start the upload and at the end press RESET/EN in the board.
After it, I had to select again the COM port, for the last time.
From this point on, it will upload and show the Serial Monitor output always, like whenn using the UART.

[image]

Richrfl, 2023-07-18:
Here is the simple code I am trying to run. After I get this working I will add a gyro board and after WiFi pier to pier to another same ESP32 to control a couple of servos

```cpp
void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);
  delay(500);
  Serial.println("Hello World!");
}

void loop() {
  // put your main code here, to run repeatedly:

}
```
I will make the suggested changes right now and will report in a couple of minutes.

Richrfl, 2023-07-19:
Sorry, but it didn't work
[image]
[image]
[image]

Richrfl, 2023-07-19:
Sorry, when you say "open the Device Manager" do you mean Windows Device Manager? If so, to do what? I see there is a USB JTAG/serial device unit and in Ports (COM and LPT) there is a USB Serial device with an assigned port that is the same I have in the IDE. However, I notice that the IDE is showing me that port with the text "DFRobotBeetle ESP32-C3" and my board is S3. Don't know if that is important.

SuGlider, 2023-07-19:
Please modify the settings marked in the image below

[image]

To this:

[image]

Then upload the sketch to use those settings in the next reset of the board.

SuGlider, 2023-07-19:
Adafruit has also its own documentation:
https://learn.adafruit.com/adafruit-qt-py-esp32-s3/arduino-neopixel-blink#start-up-arduino-ide-and-select-board-slash-port-3108136

Richrfl, 2023-07-19:
Great!!! It did work! Thank you very much!!

closed 2023-07-19 by Richrfl