espressif/arduino-esp32 #8429: “Preferences.h” doesn't work on ESP32 S3

Someijam, 2023-07-19:
### Board

ESP32S3 Dev Module

### Device Description

DevKitC-1, N32R8V, just a single board.

### Hardware Configuration

Only connected to the computer without anything else. All GPIOs are free.

### Version

v2.0.9

### IDE Name

Arduino IDE 2.1.1

### Operating System

macOS Ventura 13.4.1（M1 chip）

### Flash frequency

80 MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

Try a simple example "StartCounter"(libraries/Preferences/examples/StartCounter/StartCounter.ino) on my board. Code as follows:
```cpp
/*
 ESP32 startup counter example with Preferences library.

 This simple example demonstrates using the Preferences library to store how many times
 the ESP32 module has booted. The Preferences library is a wrapper around the Non-volatile
 storage on ESP32 processor.

 created for arduino-esp32 09 Feb 2017
 by Martin Sloup (Arcao)
*/

#include <Preferences.h>

Preferences preferences;

void setup() {
  Serial.begin(115200);
  Serial.println();

  // Open Preferences with my-app namespace. Each application module, library, etc
  // has to use a namespace name to prevent key name collisions. We will open storage in
  // RW-mode (second parameter has to be false).
  // Note: Namespace name is limited to 15 chars.
  preferences.begin("my-app", false);

  // Remove all preferences under the opened namespace
  //preferences.clear();

  // Or remove the counter key only
  //preferences.remove("counter");

  // Get the counter value, if the key does not exist, return a default value of 0
  // Note: Key name is limited to 15 chars.
  unsigned int counter = preferences.getUInt("counter", 0);

  // Increase counter by 1
  counter++;

  // Print the counter to Serial Monitor
  Serial.printf("Current counter value: %u\n", counter);

  // Store the counter to the Preferences
  preferences.putUInt("counter", counter);
  Serial.print("Free:");
  Serial.println(preferences.freeEntries());

  // Close the Preferences
  preferences.end();

  // Wait 3 seconds
  Serial.println("Restarting in 3 seconds...");
  delay(3000);

  // Restart ESP
  ESP.restart();
}

void loop() {}
```

I changed 10 seconds to 3 seconds it doesn't matter I think. After I uploaded, the counter value is always "1". However, it was supposed to add "1" everytime, instead of keeping "1".
This is just a simple example. Thus, more complex things like this, too. "putString" doesn't save the string you provided, etc.

### Sketch

```cpp
...
  unsigned int counter = preferences.getUInt("counter", 0);
  counter++;
  Serial.printf("Current counter value: %u\n", counter);
  preferences.putUInt("counter", counter);
...
```

### Debug Message

```plain
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x42022fd2
SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
load:0x3fce3808,len:0x3ac
load:0x403c9700,len:0x9b4
load:0x403cc700,len:0x28d8
entry 0x403c98bc

Current counter value: 1
Free:628
Restarting in 3 seconds...
�ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x42022fd2
SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
load:0x3fce3808,len:0x3ac
load:0x403c9700,len:0x9b4
load:0x403cc700,len:0x28d8
entry 0x403c98bc

Current counter value: 1
Free:628
Restarting in 3 seconds...
//output loops like this
```

### Other Steps to Reproduce

1. Just verify the code and upload, then open the monitor, "Current counter value" is always "1", nothing special.

2. If you check with "int pStatus = preferences.putUInt("counter", counter);", and the "pStatus" is "4", which is the right return value.

3. My friend who uses Windows 10 and uses the same ESP32S3 board also meet the problem.

4. I tested the code on "wokwi.com"(an online simulator) but it works normally...(God, That's weird?)

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-07-19:
Works for me.  Make sure you have the correct flash settings (OPI)

Someijam, 2023-07-20:
> Works for me (in Arduino 1.8.16).

Thank you very much!!!😲
After I changed the "Flash Mode" to "OPI 80MHz" in the menu, it works!

closed 2023-07-20 by Someijam