espressif/arduino-esp32 #8430: ESP32 Arduino boards package error in 2.0.10

CanHobby, 2023-07-19:
### Board

all ESP32-xxx

### Device Description

makes no difference as to which board I am using..  I am only trying to compile - not even uploading.

I use Ubuntu 23.04   -  Arduino IDE 1.8.10

any compile gives the error:

Arduino: 1.8.10 (Linux), Board: "ESP32 Dev Module, Disabled, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, Core 1, Core 1, None, Disabled"

Compiling core...
bash -c "echo '-DARDUINO_CORE_BUILD" > "/tmp/arduino_build_696887/file_opts\""
>: -c: line 1: unexpected EOF while looking for matching `''
exit status 2
Error compiling for board ESP32 Dev Module.

Many times the IDE actually just hangs - to the point I had to log out several times...

Going back to 2.0.9 removes the problem...

 

### Hardware Configuration

No

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE 1.8.10

### Operating System

Ubuntu 23.04

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

see above

### Sketch

```cpp
Blink with Builtin_LED defined
```

### Debug Message

```plain
see above
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-07-19:
Dupe #8424

VojtechBartoska, 2023-07-24:
Fix is included in new release https://github.com/espressif/arduino-esp32/releases/tag/2.0.11, please give it a try.

closed 2023-07-24 by VojtechBartoska

cmburns, 2023-10-21:
Hi, 

I used Sloeber IDE 4.4.1 and added an sketch with empty setup and loop funktion.
If I compile I get following error:

`C:/Sloeber441/arduinoPlugin/tools/make/make --no-print-directory main-build
'Building file: ..\sloeber.ino.cpp'
'Starting C++ compile'
"C:\Sloeber441\arduinoPlugin\packages\esp32\tools\xtensa-esp32s3-elf-gcc\esp-2021r2-patch5-8.4.0/bin/xtensa-esp32s3-elf-g++" -DHAVE_CONFIG_H "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -D_GNU_SOURCE -DIDF_VER="v4.4.6-dirty" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS   "-IC:\Sloeber441\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/newlib/platform_include" "-IC:\Sloeber441\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/freertos/include" "-IC:\Sloeber441\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/freertos/include/esp_additions/freertos" "-IC:\Sloeber441\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/freertos/port/xtensa/include" "-IC:\Sloeber441\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/freertos/include/esp_additions" "-IC:\Sloeber441\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/esp_hw_support/include" "-IC:\Sloeber441\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/esp_hw_support/include/soc" "-IC:\Sloeber441\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/esp_hw_support/include/soc/esp32s3" "-IC:\Sloeber441\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/esp_hw_support/port/esp32s3" "-IC:\Sloeber441\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/esp_hw_support/port/esp32s3/private_include" "-IC:\Sloeber441\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/heap/include" "-IC:\Sloeber441\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/log/include" "-IC:\Sloeber441\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/lwip/include/apps" "-IC:\Sloeber441\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/lwip/include/apps/sntp" "-IC:\Sloeber441\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/lwip/lwip/src/include" "-IC:\Sloeber441\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/lwip/port/esp32/include" "-IC:\Sloeber441\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/lwip/port/esp32/include/arch" "-IC:\Sloeber441\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/soc/include" "-IC:\Sloeber441\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/soc/esp32s3" "-IC:\Sloeber441\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/soc/esp32s3/include" "-IC:\Sloeber441\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/hal/esp32s3/include" "-IC:\Sloeber441\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/hal/include" "-IC:\Sloeber441\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/hal/platform_port/include" "-IC:\Sloeber441\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/esp_rom/include" "-IC:\Sloeber441\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/esp_rom/include/esp32s3" "-IC:\Sloeber441\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/esp_rom/esp32s3" "-IC:\Sloeber441\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/esp_common/include" "-IC:\Sloeber441\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/esp_system/include" "-IC:\Sloeber441\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/esp_system/port/soc" "-IC:\Sloeber441\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/esp_system/port/public_compat" "-IC:\Sloeber441\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/xtensa/include" "-IC:\Sloeber441\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/xtensa/esp32s3/include" "-IC:\Sloeber441\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/driver/include" "-IC:\Sloeber441\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/driver/esp32s3/include" "-IC:\Sloeber441\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/esp_pm/include" "-IC:\Sloeber441\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/esp_ringbuf/include" "-IC:\Sloeber441\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/efuse/include" "-IC:\Sloeber441\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/efuse/esp32s3/include" "-IC:\Sloeber441\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/vfs/include" "-IC:\Sloeber441\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/esp_wifi/include" "-IC:\Sloeber441\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/esp_event/include" "-IC:\Sloeber441\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/esp_netif/include" "-IC:\Sloeber441\arduinoPlugin\packages\ [... cut ...] aries\ESP32\src" -MMD -MP -MF"sloeber.ino.cpp.d" -MT"sloeber.ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "..\sloeber.ino.cpp"   -o "sloeber.ino.cpp.o"
xtensa-esp32s3-elf-g++: error: C:\Users\Thomas\Documents\sloeber441workspace\Test\Release/file_opts: No such file or directory
make[1]: *** [subdir.mk:20: sloeber.ino.cpp.o] Error 1
make: *** [makefile:97: all] Error 2
"C:/Sloeber441/arduinoPlugin/tools/make/make all" terminated with exit code 2. Build might be incomplete.`

**What I am missing please?**

me-no-dev, 2023-10-21:
This looks like Sloeber issue. Please ask there.

GitHubLionel, 2024-01-13:
Hi, I have the same problem with Sloeber.
Did you find a solution ?
I notice that the directive __IN_ECLIPSE__ is not defined in the sloeber.ino.cpp. 
Thank's,
Lionel

manmasel1975, 2024-02-29:
Hello, 
for the moment I have manually created the file "file_opts" into the folder "release". Then I compile the source and all is done.