espressif/arduino-esp32 #8431: Version 2.0.10 slows down wifi connect

jtkDvlp, 2023-07-19:
### Board

DOIT ESP32 DEVKIT V1

### Device Description

Nothing special

### Hardware Configuration

Nothing connected

### Version

v2.0.10

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

40 Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

Following code connects wifi after initial connect within about 200ms using lib version 2.0.9. With version 2.0.10 it took up to 1500ms.

### Sketch

```cpp
++
#include <WiFi.h>

void setup() {
  WiFi.setAutoConnect(false);
  WiFi.setAutoReconnect(false);

  Serial.begin(115200);
  
  long starttime = millis();
  Serial.println("Fast connect to WiFi...");
  WiFi.persistent(true);
  WiFi.config(IPAddress(XXX), IPAddress(XXX), IPAddress(XXX), IPAddress(XXX));
  WiFi.begin();
  if(WiFi.waitForConnectResult(5000) != WL_CONNECTED) {
    Serial.println("Fast connect failed");
    byte bssid[] = {XXX};
    WiFi.begin("XXX", "XXX", XXX, bssid);

    while (WiFi.status() != WL_CONNECTED) {
      delay(500);
      Serial.println("Connecting to WiFi..");
    }
  }

  Serial.print("Connected to the WiFi network in ");
  Serial.print(millis() - starttime);
  Serial.println("ms");

  Serial.println(WiFi.SSID());
  Serial.println(WiFi.BSSIDstr());
  Serial.println(WiFi.channel());
  Serial.println(WiFi.localIP());
  
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Debug Message

```plain
With Version 2.0.9:
20:38:27.165 -> Fast connect to WiFi...
20:38:32.330 -> Fast connect failed
20:38:32.862 -> Connecting to WiFi..
20:38:33.376 -> Connecting to WiFi..
20:38:33.376 -> Connected to the WiFi network in 6204ms
20:38:33.376 -> XXX
20:38:33.376 -> XXX
20:38:33.376 -> XXX
20:38:33.376 -> XXX

20:38:36.672 -> Fast connect to WiFi...
20:38:36.890 -> Connected to the WiFi network in 214ms
20:38:36.890 -> XXX
20:38:36.890 -> XXX
20:38:36.890 -> XXX
20:38:36.890 -> XXX

With Version 2.0.10
20:53:43.735 -> Fast connect to WiFi...
20:53:48.947 -> Fast connect failed
20:53:49.459 -> Connecting to WiFi..
20:53:49.974 -> Connecting to WiFi..
20:53:49.974 -> Connected to the WiFi network in 6202ms
20:53:49.974 -> XXX
20:53:49.974 -> XXX
20:53:49.974 -> XXX
20:53:49.974 -> XXX

20:53:55.699 -> Fast connect to WiFi...
20:53:56.742 -> Connected to the WiFi network in 1028ms
20:53:56.742 -> XXX
20:53:56.742 -> XXX
20:53:56.742 -> XXX
20:53:56.742 -> XXX
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

nemetila, 2023-07-30:
Hi,
I have similar problem: 
- update to Version 2.0.11: WIFI connect takes ~800ms
- downgrade to Version 2.0.8: WIFI connect takes ~200ms

It seems that releases from 2.0.10 changed the ESP-IDF version from v4.4.4 to v4.4.5.

XCIggo, 2023-08-24:
Hello, I have the same problem, and I did some tests for my understanding of versioning:

platformio.ini common:
board                   = esp32-s3-devkitc-1
framework               = espidf, arduino

CASE1:
platform                = espressif32
```
PACKAGES:
 - framework-arduinoespressif32 @ 3.20011.230801 (2.0.11)
 - framework-espidf @ 3.40405.230623 (4.4.5)
 - tool-cmake @ 3.16.4
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - tool-idf @ 1.0.1
 - tool-mconf @ 1.4060000.20190628 (406.0.0) 
 - tool-ninja @ 1.9.0
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
```
result:
```
@123:Connect: Persistent OK........[FastCo Ok 889Ms][Ping 309Ms]
@1321:Connect Success IP:192.168.10.83 [1198Ms]
```

CASE2:
platform                = https://github.com/platformio/platform-espressif32.git
platform_packages= framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git
```
PACKAGES:
 - framework-arduinoespressif32 @ 2.0.6+sha.e0657db
 - framework-espidf @ 3.40405.230623 (4.4.5) 
 - tool-cmake @ 3.16.4
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - tool-idf @ 1.0.1
 - tool-mconf @ 1.4060000.20190628 (406.0.0)
 - tool-ninja @ 1.9.0
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
```
result:
```
@122:Connect: Persistent OK........[FastCo Ok 873Ms][Ping 312Ms]
@1309:Connect Success IP:192.168.10.83 [1187Ms]
```

CASE3:
platform                = https://github.com/platformio/platform-espressif32.git
platform_packages =framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.11
```
PACKAGES:
 - framework-arduinoespressif32 @ 2.0.11+sha.ae9dae4
 - framework-espidf @ 3.40405.230623 (4.4.5)
 - tool-cmake @ 3.16.4
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - tool-idf @ 1.0.1
 - tool-mconf @ 1.4060000.20190628 (406.0.0)
 - tool-ninja @ 1.9.0
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
```
result:
```
@123:Connect: Persistent OK.........[FastCo Ok 923Ms][Ping 310Ms]
@1356:Connect Success IP:192.168.10.83 [1233Ms]
```

CASE4:
platform                = https://github.com/platformio/platform-espressif32.git
platform_packages =
    framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.9
    espressif/toolchain-xtensa-esp32s3 @ 8.4.0
    framework-espidf @ 3.40404

```
PACKAGES:
 - framework-arduinoespressif32 @ 2.0.9+sha.3670e2b
 - framework-espidf @ 3.40404.0 (4.4.4)
 - tool-cmake @ 3.16.4
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - tool-idf @ 1.0.1
 - tool-mconf @ 1.4060000.20190628 (406.0.0) 
 - tool-ninja @ 1.9.0
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
```
result:
```
@121:Connect: Persistent OK.[FastCo Ok 93Ms][Ping 308Ms]
@522:Connect Success IP:192.168.10.83 [401Ms]
```
CASE5:
    framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.11
    espressif/toolchain-xtensa-esp32s3 @ 8.4.0
    framework-espidf @ 3.40404
```
PACKAGES:
 - framework-arduinoespressif32 @ 2.0.11+sha.ae9dae4
 - framework-espidf @ 3.40404.0 (4.4.4)
 - tool-cmake @ 3.16.4
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - tool-idf @ 1.0.1
 - tool-mconf @ 1.4060000.20190628 (406.0.0)
 - tool-ninja @ 1.9.0
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
```

result:
```
@121:Connect: Persistent OK.[FastCo Ok 92Ms][Ping 309Ms]
@522:Connect Success IP:192.168.10.83 [401Ms]
```

I tried several other configurations, but it seems that the connection time is relative to framework-espidf from 4.4.4 to 4.4.5, while the change version of arduino-esp32 gives the same results.

I'm not sure how versioning works
question 1: why CASE1 return 2.0.11 (normal) and CASE2 return 2.0.6 (?!)
question 2: Is there a better parameter or method to maintain version consistency between framework-arduinoespressif32,espressif/toolchain-xtensa-esp32s3, and framework-espidf?

My partial conclusion, change from 2.0.9/4.4.4 (CASE4) to 2.0.11/4.4.5(CASE3) change connection time from 100Ms to 800-900Ms, force in 2.0.11/4.4.4 (CASE5) gives a connection at 100Ms. 

I am wrong?

SuGlider, 2023-09-04:
The time for 2.0.10/11 for Fast Connect with `WiFi.begin()` using the previous connection (NVS data) done to the same SSID is about 700~1000 ms. This is confirmed with the data presented in this issue.

### But I can't confirm that using 2.0.8/2.0.9, it can connect in 200ms or so, using WPA2.
I have found it takes about 3500 ms instead...

**What is the Wireless security encryption of the tested WiFi Network?**
My WLAN uses WPA2_PSK.

If you are using WPA_PSK, please set it with `WiFi.setMinSecurity(WIFI_AUTH_WPA_PSK);` and try it.

SuGlider, 2023-09-04:
I have changed my WiFi AP to WPA + Arduino 2.0.9 and I still can't get the Fast WiFi Connect to achieve 200ms.
It connects in 3.5 seconds.

SuGlider, 2023-09-04:
@jtkDvlp - I need more information in order to reproduce the issue by achieving the 200ms as here stated.

TD-er, 2023-09-04:
If your connect duration is like 3.5 seconds, I guess you include the WiFi scan time?
Have you tried also connecting directly to a specific BSSID at a specific channel?
Like when you already have the results of a WiFi scan and try to connect to a set of possible candidates.

XCIggo, 2023-09-04:
From my comment 2 week ago , the test are done on a WPA2 - AES (recommandé) network, with fast connect (persistent) in 200mS when ip, gateway , bssid,.. are known by a previous normal connect or scan (3-6seconds) . 

I have no short code to give, but the sketch example from [jtkDvlp](https://github.com/jtkDvlp) is good.

My persistent registering after scan is:
```
 WiFi.setHostname(DBSvc.SysDyn.strHOSTName); //define hostname   
        WiFi.setAutoReconnect(false);
        WiFi.mode(WIFI_STA);                             
        WiFi.config( DBSvc.SysDyn.NetworkSTA_IP, DBSvc.SysDyn.NetworkSTA_GATEWAY, DBSvc.SysDyn.NetworkSTA_SUBNET,  DBSvc.SysDyn.NetworkSTA_DNS1, DBSvc.SysDyn.NetworkSTA_DNS2 );  //DNS Google (8.8.8.8), CloudFlare (1.1.1.1)      
        WifiState=WiFi.begin(DBSvc.SysSettings.STAssid, DBSvc.SysSettings.STApassword, DBSvc.SysDyn.WifiSTA_Channel,  DBSvc.SysDyn.WifiSTA_BSSID);
        WiFi.persistent(true);          //set _persistent  
```

My persistent reconnect code sequence is:
```
        WiFi.setHostname(DBSvc.SysDyn.strHOSTName); //define hostname          
        WiFi.mode(WIFI_STA);        
        WiFi.config( DBSvc.SysDyn.NetworkSTA_IP, DBSvc.SysDyn.NetworkSTA_GATEWAY, DBSvc.SysDyn.NetworkSTA_SUBNET,  DBSvc.SysDyn.NetworkSTA_DNS1 , DBSvc.SysDyn.NetworkSTA_DNS2 );
        WifiState=WiFi.begin();    
```

benj919, 2024-01-17:
@SuGlider  / @VojtechBartoska 

A bit later, but in case this is helpful in general. I have struggled with the same slow connect for a day and observed vastly different results depending on which wifi network I try to connect my esp32 board to.

(Re-)connecting to my phone hotspot is ~150ms including acquiring an IP through dhcp. For my home wifi it always takes ~1500+ ms to connect, and another ~1500ms to acquire an ip address. Saving the bssid/channel to rtc_ram and restoring does not change this, neither does a fixed ip allocation in the dhcp server. A static ip on the esp obviously skips the dhcp time.

Board:
ESP32-WROOM-32 / esp32 dev-kit

Board Manager:
Espressif Systems / esp32: 2.0.14 (same ballpark results when downgrading to Arduino 2.0.9)

> WiFi.begin(ssid, password);

**2x Ubiquity UAP-AC-Pro, opnSense router**
<details>
  <summary>intial connect (once, with persistence): ~2000ms connect, ~1500ms dhcp</summary>

```
[   336][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   400][V][WiFiGeneric.cpp:340] _arduino_event_cb(): STA Started
[   401][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 2 - STA_START
[   420][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[  2395][V][WiFiGeneric.cpp:355] _arduino_event_cb(): STA Connected: [...], Auth: WPA2_WPA3_PSK
[  2397][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[  3914][V][WiFiGeneric.cpp:369] _arduino_event_cb(): STA Got New IP:[...]
[  3914][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
```

</details>

<details>
  <summary>Reconnect: ~1500ms connect, ~1500ms dhcp</summary>

```
[   337][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   350][V][WiFiGeneric.cpp:340] _arduino_event_cb(): STA Started
[   351][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[   351][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 2 - STA_START
[  1729][V][WiFiGeneric.cpp:355] _arduino_event_cb(): STA Connected: [...], Auth: WPA2_WPA3_PSK
[  1730][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[  3243][V][WiFiGeneric.cpp:369] _arduino_event_cb(): STA Got New [...]
[  3243][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
```

</details>

-----
**Fairphone 4 Hotspot**
<details>
  <summary>Initial connection (once, with persistence): 2200ms</summary>

  ```
  [   337][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 0 - WIFI_READY
  [   400][V][WiFiGeneric.cpp:340] _arduino_event_cb(): STA Started
  [   401][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 2 - STA_START
  [   421][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
  [  1338][V][WiFiGeneric.cpp:355] _arduino_event_cb(): STA Connected: [...], Auth: WPA2_PSK
  [  1340][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
  [  4552][V][WiFiGeneric.cpp:369] _arduino_event_cb(): STA Got New IP:[...]
  [  4552][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
  ```

</details>

<details>
  <summary>Reconnect: 150ms</summary>

```
[   338][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   351][V][WiFiGeneric.cpp:340] _arduino_event_cb(): STA Started
[   352][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[   352][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 2 - STA_START
[   407][V][WiFiGeneric.cpp:355] _arduino_event_cb(): STA Connected: [...], Auth: WPA2_PSK
[   409][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[   435][V][WiFiGeneric.cpp:369] _arduino_event_cb(): STA Got New IP: [...]
[   436][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
```

</details>

VojtechBartoska, 2024-02-20:
We can retest this when https://github.com/espressif/arduino-esp32/pull/8760 is merged, assigning to 3.0.0-rc1 milestone.

VojtechBartoska, 2024-04-12:
Hello, can you please validate this against 3.0.0-RC1 version? Thanks

VojtechBartoska, 2024-05-14:
Closing as solved, if needed, you can reopen.