espressif/arduino-esp32 #8434: SPI.setFrequency() not working, does not return

RudolphRiedel, 2023-07-20:
### Board

Seed XIAO ESP32-C3

### Device Description

Not relevant to the issue but display module with BT817 chip connected with one of these boards:
https://github.com/RudolphRiedel/EVE_display-adapter/tree/master/L-D5019-08

And I have a logic-analyzer connected in parallel, Saleae Logic16.

### Hardware Configuration

SPI:
SCK = 8
MOSI = 10
MISO = 9
CS = 5

### Version

v2.0.9

### IDE Name

PlatformIO / Arduino IDE 1.8.19

### Operating System

Windows 11

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

460800 / 921600

### Description

A call to SPI.setFrequency() does not change the frequency and never returns.

### Sketch

```cpp
Minimum sketch to reproduce the issue:
#include <SPI.h>

#define EVE_CS 5

void setup()
{
    pinMode(EVE_CS, OUTPUT);
    digitalWrite(EVE_CS, HIGH);

    SPI.begin();

    SPI.beginTransaction(SPISettings(8000000, MSBFIRST, SPI_MODE0));
    digitalWrite(EVE_CS, LOW);
    SPI.transfer(0x55);
    digitalWrite(EVE_CS, HIGH);
    SPI.endTransaction();

    SPI.beginTransaction(SPISettings(4000000, MSBFIRST, SPI_MODE0));
    digitalWrite(EVE_CS, LOW);
    SPI.transfer(0x55);
    digitalWrite(EVE_CS, HIGH);
    SPI.endTransaction();

    SPI.beginTransaction(SPISettings(8000000, MSBFIRST, SPI_MODE0));
    digitalWrite(EVE_CS, LOW);
    SPI.transfer(0x55);
    digitalWrite(EVE_CS, HIGH);

    digitalWrite(EVE_CS, LOW);
    SPI.setFrequency(4000000);
    digitalWrite(EVE_CS, HIGH);

    digitalWrite(EVE_CS, LOW);
    SPI.transfer(0xaa);
    digitalWrite(EVE_CS, HIGH);

    SPI.endTransaction();
}

Changing the frequency with SPI.beginTransaction() works fine.
But changing the frequency with SPI..setFrequency() after using SPI.beginTransaction() is not working.

The CS LOW / HIGH pair around SPI.setFrequency(4000000); is only there to check with the logic-analyzer that this is the part that hangs.
And it does, 0x55 is send, CS goes high, CS goes low again and then nothing else happens.
```

### Debug Message

```plain
Set the core debug level all the way up to "verbose" and there is nothing at all in the serial monitor.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

RudolphRiedel, 2023-07-20:
This is the logic-analyzer trace for the sketch:
[image]

me-no-dev, 2023-07-20:
This is expected. The transaction locked the bus. This in effect prevents other threads to change the frequency mid transfer.

RudolphRiedel, 2023-07-20:
And that makes it ok for the thread to hang indefinately if it tries to lock something that this very same thread already locked before?
And to spin this further, in case of multi-threading, using SPI.endTransaction() means that some other thread that previously got locked out can jump in and get the lock it is busy-waiting for and therefore locks out the first thread before it calls SPI.beginTransaction().
Well, if SPI.beginTransaction() had a return value, it could indicate that it just failed to get a lock but it does not.

Isn't there a way to tell that the semaphore was taken by the current thread?
I tried to hack something into spiSetClockDiv() in esp32-hal-spi.c but I have close to zero experience with FreeRTOS. 

Ok, at this point it is probably better for me to leave this alone and just not call SPI.setFrequency(). :-)

me-no-dev, 2023-07-20:
> Isn't there a way to tell that the semaphore was taken by the current thread?

There is and we used it very carefully in Wire already. I will be refactoring SPI soon anyway, so will see if that makes sense there.

> And to spin this further, in case of multi-threading, using SPI.endTransaction() means that some other thread that previously got locked out can jump in and get the lock it is busy-waiting for and therefore locks out the first thread before it calls SPI.beginTransaction().

That is exactly how it should behave in an OS environment (if the waiting thread is with higher priority). You are still safe, given that you release CS before you end the transaction. When the other thread is done, you will get it again and continue with the new frequency.

> Well, if SPI.beginTransaction() had a return value, it could indicate that it just failed to get a lock but it does not.

As explained, this would not make sense. Imagine that you have another thread that communicates with something else over SPI. If `beginTransaction` returns false, you would need to loop waiting to use it, but that looping might interrupt the other task (especially if it has lower priority) and you might not even get to it being able to release the bus, then things will really lock up.

Avoid `setFrequency` for now and make sure that you use transactions and that you end them properly. It is the proper way to do even on other Arduinos that do not run RTOS and you will be safe at the same time to explore the power of ESP and RTOS further.

RudolphRiedel, 2023-07-20:
Ok, thank you for the explanation! :-)
I can agree now that this is not actually an issue, I may not prefer the solution but it is a viable solution, it works fine.

closed 2023-07-20 by RudolphRiedel