espressif/arduino-esp32 #8436: AWS IoT Core not connecting when using with classic bluetooth.

ranjitlawand, 2023-07-20:
### Board

ESP32 Dev Module (4MB)

### Device Description

ESP32 Dev Module

### Hardware Configuration

#define RXD2 16
#define TXD2 17
#define RXD3 15
#define TXD3 13
#define RECIVE_ENABLE1 5
#define RECIVE_ENABLE2 4
#define WIFII_STATE 19
#define GREEN 32
#define RED 33
#define BLUE 25
#define WifiReset 18

### Version

v2.0.7

### IDE Name

PlatformIO

### Operating System

Windoes 11

### Flash frequency

240

### PSRAM enabled

no

### Upload speed

921600

### Description

I'm trying to connect to AWS IoT Core using ESP32. I also want to use Classic Bluetooth to communicate with the mobile phone. I have attached the sket for reference. 

Case 1:
When the wifi is connected and the AWS IoT connection is established before turning on the Classic Bluetooth core the ESP32 works till the Bluetooth connection is established post which ESP32 crashes. 
```
.
.
Connected to WiFi
Connecting to MQTT broker...
Connected to MQTT broker
Bluetooth initialized
The device started, now you can pair it with bluetooth
BT Callback Event: 0
BT Callback Event: 28

assert failed: vQueueDelete queue.c:2131 (pxQueue)

Backtrace: 0x40083695:0x3fff51d0 0x400930d1:0x3fff51f0 0x40098b3d:0x3fff5210 0x400944ee:0x3fff5340 0x40122d35:0x3fff5360 0x40121c74:0x3fff5380 0x40121cc2:0x3fff53a0 0x4010b497:0x3fff53c0 0x4010ce2a:0x3fff53f0 0x4010d094:0x3fff5410 0x4013f97d:0x3fff5450 0x4010e152:0x3fff54b0 0x4012288f:0x3fff54d0

ELF file SHA256: b384bb747de10cbb

Rebooting...
[... 263 lines cut ...]

Backtrace: 0x40083695:0x3fff51d0 0x400930d1:0x3fff51f0 0x40098b3d:0x3fff5210 0x400944ee:0x3fff5340 0x40122d35:0x3fff5360 0x40121c74:0x3fff5380 0x40121cc2:0x3fff53a0 0x4010b497:0x3fff53c0 0x4010ce2a:0x3fff53f0 0x4010d094:0x3fff5410 0x4013f97d:0x3fff5450 0x4010e152:0x3fff54b0 0x4012288f:0x3fff54d0

ELF file SHA256: b384bb747de10cbb

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
```

### Other Steps to Reproduce

The mentioned DynamicJsonDocument size is required to deserialize the data coming from other devices.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-07-20:
@ranjitlawand - I have tested it using Arduino Core 2.0.10, not the 2.0.7.
 
Right after connecting the ESP32 to my smartphone using the BT/SSP, I get an error related to lack of RAM and the WiFi is disconnected. 
It tries to connect to WiFi again but it fails.

I can send data using the Bluetooth from my smatphone to the ESP32 (`BT Callback Event: 30`), but the SSL can't recover...
```
[ 53795][E][ssl_client.cpp:37] _handle_error(): [start_ssl_client():264]: (-32512) SSL - Memory allocation failed
[ 53795][E][WiFiClientSecure.cpp:144] connect(): start_ssl_client: -32512
```
 
### Therefore, the issue is that the application is consuming too much HEAP and it fails.
### It may need some sort of "memory optimization"...

```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0
.
.
Connected to WiFi
Connecting to MQTT broker...
The client esp32-client-78:E3:6D:CD:06:A8 connects to the public mqtt broker
Connected to MQTT broker
[  3324][I][BluetoothSerial.cpp:683] _init_bt(): device name set

Bluetooth initialized
The device started, now you can pair it with bluetooth
[  3324][I][BluetoothSerial.cpp:263] esp_spp_cb(): ESP_SPP_INIT_EVT
[  3334][I][BluetoothSerial.cpp:270] esp_spp_cb(): ESP_SPP_INIT_EVT: slave: start
BT Callback Event: 0
[  3353][I][BluetoothSerial.cpp:571] esp_bt_gap_cb(): ESP_BT_GAP_CONFIG_EIR_DATA_EVT: stat:0 num:4
[  3354][I][BluetoothSerial.cpp:397] esp_spp_cb(): ESP_SPP_START_EVT
BT Callback Event: 28
[ 20967][I][BluetoothSerial.cpp:587] esp_bt_gap_cb(): ESP-BT_GAP_* unknown message: 16
[ 37041][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 200 - BEACON_TIMEOUT
[ 37047][E][ssl_client.cpp:37] _handle_error(): [send_ssl_data():382]: (-78) UNKNOWN ERROR CODE (004E)
.
.
.
.
.
[ 39458][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 201 - NO_AP_FOUND

The client esp32-client-78:E3:6D:CD:06:A8 connects to the public mqtt broker
[ 53795][E][ssl_client.cpp:37] _handle_error(): [start_ssl_client():264]: (-32512) SSL - Memory allocation failed
[ 53795][E][WiFiClientSecure.cpp:144] connect(): start_ssl_client: -32512
failed with state -2

BT Callback Event: 30
BT Callback Event: 30
BT Callback Event: 30
BT Callback Event: 30
BT Callback Event: 30

```

SuGlider, 2023-07-20:
You may try using some TCP/IP service instead of Classic Bluetooth SSP.... like HTTP for what ever your application may need it.

SuGlider, 2023-07-21:
@ranjitlawand - This information may be of your interest: When I use the ESP32-WROVER with PSRAM, everything works fine... Part of the WiFi and BT buffers are allocated automatically in the PSRAM, therefore, it doesn't fail.

`BT Callback Event: 30` message is when I send data from my smartphone to the ESP32-WROVER using BT.

Logs:
```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13924
ho 0 tail 12 room 4
load:0x40080400,len:3600
entry 0x400805f0
[   469][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
Starting WiFi...
.
.
.
.
.
.
.
[  5662][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 2 - AUTH_EXPIRE
Connected to WiFi
Connecting to MQTT broker...
The client esp32-client-78:E3:6D:CD:06:A8 connects to the public mqtt broker
Connected to MQTT broker
[ 11666][I][BluetoothSerial.cpp:571] esp_bt_gap_cb(): ESP_BT_GAP_CONFIG_EIR_DATA_EVT: stat:0 num:4
[ 11666][I][BluetoothSerial.cpp:683] _init_bt(): device name set
[ 11668][I][BluetoothSerial.cpp:263] esp_spp_cb(): ESP_SPP_INIT_EVT
Bluetooth initialized
[ 11676][I][BluetoothSerial.cpp:270] esp_spp_cb(): ESP_SPP_INIT_EVT: slave: start
The device started, now you can pair it with bluetooth
BT Callback Event: 0
[ 11701][I][BluetoothSerial.cpp:571] esp_bt_gap_cb(): ESP_BT_GAP_CONFIG_EIR_DATA_EVT: stat:0 num:4
[ 11701][I][BluetoothSerial.cpp:397] esp_spp_cb(): ESP_SPP_START_EVT
BT Callback Event: 28
[ 16645][I][BluetoothSerial.cpp:278] esp_spp_cb(): ESP_SPP_SRV_OPEN_EVT: 0
BT Callback Event: 34
BT Callback Event: 30
BT Callback Event: 30
BT Callback Event: 30
BT Callback Event: 30
BT Callback Event: 30
BT Callback Event: 30
BT Callback Event: 30
BT Callback Event: 30
[ 28763][I][BluetoothSerial.cpp:583] esp_bt_gap_cb(): ESP_BT_GAP_MODE_CHG_EVT: mode: 2
```

ranjitlawand, 2023-07-24:
Hello @SuGlider,

Thanks for the quick response. 

I do understand the code needs optimization but I do need bigger JSON buffers to deserialize and serialize JSONs. 

The current hardware which I'm using does not have PSRAM. Is it possible to get the setup working with the above code?

SuGlider, 2023-07-24:
>The current hardware which I'm using does not have PSRAM. Is it possible to get the setup working with the above code?

I tried the above code without PSRAM and it fails exactly as you said.

With PSRAM it runs fine. I tested it with a ESP32-WROVER-E that has 4MB SPRAM.
When the SoC has PSRAM, part of the `maloc()` (higher than 4KB block) calls will go to the PSRAM instead of HEAP.

VojtechBartoska, 2023-10-04:
Closing this as answered.

closed 2023-10-04 by VojtechBartoska