espressif/arduino-esp32 #8437: Whats wrong with these if i press the reset button on my esp 32 cam these happens on serial camera please help

Nyxpotato, 2023-07-20:
### Board

ESP 32 Wrover Module

### Device Description

[image]

### Hardware Configuration

esp 32 cam                ftdi
5v                              vcc
gnd                           gnd
uor                            tx
uot                            rx
gpio0 - gnd

### Version

v2.0.10

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

80mhz

### PSRAM enabled

yes

### Upload speed

256000

### Description

[image]
no errors on compiling
[image]
here is the upload
[image]
and this is my problem why is it like that in the serial monitor

### Sketch

```cpp
I just copy the code in github 
here it is

i will leave the ssid and wifi pass blank but i changed it too before uploading

#include <WebServer.h>
#include <WiFi.h>
#include <esp32cam.h>
 
const char* WIFI_SSID = "    ";
const char* WIFI_PASS = "   ";
 
WebServer server(80);
 
 
static auto loRes = esp32cam::Resolution::find(320, 240);
static auto midRes = esp32cam::Resolution::find(350, 530);
static auto hiRes = esp32cam::Resolution::find(800, 600);
void serveJpg()
{
  auto frame = esp32cam::capture();
  if (frame == nullptr) {
    Serial.println("CAPTURE FAIL");
    server.send(503, "", "");
    return;
  }
  Serial.printf("CAPTURE OK %dx%d %db\n", frame->getWidth(), frame->getHeight(),
                static_cast<int>(frame->size()));
 
  server.setContentLength(frame->size());
  server.send(200, "image/jpeg");
  WiFiClient client = server.client();
  frame->writeTo(client);
}
 
void handleJpgLo()
{
  if (!esp32cam::Camera.changeResolution(loRes)) {
    Serial.println("SET-LO-RES FAIL");
  }
  serveJpg();
}
 
void handleJpgHi()
{
  if (!esp32cam::Camera.changeResolution(hiRes)) {
    Serial.println("SET-HI-RES FAIL");
  }
  serveJpg();
}
 
void handleJpgMid()
{
  if (!esp32cam::Camera.changeResolution(midRes)) {
    Serial.println("SET-MID-RES FAIL");
  }
  serveJpg();
}
 
 
void  setup(){
  Serial.begin(115200);
  Serial.println();
  {
    using namespace esp32cam;
    Config cfg;
    cfg.setPins(pins::AiThinker);
    cfg.setResolution(hiRes);
    cfg.setBufferCount(2);
    cfg.setJpeg(80);
 
    bool ok = Camera.begin(cfg);
    Serial.println(ok ? "CAMERA OK" : "CAMERA FAIL");
  }
  WiFi.persistent(false);
  WiFi.mode(WIFI_STA);
  WiFi.begin(WIFI_SSID, WIFI_PASS);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
  }
  Serial.print("http://");
  Serial.println(WiFi.localIP());
  Serial.println("  /cam-lo.jpg");
  Serial.println("  /cam-hi.jpg");
  Serial.println("  /cam-mid.jpg");
 
  server.on("/cam-lo.jpg", handleJpgLo);
  server.on("/cam-hi.jpg", handleJpgHi);
  server.on("/cam-mid.jpg", handleJpgMid);
 
  server.begin();
}
 
void loop()
{
  server.handleClient();
}
```

### Debug Message

```plain
⸮R~⸮fgR⸮ggn⸮⸮⸮⸮bR⸮oo⸮⸮gj⸮ccCGC⸮⸮⸮⸮gr⸮⸮⸮c_k⸮⸮g⸮⸮⸮⸮W⸮g⸮⸮⸮GSnnn
what is this i just followed the tutorial in youtube for object detection but in my serial monitor this happens
```

### Other Steps to Reproduce

tried on windows 10 same process still like that

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-07-20:
Your baud rate in the monitor does not match what you declare in code.

SuGlider, 2023-07-20:
@Nyxpotato - As @lbernstone said, your code sets the baudrate to 115200:
``` cpp
void  setup(){
  Serial.begin(115200);
```

but your Serial Monitor is set to 9600. Change it to 115200 and it will work fine.

[image]

I'll close this issue.

closed 2023-07-20 by SuGlider

Nyxpotato, 2023-07-21:
Thank very much for information sir ill try it later , im a beginner for
this kind of stuff i appreciate your help

Nyxpotato, 2023-07-21:
it works thank you sir