espressif/arduino-esp32 #8439: camera PIXFORMAT_JPEG

hzhh110, 2023-07-21:
### Related area

ESP32-S3-EYE

### Hardware specification

camera PIXFORMAT_JPEG

### Is your feature request related to a problem?

ESP32-S3-EYE for PIXFORMAT_JPEG
pCodec = avcodec_find_decoder(AV_CODEC_ID_MJPEG);

### Describe the solution you'd like

ESP32-S3-EYE for PIXFORMAT_JPEG
pCodec = avcodec_find_decoder(AV_CODEC_ID_MJPEG);

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

SuGlider, 2023-07-21:
JPEG Encode: https://github.com/bitbank2/JPEGENC
JPEG Decoder: https://github.com/bitbank2/JPEGDEC

me-no-dev, 2023-07-21:
encoder and decoder are already bundled. Just check the CameraWebServer example. It does it al.

closed 2023-07-21 by me-no-dev

hzhh110, 2023-07-21:
PIXFORMAT_RGB565    than frame2jpg  Then use ffmpeg to parse out the standard YUV420, and directly use PIXFORMAT_JPEG and ffmpeg to parse out not the standard YUV420 may be YUV422 format

me-no-dev, 2023-07-21:
Sorry but I do not understand what you mean.

SuGlider, 2023-07-22:
> PIXFORMAT_RGB565    than frame2jpg  Then use ffmpeg to parse out the standard YUV420, and directly use PIXFORMAT_JPEG and ffmpeg to parse out not the standard YUV420 may be YUV422 format

This seems to be a (Linux) ffmpeg usage case.
You want ESP32 source code example that executes such sequence?

hzhh110, 2023-07-22:
[image]
[image]
[image]
static void YUV420PPackages(uint8_t *yuv420, uint8_t *y_ptr, uint8_t *u_ptr,
                uint8_t *v_ptr, int width, int height)
{
    //Y分量的长度
    int yLen = width * height;
    //U和V分量的长度
    int uvLen = width / 2 * height / 2;
    memcpy((unsigned char *)yuv420, y_ptr, yLen);
    memcpy((unsigned char *)yuv420 + yLen, u_ptr, uvLen);
    memcpy((unsigned char *)yuv420+yLen+uvLen, v_ptr, uvLen);
}

static void YUYV422Packages(uint8_t *yuv422, uint8_t *y_ptr, uint8_t *u_ptr, uint8_t *v_ptr, int width, int height)
{
    //计算出空间大小
    int numBytes = width * height + width / 2 * height * 2;
    unsigned int yIndex = 0;
    unsigned int uIndex = 0;
    unsigned int vIndex = 0;
    //封包
    for (unsigned int i = 0; i < numBytes; i+=4) {
        //y
        memcpy((unsigned char *)yuv422+i,y_ptr+yIndex,1);
        yIndex++;
        //u
        memcpy((unsigned char *)yuv422+i+1, u_ptr+uIndex, 1);
        uIndex++;
        //y
        memcpy((unsigned char *)yuv422+i+2, y_ptr+yIndex, 1);
        yIndex++;
        //v
        memcpy((unsigned char *)yuv422+i+3, v_ptr+vIndex, 1);
        vIndex++;
    }
}