espressif/arduino-esp32 #8442: Wifi scanNetworks() API

savagerex, 2023-07-21:
### Board

ESP32 S3

### Device Description

Sensor - IIS3DWB
Flash

### Hardware Configuration

NONE

### Version

v2.0.9

### IDE Name

Arduino

### Operating System

Windows 11

### Flash frequency

80M

### PSRAM enabled

no

### Upload speed

921600

### Description

if i change wifi SSID & Password.

then i use api (n = WiFi.scanNetworks())

n=-2, scan fail.

please help me

thanks.

### Sketch

```cpp
JsonObject root = jsonResult.to<JsonObject>();

          if (deviceConnected) {
                String result;
                jsonResult.clear();
                root["Version_Info"] = "Ver0.0";
                root["Command_Id"] = "GET_WIFI_LIST";
                root["Error_Code"] = "Success";

                JsonObject wifilist = root.createNestedObject("wifi_list");
                
                int n = WiFi.scanNetworks();

                Serial.print("n = ");
                Serial.println(n);
                
                if (n == 0) {
                    wifilist["ap_name"] = "no networks found";
                }else {
                    if ( n > 10)
                    {
                        n = 6;  
                    }
                    
                    for (int i = 0; i < n; ++i) {
                      String Name = "ap_name";
                      String ApName = Name + String(i);
                       wifilist[ApName] = WiFi.SSID(i);
                       String Signal = "signal_strength";
                       String SignalStrength = Signal + String(i);
                       wifilist[SignalStrength] = WiFi.RSSI(i);
                    }
                }
                
                serializeJson(root, result);

                result += "\n";
                pTxCharacteristic->setValue(result.c_str());
                pTxCharacteristic->notify();

          }
```

### Debug Message

```plain
please enter SensorId
[ 38652][D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 3] ... ESP_GATTS_WRITE_EVT
[ 38653][D][BLECharacteristic.cpp:297] handleGATTServerEvent():  - Response to write event: New value: handle: 2d, uuid: 6e400002-b5a3-f393-e0a9-e50e24dcca9e
[ 38666][D][BLECharacteristic.cpp:300] handleGATTServerEvent():  - Data: length: 22, data: 7b22576966694c697374223a224573703332227d0d0a
[ 38678][D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 3] ... ESP_GATTS_RESPONSE_EVT
Received message {"WifiList":"Esp32"} over Bluetooth
n = -2
[ 38700][D][BLECharacteristic.cpp:790] onNotify(): >> onNotify: default
[ 38700][D][BLECharacteristic.cpp:791] onNotify(): << onNotify
[ 38707][D][BLECharacteristic.cpp:796] onStatus(): >> onStatus: default
[ 38708][D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 3] ... ESP_GATTS_CONF_EVT
[ 38712][D][BLECharacteristic.cpp:797] onStatus(): << onStatus
Connecting to WiFi..
Connecting to WiFi..
please enter SensorId
[ 39725][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 39725][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 201 - NO_AP_FOUND
[ 39728][D][WiFiGeneric.cpp:1081] _eventCallback(): WiFi AutoReconnect Running
Connecting to WiFi..
Connecting to WiFi..
please enter SensorId
Connecting to WiFi..
Connecting to WiFi..
please enter SensorId
Connecting to WiFi..
Connecting to WiFi..
please enter SensorId
Connecting to WiFi..
[ 43113][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 43113][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 201 - NO_AP_FOUND
[ 43116][D][WiFiGeneric.cpp:1081] _eventCallback(): WiFi AutoReconnect Running
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-07-21:
@savagerex - I didn't understand the issue. Please provide more details.

savagerex, 2023-07-22:
@SuGlider 

After i change wifi SSID & Password.
then i use api (WiFi.scanNetworks())

scanNetworks will fail.

SuGlider, 2023-07-22:
Ok, could you please provide a complete minimal sketch that I can use to replicate the issue?

The code posted in this issue doesn't seem to help replicating it.

savagerex, 2023-07-22:
@SuGlider 

my code is very huge.

do you have email??

i send to you.

SuGlider, 2023-07-22:
Not sure that sending me the code would help... 
Let me suggest to erase the flash before uploading the sketch in order to erase NVS and any WiFi information.

[image]

savagerex, 2023-07-24:
@SuGlider 

After i enable "Erase All Flash Before Sketch Upload".
the result is same. ScanNetwork is still fail.

SuGlider, 2023-07-24:
The next possible reason for such error is that the Sketch is big and allocates/runs many libraries that the HEAP is not enough anymore to run ScanNetwork. In such case, when ScanNetwork tries to allocate memory to add a new entry, the allocation fails and the list of networks ais not completely shown.

I see that the project uses ESP32-S3 but PSRAM is disabled (based on the issue form information).

Does the S3 have PSRAM? Could you please turn PSRAM on and test it?

everslick, 2023-07-24:
IMHO, this is a known issue.

When `WiFi.scanNetworks()` returns `WIFI_SCAN_FAILED` you have to call `Wifi.disconnect()` and then  call `WiFi.scanNetworks()` again. Then you can call `WiFi.reconnect()`.

savagerex, 2023-07-24:
OK, Thanks

savagerex, 2023-07-25:
another question

when i use WiFi.scanNetworks() API.

i will get repeat AP.

is this normal???

everslick, 2023-07-25:
yes, you have to remove duplicates, if they bother you.

I would:

A) sort all found SSIDd by signal strength, then by security (encrypted vs. open), then by name.
B) remove duplicated SSIDs.

savagerex, 2023-07-25:
OK, Thanks

everslick, 2023-07-27:
Please close the issue, if your problem is solved. Thanks!

VojtechBartoska, 2023-08-21:
Closing as solved, if needed you can reopen. Thanks.

closed 2023-08-21 by VojtechBartoska