espressif/arduino-esp32 #8443: Implement remaining USBHID functions from Atmega32U4

kilofeq, 2023-07-21:
### Related area

USB HID

### Hardware specification

ESP32S2/ESP32S3

### Is your feature request related to a problem?

I cannot make FFB controller port from Atmega 32U4, because of lack of functions.

### Describe the solution you'd like

Create remaining functions for USBHID library

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

lbernstone, 2023-07-21:
usb hid features come upstream from https://github.com/hathach/tinyusb/.  Cross post there, and explain exactly what functions are missing.

kilofeq, 2023-07-22:
Hi! I need to send HID reports (done), but I would also like to implement PID, so data would be sent from computer about some forces send to HID device. Something like this: https://github.com/hoantv/VNWheel/blob/master/WHID.cpp

closed 2025-01-15 by Parsaabasi