espressif/arduino-esp32 #8445: [EEPROM.cpp:70] begin(): Unable to open NVS namespace: 4353

murarduino, 2023-07-22:
### Board

ESP32-PICO-DevKitM-2

### Device Description

分区：partitions.csv
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x100000,
spiffs,   data, spiffs,  0x110000,0x6F0000,

### Hardware Configuration

无

### Version

latest master (checkout manually)

### IDE Name

VScode+ESP-IDF+arduino-esp32库（组件方式）

### Operating System

windows10

### Flash frequency

40MHZ

### PSRAM enabled

no

### Upload speed

406800

### Description

使用Arduino-ESP32的EEPROM库，尝试读写EEPROM数据（uint8_t）

### Sketch

```cpp
#include "EEPROM.h"
 #include <esp_spi_flash.h>
EEPROM.begin(SPI_FLASH_SEC_SIZE);
```

### Debug Message

```plain
I级日志：
[    70][E][EEPROM.cpp:70] begin(): Unable to open NVS namespace: 4353
V级日志：
D (1449) nvs: nvs_open_from_partition eeprom 1
[   349][E][EEPROM.cpp:70] begin(): Unable to open NVS namespace: 4353
```

### Other Steps to Reproduce

尝试修改EEPROM大小。改为
EEPROM.begin(512);
问题依旧。

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-07-23:
Does the [eeprom_write example](https://github.com/espressif/arduino-esp32/blob/master/libraries/EEPROM/examples/eeprom_write/eeprom_write.ino) work for you?

murarduino, 2023-07-24:
Yes, I replaced ESPDuino32 and used the borrowed eeprom_write.ino code:

`#include "EEPROM.h"

int addr = 0;
#define EEPROM_SIZE 64
void setup()
{
  Serial.begin(115200);
  Serial.println("start...");
  if (!EEPROM.begin(EEPROM_SIZE))
  {
    Serial.println("failed to initialise EEPROM"); while(1){delay(10);｝;
  }
....`

Log:
“failed to initialise EEPROM”
[EEPROM.cpp:70] begin(): Unable to open NVS namespace: 4353

SuGlider, 2023-07-24:
I teste EEPROM with Arduino Core 2.0.9 and Arduino IDE 1.8.15,
The Example sketch and the code from above execute with no error.

I think that the used Arduino Development Environment with VSCode + IDF + Arduino ESP32 is not correctly set. 

Please test it using the Arduino IDE with Arduino Core 2.0.9 in order to see it working.

murarduino, 2023-07-24:
I looked at the EEPROM.cpp source code and found it in bool EEPROMClass::begin(). The nvs_flash_init() function is not called, is that correct?

SuGlider, 2023-07-24:
> I looked at the EEPROM.cpp source code and found it in bool EEPROMClass::begin(). The nvs_flash_init() function is not called, is that correct?

Yes, because it is excuted in `void initArduino()`, before the sktech is started.
EEPROM doesn't need to call it.

SuGlider, 2023-07-24:
`initArduino()` is called from the Arduino `app_main()` - only when `CONFIG_AUTOSTART_ARDUINO` is set.
I think that your Arduino environment may miss that setting.

murarduino, 2023-07-24:
I seem to find the reason why my code is based on the IDF architecture (mainly borrowing some sdkconfig configuration functionality). There are other needs). In my code, there is no call to initArduino(), so I don't want to be restricted by the Arduino framework. I just want to call some of the arduino-esp32 configured libraries if needed to reduce some of the underlying functionality rewriting.

murarduino, 2023-07-24:
but, I included arduino.h in the .h files?

murarduino, 2023-07-25:
> `initArduino()` is called from the Arduino `app_main()` - only when `CONFIG_AUTOSTART_ARDUINO` is set. I think that your Arduino environment may miss that setting.

Thanks for your guidance, I first call initArduino() in 'app_main()'; Problem solved.

closed 2023-07-25 by murarduino