espressif/arduino-esp32 #8448: BTClassic

BlackWolfDEsign, 2023-07-23:
### Board

ESP32 Dev Module

### Device Description

nothing

### Hardware Configuration

only BT-connection

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Win 10

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

in my loop(): I want to discover() available clients. after that, i want to connect() to them and after that i want to repeat this.

If I use the discover()-part only, it shows all available clients, works as expected.... 

If i use connect("DEVICE_NAME") only, it works perfectly. Also with multiple clients....

If I use both, the discover()-function works on the first pass, but after connect(DEVICE_NAME), the discover()-function is broken... (I don't get any "active" devices anymore) so I tried so many different approaches, but I didn't find a solution....

If I put the discover()-part in setup(), everything works fine, but i want to update the list of active clients, so i put this part in loop().

a connect() after discover() will stop discover()!?!

### Sketch

```cpp
// BT
#include "BluetoothSerial.h"
BluetoothSerial SerialBT;
BluetoothSerial SerialBT2; // this was only a desperate test...

#include "ListLib.h"
List<String> list;

void setup() {
  Serial.begin(9600);
}

void loop() {
  SerialBT.begin("MASTER", true); // normally in setup(), i know...
  SerialBT.discover(BT_DISCOVER_TIME);
  BTScanResults* pScanResults = SerialBT.getScanResults();
  Serial.println(pScanResults->getCount());

  for (int i = 0; i < pScanResults->getCount(); i++) {
    BTAdvertisedDevice* pDevice = pScanResults->getDevice(i);
    String deviceName = pDevice->getName().c_str();
    if (deviceName.startsWith(SLAVE_NAME_PATTERN)) {
        delay(200);
        list.Add(deviceName);
        Serial.println(list[i]);
    }
  }
  SerialBT.discoverClear();
  SerialBT.disconnect();
  SerialBT.end();

  SerialBT2.begin("MASTER", true);
  for (int i = 0; i < list.Count(); i++) {
    SerialBT2.connect(list[i]);
    SerialBT2.println("REQUEST");
    delay(200);
    Serial.println(SerialBT2.readStringUntil('\n'));
  }
  SerialBT2.disconnect();
  SerialBT2.end();
  list.Clear();
}
```

### Debug Message

```plain
none? ;/
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-08-04:
@PilnyTomas PTAL, thanks!

PilnyTomas, 2023-08-04:
Hi, I have slightly modified your code and added a few debug outputs and it is working. I'm using a second ESP32 with the example `SerialToSerialBT` with the name changed, to begin with `SLAVE` as expected by your code.
Can you please test my code to see if it works for you?
```
#include "BluetoothSerial.h"
BluetoothSerial SerialBT;
BluetoothSerial SerialBT2; // this was only a desperate test...

#include "ListLib.h"
List<String> list;

#define BT_DISCOVER_TIME 10000

void setup() {
  Serial.begin(115200);
}

void loop() {
  SerialBT.begin("MASTER", true); // normally in setup(), i know...
  SerialBT.discover(BT_DISCOVER_TIME);
  BTScanResults* pScanResults = SerialBT.getScanResults();
  Serial.println(pScanResults->getCount());

  Serial.printf("DEBUG: got %d results\n", pScanResults->getCount());
  for (int i = 0; i < pScanResults->getCount(); i++) {
    BTAdvertisedDevice* pDevice = pScanResults->getDevice(i);
    String deviceName = pDevice->getName().c_str();
    if (deviceName.startsWith("SLAVE")) {
        list.Add(deviceName);
        Serial.printf("DEBUG: Added to list \"%s\"\n", list[i].c_str());
    }
  }
  SerialBT.discoverClear();
  SerialBT.disconnect();
  SerialBT.end();

  SerialBT2.begin("MASTER", true);
  for (int i = 0; i < list.Count(); i++) {
    Serial.printf("DEBUG: Connecting to [%d] \"%s\"\n", i, list[i].c_str());
    if(SerialBT2.connect(list[i])){
      Serial.printf("DEBUG: Connected\n");
      SerialBT2.println("REQUEST");
      Serial.println(SerialBT2.readStringUntil('\n'));
      SerialBT2.disconnect();
    }else{
      Serial.printf("DEBUG: Could not connect\n");
    }
  }
  SerialBT2.end();
  list.Clear();
  delay(10); // Feed the watchdog, otherwise will crash with "st:0x7 (TG0WDT_SYS_RESET)"
}
```

BlackWolfDEsign, 2023-08-07:
21:15:37.686 -> DEBUG: got 1 results
21:15:37.717 -> DEBUG: Added to list "POD_SLAVE_A0B765596A92"
21:15:38.282 -> DEBUG: Connecting to [0] "POD_SLAVE_A0B765596A92"
21:15:39.440 -> DEBUG: Connected
21:15:40.440 -> 
21:15:41.881 -> 0
21:15:41.881 -> DEBUG: got 0 results

without requesting it is working D:

PilnyTomas, 2023-08-15:
Ok, the discover is failing on the first test:
https://github.com/espressif/arduino-esp32/blob/099b432d10fb4ca1529c52241bcadcb8a4386f17/libraries/BluetoothSerial/src/BluetoothSerial.cpp#L1095
[E][BluetoothSerial.cpp:1173] discover(): Failed if (timeoutMs(10000) < MIN_INQ_TIME(1280) || timeoutMs(10000) > MAX_INQ_TIME(61440) || **strlen(_remote_name)(14)** || **_isRemoteAddressSet(1)**){

closed 2025-01-15 by Parsaabasi