espressif/arduino-esp32 #8449: ESP32C3 : wake up from deep sleep impacted by initial power ON

spilz87, 2023-07-23:
### Board

ESP32-C3

### Device Description

different custom boards with ESP32C3

### Hardware Configuration

rotary encoder connected to GPIO 0 and 1 with 100k PULLUP resistor connected to 3.3v

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

The rotary encoder has 2 kinds of stable positions : [HIGH-HIGH] and [LOW-LOW].
I want to use the rotary encoder to wake up the chip, send data and go back to deep sleep.

I identify this weird patern : 
- when I power ON the board, if the rotary encoder is on [HIGH-HIGH], everything goes well, I'm able to enter in deep sleep mode and to wake up from GPIO (high or low, depends on the position of the encoder just before enter in deep sleep mode.

- when I power ON the board, if the rotary encoder is on [LOW-LOW], I'm not able to wake up from GPIO 0 or 1.

Powering ON or Using EN pin has the same effect here.

Consuption analyse : 
- When waking up works well : 
In deep sleep mode with encoder on [HIGH-HIGH] -> consumption is 80uA
In deep sleep mode with encoder on [LOW-LOW] -> consumption is 300uA
- When waking up doesn't work : 
in deep sleep mode, the consumption is 150uA, what ever if I change the position of the encoder.

this issue can be linked to #8447 

### Sketch

```cpp
void setup() {

  pinMode(0, INPUT);
  pinMode(1, INPUT);

  delay(1000);
  if(digitalRead(0))
    esp_deep_sleep_enable_gpio_wakeup(1<<0, ESP_GPIO_WAKEUP_GPIO_LOW);
  else
    esp_deep_sleep_enable_gpio_wakeup(1<<0, ESP_GPIO_WAKEUP_GPIO_HIGH);
  if(digitalRead(1))
    esp_deep_sleep_enable_gpio_wakeup(1<<1, ESP_GPIO_WAKEUP_GPIO_LOW);
  else
    esp_deep_sleep_enable_gpio_wakeup(1<<1, ESP_GPIO_WAKEUP_GPIO_HIGH);
  
  gpio_set_direction(GPIO_NUM_0, GPIO_MODE_INPUT);
  gpio_set_direction(GPIO_NUM_1, GPIO_MODE_INPUT);
  delay(500);
  esp_deep_sleep_start();

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Debug Message

```plain
no debug message
```

### Other Steps to Reproduce

I found a way to solve it by forcing esp_deep_sleep_enable_gpio_wakeup ESP_GPIO_WAKEUP_GPIO_LOW to low the first time its powered (or EN is used) 

but I think there is still something wrong in the initialisation of the chip.

`void setup() {

  esp_sleep_wakeup_cause_t wakeup_reason = esp_sleep_get_wakeup_cause();

  pinMode(0, INPUT);
  pinMode(1, INPUT);

  delay(1000);
  if(wakeup_reason != ESP_SLEEP_WAKEUP_GPIO){
    esp_deep_sleep_enable_gpio_wakeup(1<<0, ESP_GPIO_WAKEUP_GPIO_LOW);
    esp_deep_sleep_enable_gpio_wakeup(1<<1, ESP_GPIO_WAKEUP_GPIO_LOW);
  }
  else{
    if(digitalRead(0))
      esp_deep_sleep_enable_gpio_wakeup(1<<0, ESP_GPIO_WAKEUP_GPIO_LOW);
    else
      esp_deep_sleep_enable_gpio_wakeup(1<<0, ESP_GPIO_WAKEUP_GPIO_HIGH);
    if(digitalRead(1))
      esp_deep_sleep_enable_gpio_wakeup(1<<1, ESP_GPIO_WAKEUP_GPIO_LOW);
    else
      esp_deep_sleep_enable_gpio_wakeup(1<<1, ESP_GPIO_WAKEUP_GPIO_HIGH);
  }
  gpio_set_direction(GPIO_NUM_0, GPIO_MODE_INPUT);
  gpio_set_direction(GPIO_NUM_1, GPIO_MODE_INPUT);
  delay(500);
  esp_deep_sleep_start();

}

void loop() {
  // put your main code here, to run repeatedly:

}
`

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-08-03:
@spilz87 - It misses one code line:

``` cpp
  esp_deep_sleep_enable_gpio_wakeup(1 << INTERRUPT_PIN, ESP_GPIO_WAKEUP_GPIO_HIGH);
  gpio_set_direction((gpio_num_t)INTERRUPT_PIN, GPIO_MODE_INPUT);  // <<<<=== Add this 
  esp_deep_sleep_start();
```

SuGlider, 2023-08-03:
> @SuGlider thanks for your reply
> 
> What is the difference between : `gpio_set_direction(GPIO_NUM_1, GPIO_MODE_INPUT);` and `gpio_set_direction((gpio_num_t)INTERRUPT_PIN, GPIO_MODE_INPUT); // <<<<=== Add this `
> 
> ?
> 
> thanks

None. I think that I wrote this comment in the wrong issue... It was suppose to be in #8484 instead.

SuGlider, 2023-08-03:
@spilz87 - I still think that you should try removing the pullup resistors... Why do you need them?
GPIO 0 and 1 are always reading Levels only. The Encoder will set the GPIO level.

SuGlider, 2023-08-03:
@spilz87 - I tested this sketch using a breadboard and wires... No Resistors.
I worked as expected.

VojtechBartoska, 2023-08-18:
Closing this as answered, if needed you can reopen. Thanks for testing @SuGlider.

closed 2023-08-18 by VojtechBartoska

spilz87, 2023-08-18:
I still need to test before saying it's solved

> Closing this as answered, if needed you can reopen. Thanks for testing @SuGlider.

I'm still not confortable having no pullup nor pulldown resistor for a rotary encoder as suggested by @SuGlider 
I never saw it before

VojtechBartoska, 2023-08-18:
Ok, hopefully @SuGlider can help with this.

spilz87, 2023-08-18:
After tests : 

- as suspected, the suggestion of @SuGlider doesn't work with a rotary encoder : the level of rotary encoder changes too quickly. Without pullup or pulldown resistor, the rotary encoder is no more reliable.

With the example given in first post : 
- With Version 2.0.9 : 
the issue is present as described, but the work around I proposed works.

- With Version 2.0.10 and 2.0.11 : 
it becomes worth : depends on the level of GPIO when enter in deep sleep mode, it never wakes up anymore.
description : 
GPIO are connected to 3.3V with a 100k resistor and shorcutted to ground by rotary encoder, (so LOW level before enter in deep sleep mode).
Enter in deep sleep mode.
when I turn the rotary encoder (shortcut removed), the tension of GPIO become 0.5V instead of 3.3V, and never exit sleeping mode.
the work around doesn't work anymore :(

Conclusion : 
- the issue is still present
- something change about deep sleep between 2.0.9 and 2.0.10/2.0.11 that makes it even worth.

 @VojtechBartoska  Please re-open the issue

spilz87, 2023-08-18:
hypotesis : 
the difference between 2.0.9 and 2.0.10/11 which impact us here is linked to internal pullup/pulldown resistor when enter in deep sleep mode.

the above description lets me think the internal pulldown resistor is activated.
that makes sense if internal resistor are around 10k -> 3.3V * 10k / (10k + 100k) = 0.3V which is close to the 0.5V I mesured.

spilz87, 2023-08-18:
to help with a diagram : 
[image]