espressif/arduino-esp32 #8451: Arduino Support for WPA3 Authentication in AP mode

brindhakumar, 2023-07-24:
### Related area

WiFi

### Hardware specification

ESP32S2 Dev Module

### Is your feature request related to a problem?

I am facing a issue with configuring WPA3 in Access point Mode.
After the configuration I found that SAE Support is not there in the access point mode. It works with Station mode.
But configuring WPA3_AUTH_MODE in Access point will not work. 

### Describe the solution you'd like

When using the function  WiFi.softAP() with Authentication configuration I need to enable WPA3 Authentication with SAE Support

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

Rotzbua, 2023-08-22:
Current IDF4.4 does not support WPA3 AP.
Earliest support is IDF5.1. https://github.com/espressif/esp-idf/issues/10939
So take a ☕ until it is time.

VojtechBartoska, 2024-02-20:
Possibly related to https://github.com/espressif/arduino-esp32/pull/8760

closed 2024-04-04 by me-no-dev