espressif/arduino-esp32 #8454: ESP32C3 Wire I2C - SCL hang and 1 second timeout on transfer performed after NAK

l0ud, 2023-07-24:
### Board

WeAct ESP32-C3FH4

### Device Description

WeAct ESP32-C3FH4 with only BMI160 sensor wired in. WeAct board is very basic containing just ESP32C3FH4 chip, voltage regulator USB port and some buttons.

### Hardware Configuration

BMI160 IMU wired using i2c to ESP32C3 4 and 5 pin.

### Version

v2.0.6

### IDE Name

PlatformIO

### Operating System

Manjaro Linux

### Flash frequency

Default

### PSRAM enabled

no

### Upload speed

Default

### Description

Scenario is the following:
- I'm writing 0 bytes (only address) to device 0x68 to check if it's available. It responds with ACK and I get success status, all good here.
- I'm writing address to device 0x69 to check if it's available. It's not, I have NACK status, all good.
- I'm writing again 0 bytes to device 0x68 expecting it to succeed. However it fails with timeout.

On screenshot I'm showing overview of first 3 transfers on i2c bus:
- successful write to 0x68 (only address written, then ACK). Transfer reported as successful.
- failed write to 0x69 (no ACK after address, as expected). This transfer returns failure, as expected.
- successful write to 0x68 (with ACK) with SCL line stuck low for one second! This transfer returns timeout error.
[image]
BMI160 doesn't support clock stretching, so SCL stuck low is probably ESP fault.

Zoom in to last transfer:
[image]
And recovery after one second (I'm starting another transfer to 0x69):
[image]

### Sketch

```cpp
byte status;
while(1) {
Wire.beginTransmission(0x68);
status = Wire.endTransmission();
Serial.printf("Address 0x68 status %d\n", addr, error);
Wire.beginTransmission(0x69);
status = Wire.endTransmission();
Serial.printf("Address 0x69 status %d\n", addr, error);
}
```

### Debug Message

```plain
Address 0x68 status 0 [OK]
Address 0x69 status 2 [NACK, OK]
Address 0x68 status 5 [timeout, not ok, should be 0]
Address 0x69 status 2 [NACK, OK]
...
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-08-04:
@me-no-dev PTAL, thanks!

l0ud, 2023-08-23:
I looked around and now I'm wondering, is that not related to https://github.com/espressif/esp-idf/issues/9777 ?
I looked at files I have locally and it seems like my sources doesn't have that patch included.

VojtechBartoska, 2023-08-23:
Closing as solved, thanks @me-no-dev for the info.