espressif/arduino-esp32 #8457: ESP32 Arduino Core SD Library is ignoring custom pin definition

momonator25, 2023-07-25:
### Board

probably all ESP32 Boards?

### Device Description

- ESP32 Lolin C3 Pico Board
- Micro SD Breakout 

### Hardware Configuration

- MISO: 0
- SCK: 1
- MOSI: 4
- CS: 2

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

Windows 11

### Flash frequency

40MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

When using a ESP32 board that supports Hardware SPI on different Pins the setup of the the SD Card library is ignoring the previous `spi.begin(SD_SCK, SD_MISO, SD_MOSI, SD_CS);`

In the `.platformio\packages\framework-arduinoespressif32\libraries\SD\src\SD.cpp`  `begin` function the call of `spi.begin();` is overwriting the setup so the default SPI pins are used in every case.

Possible solution: 
Add optional SPI Pins to the `begin` function in `SD.cpp` (will break std. Arduino compatibility):
```cpp
bool SDFS::begin(uint8_t sckPin, uint8_t misoPin, uint8_t mosiPin, uint8_t ssPin, SPIClass &spi, uint32_t frequency, const char * mountpoint, uint8_t max_files, bool format_if_empty)
{
    if(_pdrv != 0xFF) {
        return true;
    }

    // spi.begin();
    spi.begin(sckPin, misoPin, mosiPin, ssPin);
   ...
}
```

and change of function definition in `SD.h` to use standard SPI pins as default

```h
bool begin(uint8_t sckPin=SCK, uint8_t misoPin=MISO, uint8_t mosiPin=MOSI, uint8_t ssPin=SS, SPIClass &spi=SPI, uint32_t frequency=4000000, const char * mountpoint="/sd", uint8_t max_files=5, bool format_if_empty=false);
```

and then call
```cpp
SD.begin(SD_SCK, SD_MISO, SD_MOSI, SD_CS)
```

### Sketch

```cpp
The default __SD_Test.ino__ example
```

### Debug Message

```plain
[  7770][E][sd_diskio.cpp:802] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[  8270][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
Card Mount Failed
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-07-25:
Can you please provide a minimal example to show your code?  SD simply uses the existing spi definition.  If you don't use the default (SPI), then you need to pass that spi object to SD::begin.
I don't have an SD module handy, but this code is giving me whatever I pass into spi.begin():
```
#include "FS.h"
#include "SD.h"
#include "SPI.h"

void setup() {
  Serial.begin(115200);
  SPI.begin(1,0,4,2);
  SD.begin(2);
  Serial.println(SPI.pinSS());
}

void loop() {}
```

momonator25, 2023-07-27:
> Can you please provide a minimal example to show your code? SD simply uses the existing spi definition. If you don't use the default (SPI), then you need to pass that spi object to SD::begin. I [...]

MWE:
```cpp
#include <Arduino.h>
#include "FS.h"
#include "SD.h"
#include "SPI.h"

#define SD_SCK 14
#define SD_MISO 2
#define SD_MOSI 15
#define SD_CS 13

SPIClass myspi = SPIClass(HSPI);

void setup()
{
  Serial.begin(115200);
  myspi.begin(SD_SCK, SD_MISO, SD_MOSI, SD_CS);
  if (!SD.begin(SD_CS, myspi))
  {
    Serial.println("Card Mount Failed");
    return;
  }
}
```

Sadly it's not working as expected and is ignoring the spi init

closed 2023-07-27 by momonator25