espressif/arduino-esp32 #8463: ESP32 Connect External flash

savagerex, 2023-07-27:
### Board

ESP32 S3

### Device Description

Sensor - IIS3DWB
external flash

### Hardware Configuration

NONE

### Version

v2.0.9

### IDE Name

Arduino

### Operating System

Windows 11

### Flash frequency

80M

### PSRAM enabled

no

### Upload speed

921600

### Description

I use ESP32 to connect external Flash.
I already fomrat flash.

After test about thirty minutes.
there is an error (open file error).

debug message as below.
"There was an error opening the file for writing" 

please help me to solve this problem.

### Sketch

```cpp
#include "SPI.h"
#include <SerialFlash.h>
#include "BoardConfig.h"
#include "Sensor.h"
#include "IIS3DWB.h"
#include <WiFi.h>
#include <MQTT.h>

const char* ssid = "ASUS-38";     //SW1-WIFI6   ASUS-38   vivo Y21 
const char* pass = "12345678";   //wireless    12345678   1234567890

const char* mqtt_server = "192.168.50.129";   //broker.emqx.io  192.168.50.128
//const char *topic = "DataTopic";
//const char *topic = "1/report";
const char *mqtt_username = "emqx";
const char *mqtt_password = "public";

[... 471 lines cut ...]
                Serial.println("Task2 Going to sleep now");
                delay(100);
                esp_deep_sleep_start();
              }
            
        }
   }
   vTaskDelete( NULL );
}

void loop() {
  
}
```

### Debug Message

```plain
debug message as below.
"There was an error opening the file for writing"
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-07-27:
@savagerex - This seems to be a problem with the `SerialFlash` library. 
Maybe someone from the community may help with it.

VojtechBartoska, 2023-08-18:
Seems we are not able to help here, closing as won't do. If needed, you can reopen. Thanks for understanding.

closed 2023-08-18 by VojtechBartoska