espressif/arduino-esp32 #8469: esp32c3 Serial1 hardweired to pin 18/19 (USB Pins)

mcenderdragon, 2023-07-29:
### Board

ESP32-c3-WROOM

### Device Description

plain module, on selfmade PCB.

### Hardware Configuration

gpio18 & 19 are connected to usb gpio10 is for RX1

### Version

v2.0.4

### IDE Name

PLatfromIO

### Operating System

Win 10

### Flash frequency

N/A

### PSRAM enabled

yes

### Upload speed

115200

### Description

https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/HardwareSerial.cpp#L55

https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/HardwareSerial.cpp#L355

Passed arguments are ignored and the defied RX1 and TX1 pins are sued instead.

only workoutaround is to define RX1 and TX1 before including the hardware serial, which si bad design.

### Sketch

```cpp
SerialGPS.begin(9600, SERIAL_8N1, 10, 0);
```

### Debug Message

```plain
11:50:01.330 > ESP-ROM:esp32c3-api1-20210207
11:50:01.376 > Build:Feb  7 2021
11:50:01.381 > rst:0x1 (POWERON),boot:0x9 (SPI_FAST_FLASH_BOOT)
11:50:01.381 > SPIWP:0xee
11:50:01.381 > mode:DIO, clock div:1
11:50:01.381 > load:0x3fcd6100,len:0x438
11:50:01.381 > load:0x403ce000,len:0x918
11:50:01.381 > load:0x403d0000,len:0x24e4
11:50:01.381 > SHA-256 comparison failed:
11:50:01.381 > Calculated: 080c5cb68a075ced55f248b97bca965e3e5bd5da80a64e34e6a1638f89d6f64e
11:50:01.381 > Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff  
11:50:01.381 > Attempting to boot anyway...
11:50:01.381 > entry 0x403ce000
```

### Other Steps to Reproduce

just read the code, the flaw is obvious, no need to provide example code.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-07-30:
@mcenderdragon - Not sure about what you are considering an issue here.
I understand that you want to set just the SerialGPS RX pin and not the TX pin?

for that just use: 
`SerialGPS.begin(9600, SERIAL_8N1, 10); // just sets the RX pin - TX pin is not modified from its default` 
or 
`SerialGPS.setPins(10, -1); // changes only the RX pin, keeping the TX pin the same it was before with -1`

mcenderdragon, 2023-08-04:
> for that just use: `SerialGPS.begin(9600, SERIAL_8N1, 10); // just sets the RX pin - TX pin is not modified from its default` or `SerialGPS.setPins(10, -1); // changes only the RX pin, keeping the TX pin the same it was before with -1`
> 
> Both ways work perfectly fine. Please update your Arduino Core to the latest 2.0.11.

This will not work if Serial1 is used, becvuase there the supplied pins are overwritten in the cisntructer (the part I did higlight in the linked source code)

mcenderdragon, 2023-08-04:
also uart has only on 80 and 160Mhz the correct baud rate (even if updateBaudRate is called again). 40Mhz works fine chode wise, but the uart signals are gibberish

SuGlider, 2023-08-04:
>Version v2.0.4

All those issues should be fixed in the latest Arduino Core 2.0.9 (I recommend this version as the latest more stable).
Please update it to fix the issues you have reported.

SuGlider, 2024-08-02:
Closing this issue as we are now on Arduino Core 3.0.x and it is all fixed.

closed 2024-08-02 by SuGlider