espressif/arduino-esp32 #8471: Issue with MIDI with 2.0.10 and 2.0.11

corrados, 2023-07-29:
### Board

ESP32

### Device Description

ESP32 Dev Module

### Hardware Configuration

Hardware fron-end needed for Edrumulus.

### Version

v2.0.10

### IDE Name

Arduino IDE

### Operating System

Linux

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

I am developing an open source edrum module (source code at Github: https://github.com/corrados/edrumulus) on my ESP32 dev module (NodeMCU with dual core ESP32). Since version 2.0.10 and 2.0.11, the MIDI transfer from the board to the PC is unreliable. It seems that some MIDI messages do not come through anymore. On versions <= 2.0.9, everything works fine.

### Sketch

```cpp
Full source code is available at https://github.com/corrados/edrumulus
Flash the latest Git code at main and use tools/run_edrumulus.sh
```

### Debug Message

```plain
Incorrect behaviour: The GUI will not start correctly since the MIDI messages are missing. Using the same script with 2.0.9 will run fine.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2023-07-31:
@me-no-dev Please help with the triage, thanks

corrados, 2023-07-31:
I fear that the cause of the problem might be related to the IDF version: https://github.com/espressif/arduino-esp32/issues/8467#issuecomment-1656803833.

SuGlider, 2023-08-02:
@corrados - Could you give us more information about what is not working?
How does this project work? Does it use I2S and DAC (as pointed as potential issue above)?

corrados, 2023-08-03:
Yes, you are right. I am using both ADCs. But the problem are not the ADCs but the MIDI communication between the ESP32 and the PC. It seems that some MIDI messages get lost.

Edrumulus not only uses the MIDI channel for transmitting the actual pad strikes but also for settings changes. At the startup phase I am configuring Edrumulus by transmitting all settings from the PC to the ESP32 via MIDI. Edrumulus sends a MIDI response for each received settings change. The GUI on the PC checks for these confirmation MIDI events and if they do to arrive, the GUI gets stuck. This is the case if, e.g., some MIDI messages get lost. I never experienced problems with that MIDI communication using 2.0.9 but with 2.0.10/2.0.11 (I tried them both), I always get problems with lost MIDI messages.

On the PC, the GUI is a python script and I then inserted some debugging prints which clearly showed that not all expected MIDI messages came back from the ESP32.

In Edrumulus, I am using the "MIDI.h" library and the serial communication is tunneled through USB.

SuGlider, 2023-08-04:
@corrados - I think that this issue may be related to a heap problem with the Arduino Core 2.0.10/11 as I wrote in https://github.com/espressif/arduino-esp32/issues/8482#issuecomment-1664769461

If possible, please test your project adding `Serial.printf("Available Heap: %d\n",ESP.getFreeHeap());`  in a few strategic places of the code.

Thanks.

corrados, 2023-08-04:
I fear I cannot do this test. When I use the Serial.printf in my debug session, I always have to disable the MIDI communication since it uses exactly the same channel. And if I disable MIDI, I cannot see the issue.

I did a quick test by inserting the code you have given but as expected, the serial-to-MIDI bridge on my PC is confused if invalid strings are received. So I could not get any output showing the free heap, unfortunately.

SuGlider, 2023-08-04:
I see. If you have any UART-USB board with you, you may try to use the `Serial2` to print the HEAP available space.
Or use a second ESP32 to connect `edrumulus_ESP32_UART2` <--> `Second_ESP32_UART2` and then use a sketch the just reads from `Serial2` and prints to `Serial` (using the second ESP32).

Let me know if you want to try it / need any further information about it.

corrados, 2023-08-05:
I have now disabled the MIDI write but can receive MIDI messages. If I startup Edrumulus, it shows:
Available Heap: 151236
Does this help?

corrados, 2023-08-05:
I forgot to give you a value for both versions:
2.0.9: Available Heap: 151236
2.0.11: Available Heap: 116616

corrados, 2023-10-01:
Sorry for the delay. I have just tested versions 2.0.12 and 2.0.13 and they again work fine with Edrumulus. So, the issue was only present in versions 2.0.10 and 2.0.11.

Thanks for fixing :-).

I'll close this issue now.

closed 2023-10-01 by corrados